The-OpenROAD-Project/OpenROAD-flow-scripts #996: Create script to handle io.cfg file to restrict pins on specific side

vijayank88, 2023-04-19:
### Description

Currently through  `set_io_pin_constraint` we can restrict pin placement either side like `top`, `bottom`, `left`, `right`.
But it should handled through manually creating `io.tcl` script and using `IO_CONSTRAINTS` variable to pass customized IO placement during floorplan stage.

Instead of creating `io.tcl` for each design, create custom template with `io.cfg` for specific direction IO placement to automate this.

### Suggested Solution

With the help of our trainees, created tcl script to handle above requirements.

### Additional Context

Updated `gf180/aes` design with our script to read `io.cfg`, which contains IO port list (User input) and place pins on specific direction on #top #bottom #left #right.

`io.cfg` in following format for `aes` design
```
#Note: top,bottom,left, right sides should not modify or alter its order.
#Based on requirement update port list in respective core side.
#        top
#      _________
#     |		|
#left |		| right
#     |_________|
#        bottom
#If either side doesn't have pins to place keep those sides as empty.
#But don't remove any of fixed # side names.
#
#top
text_in*

#bottom
key*

#left
clk
rst
ld

#right
done
text_out*
```

If this enhancement is OK, I request our trainee to create PR for the same. Thanks.

maliberty, 2023-04-19:
The constraints provided in tcl are more capable the just the side (range, grouping, mirroring, etc).  This would create a less capable format.  I'm not sure why a .cfg is preferrable to a .tcl.  It will necessarily be design specific in either case.

vijayank88, 2023-04-19:
Yes. It is design specific requirement to place IO's on specific side based on port list.

oharboe, 2023-04-19:
I thought the io.cfg and macro placement .cfg files were anachronisms as soon as .tcl config files were supported. If someone took the trouble of replacing all example .cfg in ORFS with .tcl files, then perhaps delete .cfg support to reduce amount of code to be maintained?

No?

maliberty, 2023-04-20:
I prefer to avoid config files and stick to the tcl commands.

closed 2023-04-20 by maliberty