The-OpenROAD-Project/OpenROAD-flow-scripts #1001: Speed up load of .odb files in >100mbyte to gigabyte range

oharboe, 2023-04-20:
### Description

Loading a .odb file that is gigabytes takes a long time. Also, the rest of the GUI doesn't scale well.

### Suggested Solution

For large .odb files (100s of megabytes to gigabytes), prompt user to filter out stuff on load to speed up the GUI.

Ideas for things to filter out, based on where the .odb file grows:

- fill cells
- pdn

oharboe, 2023-04-21:
To test this, .odb files in gigabyte range is needed. These can be generated using the mock-array-big.

Below is a 2200um x 2200 um design:

[image]

To generate:

Install Scala as the script below uses Chisel to generate Verilog: https://www.scala-lang.org/download/

Generate Verilog:

```shell
MOCK_ARRAY_WIDTH=16 MOCK_ARRAY_HEIGHT=16 MOCK_ARRAY_DATAWIDTH=64 MOCK_ARRAY_PITCH_SCALE=20 designs/asap7/mock-array-big/configure.sh
```

Run flow:

```
MOCK_ARRAY_WIDTH=16 MOCK_ARRAY_HEIGHT=16 MOCK_ARRAY_DATAWIDTH=64 MOCK_ARRAY_PITCH_SCALE=20 make DESIGN_CONFIG=designs/asap7/mock-array-big/config.mk
```

oharboe, 2023-04-21:
A perf record + report on the above:

```
Samples: 382K of event 'cycles', Event count (approx.): 470955921617
Overhead  Command          Shared Object                   Symbol
  20,17%  openroad         openroad                        [.] odb::operator>>                            ◆
  14,97%  openroad         openroad                        [.] boost::geometry::index::detail::rtree::visi▒
  11,23%  openroad         libc.so.6                       [.] _IO_fread                                  ▒
   8,52%  openroad         openroad                        [.] boost::geometry::index::detail::rtree::redi▒
   4,00%  openroad         libc.so.6                       [.] malloc_consolidate                         ▒
   2,97%  openroad         libc.so.6                       [.] _int_malloc                                ▒
   2,95%  openroad         libc.so.6                       [.] _int_free                                  ▒
   2,91%  openroad         libc.so.6                       [.] malloc                                     ▒
   2,16%  openroad         openroad                        [.] std::_Rb_tree<std::pair<unsigned int, unsig▒
   1,81%  openroad         libc.so.6                       [.] _IO_file_xsgetn                            ▒
   1,78%  openroad         openroad                        [.] std::_Rb_tree<std::pair<unsigned int, unsig▒
   1,33%  openroad         libc.so.6                       [.] unlink_chunk.constprop.0                   ▒
   1,28%  openroad         libc.so.6                       [.] __memmove_avx_unaligned_erms               ▒
   1,25%  openroad         libQt5Gui.so.5.15.6             [.] 0x000000000052d7c6                         ▒
   0,93%  openroad         openroad                        [.] boost::geometry::detail::expand::expand_ind▒
   0,83%  openroad         libstdc++.so.6.0.30             [.] std::_Rb_tree_insert_and_rebalance         ▒
```

maliberty, 2023-04-21:
The runtime is mostly in building the rtrees in the gui.  Secondarily in reading the gcell grid in odb.

oharboe, 2023-04-21:
> The runtime is mostly in building the rtrees in the gui. Secondarily in reading the gcell grid in odb.

This test-case doesn't quite match the most severe performance problems I'm struggling with, but worth a pass at optimizing. Perhaps it can be used as a test-case to skip loading e.g. fill cells?

oharboe, 2023-04-21:
After https://github.com/The-OpenROAD-Project/OpenROAD/pull/3224, the perf report is:

```
Samples: 356K of event 'cycles', Event count (approx.): 441096304269
Overhead  Command          Shared Object                   Symbol
  21,13%  openroad         openroad                        [.] odb::operator>>
  16,47%  openroad         openroad                        [.] boost::geometry::index::detail::rtree::visit
   7,81%  openroad         openroad                        [.] boost::geometry::index::detail::rtree::redis
   4,35%  openroad         libc.so.6                       [.] malloc_consolidate
   3,26%  openroad         libc.so.6                       [.] _int_free
   3,25%  openroad         libstdc++.so.6.0.30             [.] std::istream::read
   3,02%  openroad         libc.so.6                       [.] malloc
   3,01%  openroad         libc.so.6                       [.] _int_malloc
   2,48%  openroad         libstdc++.so.6.0.30             [.] std::basic_filebuf<char, std::char_traits<ch
   2,41%  openroad         openroad                        [.] std::_Rb_tree<std::pair<unsigned int, unsign
   1,79%  openroad         openroad                        [.] std::_Rb_tree<std::pair<unsigned int, unsign
   1,51%  openroad         libstdc++.so.6.0.30             [.] std::basic_streambuf<char, std::char_traits<
   1,44%  openroad         libc.so.6                       [.] unlink_chunk.constprop.0
   1,37%  openroad         libstdc++.so.6.0.30             [.] std::istream::sentry::sentry
```

oharboe, 2023-04-21:
Another profiling run, this time on a large 2_floorplan.odb of a proprietary design:

```
Samples: 620K of event 'cycles', Event count (approx.): 660409318638
Overhead  Command          Shared Object                   Symbol
  57,74%  openroad         openroad                        [.] boost::geometry::index::detail::rtree::visitors::insert<s
  24,54%  openroad         openroad                        [.] boost::geometry::index::detail::rtree::redistribute_eleme
   1,76%  openroad         openroad                        [.] gui::Search::addSNet
   1,54%  openroad         openroad                        [.] odb::dbBox::getViaBoxes
   1,18%  openroad         libc.so.6                       [.] _int_free
   1,15%  openroad         openroad                        [.] boost::geometry::detail::expand::expand_indexed<0ul, 2ul>
   0,91%  openroad         [unknown]                       [k] 0xffffffffaad87f27
   0,91%  openroad         openroad                        [.] boost::variant<boost::geometry::index::detail::rtree::var
   0,79%  openroad         openroad                        [.] odb::_dbBox::getTechLayer
   0,62%  openroad         libc.so.6                       [.] _IO_fread
   0,52%  openroad         libc.so.6                       [.] _int_malloc
   0,50%  openroad         openroad                        [.] boost::variant<boost::geometry::index::detail::rtree::var
   0,47%  openroad         libc.so.6                       [.] malloc
   0,31%  openroad         libc.so.6                       [.] __memmove_avx_unaligned_erms
   0,28%  openroad         libc.so.6                       [.] unlink_chunk.constprop.0
```

oharboe, 2023-04-21:
This is perhaps only modestly interesting, when I disable this code, the design loads in the GUI in a minute or two, rather than 10(?):

```
git diff
diff --git a/src/gui/src/browserWidget.cpp b/src/gui/src/browserWidget.cpp
index 3f5ada5a7..d48b4c279 100644
--- a/src/gui/src/browserWidget.cpp
+++ b/src/gui/src/browserWidget.cpp
@@ -405,6 +405,8 @@ BrowserWidget::ModuleStats BrowserWidget::addInstanceItems(
   };
   std::map<DbInstDescriptor::Type, Leaf> leaf_types;
   for (auto* inst : insts) {
+    if (inst->getMaster()->isFiller())
+    continue;
     auto type = inst_descriptor_->getInstanceType(inst);
     auto& leaf_parent = leaf_types[type];
     if (leaf_parent.item == nullptr) {
diff --git a/src/gui/src/search.cpp b/src/gui/src/search.cpp
index add8e09c8..f8e4c1583 100644
--- a/src/gui/src/search.cpp
+++ b/src/gui/src/search.cpp
@@ -238,10 +238,10 @@ void Search::updateShapes()
   box_shapes_.clear();
   polygon_shapes_.clear();
 
-  for (odb::dbNet* net : block_->getNets()) {
-    addNet(net);
-    addSNet(net);
-  }
+  // for (odb::dbNet* net : block_->getNets()) {
+  //   addNet(net);
+  //   addSNet(net);
+  // }
 
   for (odb::dbBTerm* term : block_->getBTerms()) {
     for (odb::dbBPin* pin : term->getBPins()) {
```

maliberty, 2023-04-22:
I've run it through vtune and there are two issues

1) Loading the gcell data is quite slow.  This is pretty straightforward to fix but is also the smaller problem.  This shows up even in a non-gui load.

2) The pdn is very slow to load.  The issue in V1 has a huge number of cuts due to the m1/m2 follow pins.  I've been using a test case you send earlier and it has >600M cuts.  I'm thinking about better representations for that data as it is mostly repetitive.

Short term you might try dropping ```add_pdn_connect -grid {top} -layers {M1 M2}``` from the pdn and see if that helps for prototyping.  You'll have to skip psm as well if you do so.

rovinski, 2023-04-22:
I recall in a commercial tool for both PDN generation and loading, the PDN size was a huge issue for larger chips. We found out that there was a way to create a "via array" instead of many individual vias, where I believe it essentially just stored (cut type, x-pitch, y-pitch, x-length, y-length) in order to represent e.g. rails with thousands of vias on it.

I don't know if there's anything implicit to LEF/DEF to represent this, but it could definitely be an internal representation thing worth looking at.

maliberty, 2023-04-24:
@rovinski I was thinking along those lines as one approach.  Note that we do sometimes depopulate some vias in order to improve pin access.  In those cases we would have to break up the large vias and give back some of the efficiency.  pdn & drt would both have to be enhanced.

The GUI would have to store the vias rather than the cuts as it does today.  That is certainly doable but only makes sense if we are going to have these big vias.

maliberty, 2023-05-01:
All planned changes are now merged.  Please re-run your design from scratch and report on the new results.

maliberty, 2023-05-02:
Is this with a heat map displayed?  I see gui::HeatMapDataSource in your list.

oharboe, 2023-05-02:
> Is this with a heat map displayed? I see gui::HeatMapDataSource in your list.

No. The heatmap on/off is not persisted across GUI runs.

oharboe, 2023-05-02:
> > > Is this with a heat map displayed? I see gui::HeatMapDataSource in your list.
> > 
> > 
> > No. The heatmap on/off is not persisted across GUI runs.
> 
> It is persisted across runs.

Is there a command line option not to read persisted options? A file I can delete?

Is there a reset to default button?

maliberty, 2023-05-02:
It depends on your platform but for linux they are stored in ~/.config/OpenRoad Project/openroad.conf

oharboe, 2023-05-02:
```

oharboe, 2023-05-02:
If I press ctrl-c as soon as the command below prints "Estimating parasitics", I get this profile:

```

oharboe, 2023-05-03:
From make gui_final with 3.0gByte .odb and real abstracts:

```
Overhead  Command          Shared Object                Symbol
   9,86%  openroad         libstdc++.so.6.0.30          [.] std::istream::read
   6,85%  openroad         openroad                     [.] odb::_dbBox::getTechLayer
   5,83%  openroad         libstdc++.so.6.0.30          [.] std::basic_filebuf<char, std::char_traits<char> >::xsgetn
   5,79%  openroad         libstdc++.so.6.0.30          [.] std::istream::sentry::sentry
   4,85%  openroad         openroad                     [.] SpefLex_lex
   3,81%  openroad         openroad                     [.] boost::geometry::index::detail::rtree::visitors::insert<std::tu
   2,59%  openroad         openroad                     [.] sta::SpefReader::makeNameMapEntry
   2,39%  openroad         openroad                     [.] boost::geometry::index::detail::rtree::visitors::detail::insert
   2,29%  openroad         libstdc++.so.6.0.30          [.] std::basic_streambuf<char, std::char_traits<char> >::xsgetn
   1,93%  openroad         openroad                     [.] boost::geometry::index::detail::rtree::visitors::insert<std::tu
   1,73%  openroad         openroad                     [.] boost::geometry::index::detail::rtree::visitors::insert<std::tu
   1,73%  openroad         libc.so.6                    [.] __strcmp_avx2
   1,71%  openroad         openroad                     [.] sta::stringDeleteCheck
   1,68%  openroad         libc.so.6                    [.] malloc
   1,61%  openroad         openroad                     [.] sta::SpefReader::findParasiticNode
   1,56%  openroad         libc.so.6                    [.] _int_free
   1,55%  openroad         openroad                     [.] boost::geometry::index::detail::rtree::redistribute_elements<bo
   1,47%  openroad         openroad                     [.] SpefParse_parse
   1,36%  openroad         [unknown]                    [k] 0xffffffff8f587f27
   1,21%  openroad         openroad                     [.] odb::dbITerm::getMTerm
   1,18%  openroad         libc.so.6                    [.] _int_malloc
   1,16%  openroad         libc.so.6                    [.] __memmove_avx_unaligned_erms
   1,13%  openroad         openroad                     [.] boost::geometry::index::detail::rtree::redistribute_elements<bo
   1,08%  openroad         openroad                     [.] LibertyLex_lex
   1,03%  openroad         openroad                     [.] odb::dbHashTable<odb::_dbInst>::find
   1,00%  openroad         openroad                     [.] odb::operator>>
```

Same, but ctrl-c as soon as "Loading spef" is printed:

```
Overhead  Command          Shared Object                         Symbol
  19,82%  openroad         libstdc++.so.6.0.30                   [.] std::istream::read
  11,32%  openroad         libstdc++.so.6.0.30                   [.] std::basic_filebuf<char, std::char_traits<char> >::xsg
  11,04%  openroad         libstdc++.so.6.0.30                   [.] std::istream::sentry::sentry
   5,84%  openroad         openroad                              [.] boost::geometry::index::detail::rtree::visitors::inser
   4,43%  openroad         libstdc++.so.6.0.30                   [.] std::basic_streambuf<char, std::char_traits<char> >::x
   3,75%  openroad         openroad                              [.] boost::geometry::index::detail::rtree::visitors::detai
   2,18%  openroad         [unknown]                             [k] 0xffffffff8f587f27
   2,13%  openroad         libc.so.6                             [.] __memmove_avx_unaligned_erms
   2,13%  openroad         openroad                              [.] LibertyLex_lex
   2,06%  openroad         openroad                              [.] odb::operator>>
   1,56%  openroad         libc.so.6                             [.] malloc
   1,50%  openroad         libc.so.6                             [.] _int_malloc
   1,49%  openroad         libstdc++.so.6.0.30                   [.] 0x00000000000a19c0
   1,47%  openroad         openroad                              [.] odb::operator>>
   1,14%  openroad         libc.so.6                             [.] __memset_avx2_unaligned_erms
   1,02%  openroad         [unknown]                             [k] 0xffffffff8f58866c
   0,96%  openroad         libstdc++.so.6.0.30                   [.] 0x00000000000a2120
```

oharboe, 2023-05-03:
Some suspend and resume in debugger:

[image]

gadfort, 2023-05-07:
@maliberty  @eder-matheus when running this example, I noticed
```
[INFO GPL-0030] NumBins: 1048576
[NesterovSolve] Iter: 1 overflow: 0.993684 HPWL: -30794661463191
[NesterovSolve] Iter: 10 overflow: 0.99812 HPWL: -30794665747327
[NesterovSolve] Iter: 20 overflow: 0.99812 HPWL: -30794665747098
```
the HPWL is being printed as negative, so some kind of overflow is occurring.

maliberty, 2023-05-09:
io_lsbIns_0 has no connections and its hpwl should be zero but is coming out as a large negative value.  I'll fix it.  Is it intended to be dangling?

oharboe, 2023-05-09:
> io_lsbIns_0 has no connections and its hpwl should be zero but is coming out as a large negative value. I'll fix it. Is it intended to be dangling?

Sort of. By design it shouldn't connect to anything, I should not be generating it from Chisel, but I was lazy when I wrote the tcl, and at the time I wanted to have as many lsbIns as Outs. Now that I'm using regex for set_io_constraint, I could fix that in the Chisel.

oharboe, 2023-05-09:
@maliberty The performance problems for me are mainly in the GUI at this point. Is there anything else that you want to do before this github issue is closed?