The-OpenROAD-Project/OpenROAD-flow-scripts #1010: All the demo designs of  ASAP7 failed with the same error in yosys synthesis stage: Can't open ABC output file

JiaDYuan, 2023-04-25:
### Subject

[Stage]: Synthesis.

### Describe the bug

All the demo projects with ASAP7 platform (gcd, ibex, jpeg..) failed with the same error messages: Can't open ABC output file .  These are all demos in the designs folder, which shouldn't be. Please check it out yourself.

Here are the ABC synthesis error messages of ibex core:

 53.1.1. Executing ABC.
ERROR: Can't open ABC output file `/tmp/yosys-abc-cXrezK/output.blif'.
Command exited with non-zero status 1
Elapsed time: 1:04.80[h:]min:sec. CPU time: user 64.38 sys 0.38 (99%). Peak memory: 219376KB.
make: *** [Makefile:395: results/asap7/ibex/base/1_1_yosys.v] Error 1

### Expected Behavior

All the demo designs on the ASAP7 platform should pass the whole flow successfully without any error messages, like I did before. 

### Environment

```shell
Git commit: d5b908a31e00957ed6c695144c76859863d29a4c
kernel: Linux 5.10.0-18-amd64
os: Debian GNU/Linux 11 (bullseye)
cmake version 3.24.2
-- The CXX compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-7907-gd5b908a31
-- System name: Linux
-- Compiler: GNU 10.2.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY)
-- The C compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /home/yuanjiadong/mambaforge/bin/python3.10 (found version "3.10.8") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
[... 149 lines cut ...]
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib
    runtime library [libtcl8.6.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/yuanjiadong/mambaforge/lib

  Some of these libraries may not be found correctly.

ponents: serialization system thread
-- Found Boost: /home/yuanjiadong/openroad/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.9HG3ICaC39
```

### To Reproduce

[1_1_yosys.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11324791/1_1_yosys.log)

 I only ran the command  "make DESIGN_CONFIG=./designs/sky130hd/ibex/config.mk" under the folder /OpenROAD-flow-scripts/flow.

JiaDYuan, 2023-04-25:
> Are you out of disk space or a permission problem perhaps? This runs in the nightly CI so it must be a local issue.

Nope, the disk space is enough. 

Filesystem      Size  Used Avail Use% Mounted on
udev             56G     0   56G   0% /dev
tmpfs            12G  2.2M   12G   1% /run
/dev/md0         67G   28G   39G  42% /
tmpfs            56G  336K   56G   1% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
/dev/nvme3n1p1  813G  237G  535G  31% /opt
/dev/sde1       236G  236G     0 100% /mnt/hddroot
/dev/md1p1      402G  239G  143G  63% /home
/dev/md2p1      1.1T  724G  320G  70% /mnt/bck
/dev/sda1       1.8T  947G  793G  55% /mnt/USRP_Exp
tmpfs            12G   96K   12G   1% /run/user/1019
tmpfs            12G  704K   12G   1% /run/user/1028

Would you try it out on your local mechine?  And notice that this error only happen with the PDK ASAP7. Other platforms (nangate45, sky130hd, etc.) are all ok. So I am not sure it is a local problem or not..

maliberty, 2023-04-25:
You can see it passing here https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Public/job/public_tests_all/job/master/684/execution/node/531/log/?consoleFull
```
8.24. Executing ABC pass (technology mapping using ABC).
8.24.1. Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
8.25. Executing OPT pass (performing simple optimizations).
```

maliberty, 2023-04-25:
I don't know if it relates but
```
/dev/sde1 236G 236G 0 100% /mnt/hddroot
```

JiaDYuan, 2023-04-25:
> I don't know if it relates but
> 
> ```
> /dev/sde1 236G 236G 0 100% /mnt/hddroot
> ```

No it's not related, it's running on a server. The process is running on  /dev/md1p1 402G 239G 143G 63% /home

JiaDYuan, 2023-04-25:
> You can see it passing here https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Public/job/public_tests_all/job/master/684/execution/node/531/log/?consoleFull
>
> ```
> 8.24. Executing ABC pass [...]

May I ask when did this be run? I noticed there were some modifications in the ASAP7 platform folder recently.

rovinski, 2023-04-25:
> No it's not related, it's running on a server. The process is running on /dev/md1p1 402G 239G 143G 63% /home

Check where your `/tmp` directory is mounted. Yosys uses this for temporary storage (as seen in `ERROR: Can't open ABC output file '/tmp/yosys-abc-cXrezK/output.blif'.`)

maliberty, 2023-04-25:
https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Public/job/public_tests_all/job/master/684/

[image]

JiaDYuan, 2023-04-27:
@maliberty @rovinski 

Hi, I re-installed OpenROAD-flow-scripts and spare some disk sapce, but it's still not solved.

Then I found out what the problem is. I ran the command: 
make DESIGN_CONFIG=./designs/asap7/ibex/config.mk -j8

So I tried to accelerate the process with "-j8", but it ended up with the errors, when I removed it, all the errors are gone. But it is weird, I don't understand why adding "-j8" option would cause errors?

maliberty, 2023-04-27:
Perhaps there is some missing dependency?  In any case -j won't speed things up as the flow is quite linear.

maliberty, 2023-04-27:
OpenrRoad itself uses threads but the -j is to make not OR.

closed 2023-04-27 by maliberty