The-OpenROAD-Project/OpenROAD-flow-scripts #1028: genElapsedTime.py generates wrong values

oharboe, 2023-05-03:
### Subject

[Documentation] for documentation errors.

### Describe the bug

```
Log                       Elapsed seconds
1_1_yosys                       2678
2_1_floorplan                    507
2_2_floorplan_io                 128
2_4_mplace                       125
2_5_tapcell                      141
2_6_pdn                         1885
3_1_place_gp_skip_io            3661
3_2_place_iop                    253
3_3_place_gp                   15782
3_4_resizer                     3169
3_5_opendp                      5736
4_1_cts                        10991
4_2_cts_fillcell                3083
5_1_fastroute                  11825
5_2_TritonRoute                 8689
6_1_merge                       6996
6_report                       20193
```

Should be:

```
$ grep Elapsed logs/FooBar/*.log | awk 'match($0, /[a-zA-Z_0-9/]+\/([a-zA-Z_0-9]+)\.log:Elapsed time: ([0-9]+:)?([0-9]+):([0-9]+)/ , arr) { printf "%-25s %10d\n", arr[1], arr[2]*3600 + arr[3] * 60 + arr[4] }'
1_1_yosys                         44
2_1_floorplan                      7
2_2_floorplan_io                   2
2_4_mplace                         2
2_5_tapcell                        2
2_6_pdn                           30
3_1_place_gp_skip_io              61
3_2_place_iop                      4
3_3_place_gp                     263
3_4_resizer                       52
3_5_opendp                        94
4_1_cts                          183
4_2_cts_fillcell                  51
5_1_fastroute                    196
5_2_TritonRoute                 8689
6_1_merge                        115
6_report                         336
generate_abstract                 93
```

Raw data:

```
$ grep Elapsed logs/asap7/FooBar/base/*.log 
logs/asap7/FooBar/base/1_1_yosys.log:Elapsed time: 0:44.38[h:]min:sec. CPU time: user 43.30 sys 1.07 (99%). Peak memory: 319928KB.
logs/asap7/FooBar/base/2_1_floorplan.log:Elapsed time: 0:07.87[h:]min:sec. CPU time: user 7.71 sys 0.16 (99%). Peak memory: 398996KB.
logs/asap7/FooBar/base/2_2_floorplan_io.log:Elapsed time: 0:02.08[h:]min:sec. CPU time: user 1.94 sys 0.12 (99%). Peak memory: 313160KB.
logs/asap7/FooBar/base/2_4_mplace.log:Elapsed time: 0:02.05[h:]min:sec. CPU time: user 1.95 sys 0.09 (99%). Peak memory: 306552KB.
logs/asap7/FooBar/base/2_5_tapcell.log:Elapsed time: 0:02.21[h:]min:sec. CPU time: user 2.06 sys 0.15 (99%). Peak memory: 345612KB.
logs/asap7/FooBar/base/2_6_pdn.log:Elapsed time: 0:30.85[h:]min:sec. CPU time: user 30.54 sys 0.30 (99%). Peak memory: 835220KB.
logs/asap7/FooBar/base/3_1_place_gp_skip_io.log:Elapsed time: 1:01.01[h:]min:sec. CPU time: user 60.43 sys 0.56 (99%). Peak memory: 1366664KB.
logs/asap7/FooBar/base/3_2_place_iop.log:Elapsed time: 0:04.13[h:]min:sec. CPU time: user 3.78 sys 0.34 (99%). Peak memory: 672836KB.
logs/asap7/FooBar/base/3_3_place_gp.log:Elapsed time: 4:23.02[h:]min:sec. CPU time: user 258.54 sys 4.41 (99%). Peak memory: 9342860KB.
logs/asap7/FooBar/base/3_4_resizer.log:Elapsed time: 0:52.49[h:]min:sec. CPU time: user 49.73 sys 2.74 (99%). Peak memory: 4439744KB.
logs/asap7/FooBar/base/3_5_opendp.log:Elapsed time: 1:34.96[h:]min:sec. CPU time: user 89.06 sys 5.86 (99%). Peak memory: 11090868KB.
logs/asap7/FooBar/base/4_1_cts.log:Elapsed time: 3:03.11[h:]min:sec. CPU time: user 179.07 sys 3.99 (99%). Peak memory: 10873344KB.
logs/asap7/FooBar/base/4_2_cts_fillcell.log:Elapsed time: 0:51.23[h:]min:sec. CPU time: user 44.17 sys 7.04 (99%). Peak memory: 13190904KB.
logs/asap7/FooBar/base/5_1_fastroute.log:Elapsed time: 3:16.65[h:]min:sec. CPU time: user 186.75 sys 9.84 (99%). Peak memory: 13511260KB.
logs/asap7/FooBar/base/5_2_TritonRoute.log:Elapsed time: 2:24:49[h:]min:sec. CPU time: user 259015.71 sys 23279.34 (3248%). Peak memory: 37466284KB.
logs/asap7/FooBar/base/6_1_merge.log:Elapsed time: 1:55.96[h:]min:sec. CPU time: user 112.05 sys 3.77 (99%). Peak memory: 10317524KB.
logs/asap7/FooBar/base/6_report.log:Elapsed time: 5:36.33[h:]min:sec. CPU time: user 328.67 sys 7.37 (99%). Peak memory: 12458828KB.
logs/asap7/FooBar/base/generate_abstract.log:Elapsed time: 1:33.25[h:]min:sec. CPU time: user 89.69 sys 3.53 (99%). Peak memory: 9052764KB.
```

### Expected Behavior

See above

### Environment

```shell
ORFS: 200988f8a6d8e659cc4efbed479181a17ecfaf39
```

### To Reproduce

See above

maliberty, 2023-05-03:
@habibayassin I'll fix this with #1036 but your unit tests don't run for me and the result in test_longer_elapsed_time is incorrect.  Please fix the tests.

habibayassin, 2023-05-03:
@maliberty for this version of test file you can run each individual test case on its own `python -m unittest test_genElapsedTime.TestElapsedTime.test_longer_elapsed_time`. There’s a [PR](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1014) that will be merged soon that runs the tests collectively

closed 2023-05-03 by maliberty