The-OpenROAD-Project/OpenROAD-flow-scripts #1055: [ERROR GUI-0077] Could not connect to any X display.

oharboe, 2023-05-09:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

The normal flow is trying to connect to a screen, which fails. This used to work.

To reproduce:

1. On a machine with a display, run e.g. "screen", used to run commands while logged out.
2. Run a flow.

```
==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 1776 u^2 13% utilization.

[WARNING GUI-0076] QXcbConnection: Could not connect to display localhost:10.0
[ERROR GUI-0077] Could not connect to any X display.
Error: final_report.tcl, 66 GUI-0077
Elapsed time: 0:04.27[h:]min:sec. CPU time: user 4.16 sys 0.08 (99%). Peak memory: 253244KB.
```

### Expected Behavior

The normal flow should not require a screen.

### Environment

```shell
latest master
```

### To Reproduce

See above

maliberty, 2023-05-11:
I tried "DISPLAY=NOTHING make" and I get:

```
[WARNING GUI-0076] QXcbConnection: Could not connect to display NOTHING
[ERROR GUI-0077] Could not connect to any X display.
Error: final_report.tcl, 66 GUI-0077
```
but no crash.  The error itself is reasonable.

oharboe, 2023-05-12:
> I tried "DISPLAY=NOTHING make" and I get:
> 
> ```
> [WARNING GUI-0076] QXcbConnection: Could not connect to display NOTHING
> [ERROR GUI-0077] Could not connect to any X display.
> Error: final_report.tcl, 66 GUI-0077
> ```
> 
> but no crash. The error itself is reasonable.

I agree the error message is reasonable, but the build should not attempt to connect to a GUI.

The SEGFAULT is a separate minor concern, I'm sure easily fixed if it can be reproduced.

oharboe, 2023-05-12:
```
rm -f logs/nangate45/gcd/base/6_report.log && DISPLAY=nothing make
```

Results in:

```
terminate called after throwing an instance of 'std::runtime_error'
  what():  GUI-0077
Signal 6 received
Stack trace:
 0# 0x000055D35D274F4B in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007FD445AD7520 in /lib/x86_64-linux-gnu/libc.so.6
 2# pthread_kill in /lib/x86_64-linux-gnu/libc.so.6
 3# raise in /lib/x86_64-linux-gnu/libc.so.6
 4# abort in /lib/x86_64-linux-gnu/libc.so.6
 5# 0x00007FD445E68BBE in /lib/x86_64-linux-gnu/libstdc++.so.6
 6# 0x00007FD445E7424C in /lib/x86_64-linux-gnu/libstdc++.so.6
 7# 0x00007FD445E73229 in /lib/x86_64-linux-gnu/libstdc++.so.6
 8# __gxx_personality_v0 in /lib/x86_64-linux-gnu/libstdc++.so.6
 9# 0x00007FD445CD3C64 in /lib/x86_64-linux-gnu/libgcc_s.so.1
10# _Unwind_Resume in /lib/x86_64-linux-gnu/libgcc_s.so.1
11# 0x00007FD44665A2F9 in /lib/x86_64-linux-gnu/libQt5Core.so.5
12# QMessageLogger::fatal(char const*, ...) const in /lib/x86_64-linux-gnu/libQt5Core.so.5
13# QGuiApplicationPrivate::createPlatformIntegration() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
14# QGuiApplicationPrivate::createEventDispatcher() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
15# QCoreApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Core.so.5
16# QGuiApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Gui.so.5
17# QApplicationPrivate::init() in /lib/x86_64-linux-gnu/libQt5Widgets.so.5
18# gui::startGui(int&, char**, Tcl_Interp*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
19# show(char const*, bool) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
20# 0x000055D35DA33A18 in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
21# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
22# 0x00007FD449970028 in /lib/x86_64-linux-gnu/libtcl8.6.so
23# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
24# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
25# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
26# ord::tclAppInit(Tcl_Interp*) in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
27# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
28# main in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
29# 0x00007FD445ABED90 in /lib/x86_64-linux-gnu/libc.so.6
30# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
31# _start in /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 6
Elapsed time: 0:00.58[h:]min:sec. CPU time: user 0.47 sys 0.03 (87%). Peak memory: 134528KB.
make: *** [Makefile:683: logs/nangate45/gcd/base/6_report.log] Error 134

```

oharboe, 2023-05-12:
Ah, it is not a crash it is an error with a stack trace.

Here the GUI is being initialized:

[image]

oharboe, 2023-05-12:
If DISPLAY is set to an invalid value, then the build will fail with a runtime error and a stack trace.

The workaround is to run `unset DISPLAY` if the display is not valid.

It would appear that OpenROAD is trying to connect to the DISPLAY, even if it is not required for save_images.tcl to work.

Ideally, OpenROAD would not even initialize the unecessary GUI component when running save_images.tcl from the make flow.

oharboe, 2024-07-14:
@maliberty Isn't this fixed now?

oharboe, 2024-07-21:
Tried it. Fixed.

closed 2024-07-21 by oharboe