The-OpenROAD-Project/OpenROAD-flow-scripts #1071: OpenRoad Gui not openning  .. Loading spef

mbautista-lab, 2023-05-15:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I am attempting to execute the command "make gui_final DESIGN_CONFIG=./designs/sky130hd/gcd/config.mk", but it appears to be stuck at the "Loading spef" stage.

[INFO][FLOW] Using platform directory ./platforms/sky130hd
ODB_FILE=./results/sky130hd/gcd/base/6_final.odb /home/mgbautista/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  ./scripts/gui.tcl
This plugin does not support propagateSizeHints()
OpenROAD v2.0-7988-g20b907452
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Loading spef

### Expected Behavior

expect to load and see  GUI

### Environment

```shell
NAME="Ubuntu"
VERSION="20.04.6 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.6 LTS"
VERSION_ID="20.04"
```

### To Reproduce

n/a

### Relevant log output

```shell
n/a
```

maliberty, 2023-05-16:
You need to provide a test case that shows the problem.

vijayank88, 2023-05-16:
@mbautista-lab 
Is that local installation or docker based?

mbautista-lab, 2023-05-16:
I installed it on our remote server and run locally. Im not using docker. 
I think I figured out a solution.  So I tried  " ssh -Y " , and it works.  This allows to establish a secure connection and enables X11 forwarding. And now I'm able to see the GUI.

maliberty, 2023-05-17:
You will need -Y (or -X) for any X11 app to work over the network so this isn't particular to OR.  Would you like to make a PR for a doc update?

closed 2023-05-17 by maliberty

mbautista-lab, 2023-05-17:
I open a PR here https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1073