The-OpenROAD-Project/OpenROAD-flow-scripts #1080: Dependencies should not exist if required package exists

vijayank88, 2023-05-19:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Trying to run latest `sudo ./setup.sh` to install dependencies.
Facing following error and package installer exit as KLayout already installed.
```
klayout-0.27.10-0.x86_64.rpm                                                                                                                                           |  29 MB  00:00:03     
Examining /var/tmp/yum-root-yDnTtz/klayout-0.27.10-0.x86_64.rpm: klayout-0.27.10-0.x86_64
/var/tmp/yum-root-yDnTtz/klayout-0.27.10-0.x86_64.rpm: does not update installed package.
Error: Nothing to do
```
Please update the dependency installer script to not to exit if required or above version already installed.

### Expected Behavior

dependency installer script to not to exit if required or above version already installed.

### Environment

```shell
With latest master
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.49.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
```

### To Reproduce

Install KLayout and  try running
```
sudo ./setup.sh
```

maliberty, 2023-05-20:
@vijayank88 can you fix this or do you need help?

vijayank88, 2023-05-20:
Currently I used 
yum remove klayout 
to remove previous installation and continue to complete the setup.sh script.

So expecting permanent fix, not a high priority.

maliberty, 2023-05-20:
I meant do you want to try to fix the setup.sh?

oharboe, 2023-05-27:
@maliberty see similar issue #989

vijayank88, 2023-05-27:
> I meant do you want to try to fix the setup.sh?

Ok. I will test with different environment and make a PR for this.

closed 2023-06-08 by vvbandeira