The-OpenROAD-Project/OpenROAD-flow-scripts #1095: build error in Open ROAD

yusc65, 2023-05-24:
### Subject

[Build] for compilation-related issues.

### Describe the bug

A error below occurs during openroad build.

I'm running on Ubuntu 20.04.5 LTS
here's my error code

-----------------------------------------------------------------------------------------
[100%] Linking CXX executable openroad
cd /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/tools/OpenROAD/build/src && /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/bin/cmake -E cmake_link_script CMakeFiles/openroad.dir/link.txt --verbose=1
/usr/bin/c++ -O3 -DNDEBUG -Wl,--export-dynamic -rdynamic CMakeFiles/openroad.dir/Design.cc.o CMakeFiles/openroad.dir/Tech.cc.o CMakeFiles/openroad.dir/OpenRoad.cc.o CMakeFiles/openroad.dir/Main.cc.o -o openroad  -Wl,-rpath,/data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib:/home/scyu/anaconda3/lib:::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::::: ifp/src/ifp.a pad/pad.a openroad_swig.a gpl/gpl.a dpl/dpl.a dpo/dpo.a fin/fin.a rsz/src/rsz.a ppl/ppl.a stt/stt.a dbSta/src/dbSta.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/libOpenSTA.a odb/src/swig/tcl/odbtcl.a rcx/src/rcx.a rmp/src/rmp.a cts/src/cts.a grt/grt.a tap/src/tap.a gui/gui.a drt/drt.a dst/dst.a mpl/mpl.a psm/src/psm.a ant/src/ant.a upf/src/upf.a utl/utl.a pdn/src/pdn.a dft/dft.a ../third-party/abc/libabc.a /usr/lib/x86_64-linux-gnu/libtcl.so -lpthread mpl2/mpl2.a par/par.a /usr/lib/x86_64-linux-gnu/libtclreadline.so /usr/lib/x86_64-linux-gnu/libz.so _openroad_swig_py.a odb/src/swig/python/_odb_py.a ifp/src/_ifp_py.a utl/_utl_py.a ant/src/_ant_py.a grt/_grt_py.a gpl/_gpl_py.a dpl/_dpl_py.a mpl/_mpl_py.a ppl/_ppl_py.a tap/src/_tap_py.a cts/src/_cts_py.a drt/_drt_py.a dpo/_dpo_py.a fin/_fin_py.a rcx/src/_rcx_py.a rmp/src/_rmp_py.a stt/_stt_py.a psm/src/_psm_py.a pdn/src/_pdn_py.a dft/_dft_py.a par/_par_py.a /usr/lib/x86_64-linux-gnu/librt.so upf/src/libupf_lib.a mpl2/libmpl2_lib.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libortools.so.9.5.2237 /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_parse.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_usage.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_usage_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_marshalling.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_reflection.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_config.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_private_handle_accessor.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_commandlineflag.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_commandlineflag_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_flags_program_name.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_distributions.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_seed_sequences.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_internal_pool_urbg.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_internal_randen.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_internal_randen_hwaes.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_internal_randen_hwaes_impl.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_internal_randen_slow.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_internal_platform.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_internal_seed_material.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_random_seed_gen_exception.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_raw_hash_set.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_hashtablez_sampler.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_hash.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_city.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_low_level_hash.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_leak_check.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_statusor.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_status.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_cord.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_cordz_info.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_cord_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_cordz_functions.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_exponential_biased.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_cordz_handle.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_bad_optional_access.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_strerror.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_bad_variant_access.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_str_format_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_synchronization.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_stacktrace.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_symbolize.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_debugging_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_demangle_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_graphcycles_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_malloc_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_time.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_strings.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_strings_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_base.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_spinlock_wait.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_throw_delegate.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_int128.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_civil_time.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_time_zone.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_bad_any_cast_impl.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_raw_logging_internal.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libabsl_log_severity.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libprotobuf.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libre2.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libCbcSolver.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libOsiCbc.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libCbc.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libCgl.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libClpSolver.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libOsiClp.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libClp.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/lib/libOsi.a /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/dependencies/ [... cut ...]
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.LPXT2EpU9I
```

### To Reproduce

build error

vijayank88, 2023-05-24:
@yusc65 
Can you try with following option?
```
./build_openroad.sh --local --threads 1
```

yusc65, 2023-05-24:
@vijayank88 
Thank you for your quick response.
After running the clean command, I ran build again(./build_openroad.sh --local). The same error message occurs.
Here's the new Env.sh message. 

---------------------------------------------------------------------------------------------
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.4.0-135-generic
os: Ubuntu 20.04.5 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8163-g9a5b2cd65
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/libgtest.a  
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
[... 58 lines cut ...]
-- TCL header: /usr/include/tcl/tcl.h
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- SSTA: 0
-- STA executable: /data/scyu/project/OpenROAD/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- Found re2: /opt/or-tools/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found Clp: /opt/or-tools/lib/cmake/Clp/ClpConfig.cmake (found version "1.17.7") 
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.EEuYDrA6YK

maliberty, 2023-05-24:
Can you remove anaconda from your path and try again?  It seems like it is has competing libraries to the system ones.

vvbandeira, 2023-05-24:
Running with any anaconda env activate can cause these problems. On my machine, I just run `conda deactivate` before compiling.

yusc65, 2023-05-25:
@maliberty @vvbandeira 

After conda deactivate, I started again from the beginning(git clone).
It works!! 
Thank you very much!!!

closed 2023-05-25 by yusc65