The-OpenROAD-Project/OpenROAD-flow-scripts #1099: Failing to Build w/ Docker (Klayout)

RaidToRadar, 2023-05-27:
### Subject

[Documentation] for documentation errors.

### Describe the bug

When building the using docker, this issue arises, perhaps there is a broken link:

#0 335.3 Setting up ruby3.0 (3.0.2-7ubuntu2.3) ...
#0 335.3 Setting up ruby (1:3.0~exp1) ...
#0 335.3 Setting up rake (13.0.6-2) ...
#0 335.4 Setting up libruby3.0:amd64 (3.0.2-7ubuntu2.3) ...
#0 335.4 Setting up ruby-rubygems (3.3.5-2) ...
#0 335.4 Setting up ruby3.0-dev:amd64 (3.0.2-7ubuntu2.3) ...
#0 335.4 Setting up ruby-dev:amd64 (1:3.0~exp1) ...
#0 335.4 Processing triggers for man-db (2.10.2-1) ...
#0 335.6 Processing triggers for dbus (1.12.20-2ubuntu4.1) ...
#0 335.6 Processing triggers for fontconfig (2.13.1-4.2ubuntu5) ...
#0 335.7 Processing triggers for hicolor-icon-theme (0.17-2) ...
#0 335.7 Processing triggers for libglib2.0-0:amd64 (2.72.4-0ubuntu2) ...
#0 335.8 Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
#0 335.8 Setting up qtmultimedia5-dev:amd64 (5.15.3-1) ...
#0 335.9 --2023-05-27 22:56:25--  https://www.klayout.org/downloads/Ubuntu-22/klayout_0.27.10-1_amd64.deb
#0 335.9 Resolving www.klayout.org (www.klayout.org)... 178.77.102.107
#0 335.9 Connecting to www.klayout.org (www.klayout.org)|178.77.102.107|:443... connected.
#0 336.4 HTTP request sent, awaiting response... 404 Not Found
#0 336.5 2023-05-27 22:56:26 ERROR 404: Not Found.
#0 336.5 
------
Dockerfile.dev:15
--------------------
  14 |     
  15 | >>> RUN ./DependencyInstaller.sh \
  16 | >>>       && rm -rf /tmp/installer
  17 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c ./DependencyInstaller.sh       && rm -rf /tmp/installer" did not complete successfully: exit code: 8

### Expected Behavior

Build the docker image to run ORFS

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-41-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
./etc/Env.sh: line 54: cmake: command not found

**** Note that the repository was just pulled when this code was ran
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
export OS_NAME=ubuntu22.04
./etc/DockerHelper.sh create -target=dev -os=$OS_NAME

AmirKavousi, 2023-05-28:
I got the same error. Is there any fix for that?

RaidToRadar, 2023-05-28:
> I got the same error. Is there any fix for that?

I think it may just be a broken link in the docker that builds all the dependencies. I think we just have to sit tight until that link is broken. About 2 days ago openroad had built correctly and ran for on this same computer, so to iterate, I believe the link for downloading Klayout just needs to be updated.

RaidToRadar, 2023-05-28:
> I got the same error. Is there any fix for that?

I am going to try to see if it's an issue with the link, by changing the link in /etc/DependencyInstaller.sh

rovinski, 2023-05-28:
See #1101

maliberty, 2023-05-28:
#1101 is merged - please try again

RaidToRadar, 2023-05-28:
Okay everything is working as expected now, Thank you for your quick response time :)

closed 2023-05-28 by RaidToRadar