The-OpenROAD-Project/OpenROAD-flow-scripts #1100: Error while ./build_openroad.sh

AmirKavousi, 2023-05-27:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Hello, I am using docker installation. I clone then build, but ./build_openroad.sh does not complete successfully.

### Expected Behavior

Should have run successfully, and should have verified installation correctly.

### Environment

```shell
OPENROAD: /Users/apple/OpenROAD-flow-scripts/tools/OpenROAD
```

### To Reproduce

[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11583246/build_openroad.log)

When I using:
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh

### Relevant log output

```shell
Loaded plugins: [fastestmirror,]([url](url)) ovl
#9 26.17 Loading mirror speeds from cached hostfile
#9 29.47  * base: mirrors.xtom.com
#9 29.47  * epel: mirrors.sonic.net
#9 29.47  * extras: mirrors.xtom.com
#9 29.47  * updates: mirrors.xtom.com
#9 29.86 https://repo.ius.io/7/aarch64/repodata/repomd.xml: [Errno 14] HTTPS Error 404 - Not Found
#9 29.86 Trying other mirror.
#9 29.86 To address this issue please refer to the below wiki article 
#9 29.86 
#9 29.86 https://wiki.centos.org/yum-errors
#9 29.86 
#9 29.86 If above article doesn't help to resolve this issue please use https://bugs.centos.org/.
#9 29.86 
#9 29.87 
#9 29.87 
#9 29.87  One of the configured repositories failed (IUS for Enterprise Linux 7 - aarch64),
#9 29.87  and yum doesn't have enough cached data to continue. At this point the only
#9 29.87  safe thing yum can do is fail. There are a few ways to work "fix" this:
#9 29.87 
#9 29.87      1. Contact the upstream for the repository and get them to fix the problem.
#9 29.87 
#9 29.87      2. Reconfigure the baseurl/etc. for the repository, to point to a working
#9 29.87         upstream. This is most often useful if you are using a newer
#9 29.87         distribution release than is supported by the repository (and the
#9 29.87         packages for the previous distribution release still work).
#9 29.87 
#9 29.87      3. Run the command with the repository temporarily disabled
#9 29.87             yum --disablerepo=ius ...
#9 29.87 
#9 29.87      4. Disable the repository permanently, so yum won't use it by default. Yum
#9 29.87         will then just ignore the repository until you permanently enable it
#9 29.87         again or use --enablerepo for temporary usage:
#9 29.87 
#9 29.87             yum-config-manager --disable ius
#9 29.87         or
#9 29.87             subscription-manager repos --disable=ius
#9 29.87 
#9 29.87      5. Configure the failing repository to be skipped, if it is unavailable.
#9 29.87         Note that yum will try to contact the repo. when it runs most commands,
#9 29.87         so will have to try and fail each time (and thus. yum will be be much
#9 29.87         slower). If it is a very temporary problem though, this is often a nice
#9 29.87         compromise:
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11583235/build_openroad.log)

#9 29.87 
#9 29.87             yum-config-manager --save --setopt=ius.skip_if_unavailable=true
#9 29.87 
#9 29.87 failure: repodata/repomd.xml from ius: [Errno 256] No more mirrors to try.
#9 29.87 https://repo.ius.io/7/aarch64/repodata/repomd.xml: [Errno 14] HTTPS Error 404 - Not Found
#9 ERROR: process "/bin/sh -c ./DependencyInstaller.sh       && rm -rf /tmp/installer" did not complete successfully: exit code: 1
------
 > [5/5] RUN ./DependencyInstaller.sh       && rm -rf /tmp/installer:
#9 29.87      5. Configure the failing repository to be skipped, if it is unavailable.
#9 29.87         Note that yum will try to contact the repo. when it runs most commands,
#9 29.87         so will have to try and fail each time (and thus. yum will be be much
#9 29.87         slower). If it is a very temporary problem though, this is often a nice
#9 29.87         compromise:
#9 29.87 
#9 29.87             yum-config-manager --save --setopt=ius.skip_if_unavailable=true
#9 29.87 
#9 29.87 failure: repodata/repomd.xml from ius: [Errno 256] No more mirrors to try.
#9 29.87 https://repo.ius.io/7/aarch64/repodata/repomd.xml: [Errno 14] HTTPS Error 404 - Not Found
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11583234/build_openroad.log)

------
Dockerfile.dev:15
--------------------
  14 |     
  15 | >>> RUN ./DependencyInstaller.sh \
  16 | >>>       && rm -rf /tmp/installer
  17 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c ./DependencyInstaller.sh       && rm -rf /tmp/installer" did not complete successfully: exit code: 1
```

rovinski, 2023-05-28:
I don't know if we support docker with Apple silicon. You are probably going to run into issues with the package manager support for arm64. Try building natively with Homebrew.

AmirKavousi, 2023-05-28:
Thank you for your prompt reply. In order to meet the deadline of my forthcoming journal article, I have decided to use an Intel-based Mac for running OpenRoad, considering potential compatibility issues with the Apple Silicon Mac. Unfortunately, I'm currently experiencing an error during the process. I would greatly appreciate any assistance you can provide to help me troubleshoot and resolve this issue:

508.3 
#9 508.3 Complete!
#9 508.5 Loaded plugins: fastestmirror, ovl
#9 509.4 Cannot open: https://www.klayout.org/downloads/CentOS_7/klayout-0.27.10-0.x86_64.rpm. Skipping.
#9 509.4 Error: Nothing to do
#9 ERROR: process "/bin/sh -c ./DependencyInstaller.sh       && rm -rf /tmp/installer" did not complete successfully: exit code: 1
------
 > [5/5] RUN ./DependencyInstaller.sh       && rm -rf /tmp/installer:
#9 508.3   rubygem-io-console.x86_64 0:0.4.2-39.el7_9                                    
#9 508.3   rubygem-json.x86_64 0:1.7.7-39.el7_9                                          
#9 508.3   rubygem-psych.x86_64 0:2.0.0-39.el7_9                                         
#9 508.3   rubygem-rdoc.noarch 0:4.0.0-39.el7_9                                          
#9 508.3   rubygems.noarch 0:2.0.14.1-39.el7_9                                           
#9 508.3 
#9 508.3 Complete!
#9 508.5 Loaded plugins: fastestmirror, ovl
#9 509.4 Cannot open: https://www.klayout.org/downloads/CentOS_7/klayout-0.27.10-0.x86_64.rpm. Skipping.
#9 509.4 Error: Nothing to do
------
Dockerfile.dev:15
--------------------
  14 |     
  15 | >>> RUN ./DependencyInstaller.sh \
  16 | >>>       && rm -rf /tmp/installer
  17 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c ./DependencyInstaller.sh       && rm -rf /tmp/installer" did not complete successfully: exit code: 1

rovinski, 2023-05-28:
Seems like the KLayout version upstream has disappeared. That's unfortunate that they don't keep releases archived indefinitely.
Try checking out #1101 and building that.

maliberty, 2023-05-28:
#1101 is merged - please try again

vvbandeira, 2023-06-07:
@rovinski can we close this? did you had a chance to try again?

StrawHatVigz, 2023-06-13:
I am also facing similar error even when i built docker natively using homebrew