The-OpenROAD-Project/OpenROAD-flow-scripts #1108: ./build_openroad.sh fails linking mpl2_test despite mpl2 being removed on M1

stefanottili, 2023-05-30:
### Subject

[Build] for compilation-related issues.

### Describe the bug

It used to be possible to build openroad on M1, but it currently fails with mpl2 issues despite mpl2 being "removed".

./build_openroad.sh --local
..
-- Removing MPL2 and PAR to avoid run time fatal error.
..
[ 98%] Linking CXX executable mpl2_test
cd /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E cmake_link_script CMa ...
/Library/Developer/CommandLineTools/usr/bin/c++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -Wl,-sea .....
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks'
Undefined symbols for architecture arm64:
  "absl::lts_20230125::ParseCommandLine(int, char**)", referenced from:
      InitGoogle(char const*, int*, char***, bool) in libmpl2_lib.a(bus_synthesis.cpp.o)
ld: symbol(s) not found for architecture arm64

### Expected Behavior

a) it would be nice if mpl2 would be enabled on M1, the DependencyInstaller installs the or-tools after all.
b) since mpl2 is currently disabled, should linking mpl2_test be skipped for the build to succeed ?

### Environment

```shell
./etc/Env.sh 
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Darwin 22.5.0
os:             macOS           13.4
cmake version 3.26.4
CMake Warning at src/CMakeLists.txt:226 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-7988-g20b907452
-- System name: Darwin
-- Compiler: AppleClang 14.0.3.14030022
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
-- The C compiler identification is AppleClang 14.0.3.14030022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /opt/homebrew/Frameworks/Python.framework/Versions/3.11/bin/python3.11 (found version "3.11.3") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
[... 172 lines cut ...]
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Removing PAR_PY to avoid run time fatal error.
-- Configuring done (1.7s)
-- Generating done (0.8s)
-- Build files have been written to: /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build
...
[ 98%] Built target mpl2
/Library/Developer/CommandLineTools/usr/bin/make  -f src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/build.make src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/depend
cd /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E cmake_depends "Unix Makefiles" /Users/user/OpenROAD-flow-scripts/tools/OpenROAD /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/src/mpl2/test/cpp /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/DependInfo.cmake --color=
/Library/Developer/CommandLineTools/usr/bin/make  -f src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/build.make src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/build
[ 98%] Linking CXX executable mpl2_test
cd /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E cmake_link_script CMakeFiles/mpl2_test.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/mpl2_test.dir/mpl2_test.cc.o -o mpl2_test -F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks  -Wl,-rpath,/opt/homebrew/opt/or-tools/lib -Wl,-rpath,/opt/homebrew/lib ../../../../lib/libgtest.a ../../../../lib/libgtest_main.a ../../libmpl2_lib.a ../../../../lib/libgtest.a /opt/homebrew/opt/or-tools/lib/libortools.9.5.dylib /opt/homebrew/opt/or-tools/lib/libabsl_flags_parse.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_marshalling.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_reflection.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_config.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_private_handle_accessor.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_program_name.a /opt/homebrew/opt/or-tools/lib/libabsl_random_distributions.a /opt/homebrew/opt/or-tools/lib/libabsl_random_seed_sequences.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_pool_urbg.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes_impl.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_slow.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_platform.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_seed_material.a /opt/homebrew/opt/or-tools/lib/libabsl_random_seed_gen_exception.a /opt/homebrew/opt/or-tools/lib/libabsl_raw_hash_set.a /opt/homebrew/opt/or-tools/lib/libabsl_hashtablez_sampler.a /opt/homebrew/opt/or-tools/lib/libabsl_hash.a /opt/homebrew/opt/or-tools/lib/libabsl_city.a /opt/homebrew/opt/or-tools/lib/libabsl_low_level_hash.a /opt/homebrew/opt/or-tools/lib/libabsl_leak_check.a /opt/homebrew/opt/or-tools/lib/libabsl_statusor.a /opt/homebrew/opt/or-tools/lib/libabsl_status.a /opt/homebrew/opt/or-tools/lib/libabsl_cord.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_info.a /opt/homebrew/opt/or-tools/lib/libabsl_cord_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_functions.a /opt/homebrew/opt/or-tools/lib/libabsl_exponential_biased.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_handle.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_optional_access.a /opt/homebrew/opt/or-tools/lib/libabsl_strerror.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_variant_access.a /opt/homebrew/opt/or-tools/lib/libabsl_str_format_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_synchronization.a /opt/homebrew/opt/or-tools/lib/libabsl_stacktrace.a /opt/homebrew/opt/or-tools/lib/libabsl_symbolize.a /opt/homebrew/opt/or-tools/lib/libabsl_debugging_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_demangle_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_graphcycles_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_malloc_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_time.a /opt/homebrew/opt/or-tools/lib/libabsl_strings.a /opt/homebrew/opt/or-tools/lib/libabsl_strings_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_base.a /opt/homebrew/opt/or-tools/lib/libabsl_spinlock_wait.a /opt/homebrew/opt/or-tools/lib/libabsl_throw_delegate.a /opt/homebrew/opt/or-tools/lib/libabsl_int128.a /opt/homebrew/opt/or-tools/lib/libabsl_civil_time.a /opt/homebrew/opt/or-tools/lib/libabsl_time_zone.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_any_cast_impl.a /opt/homebrew/opt/or-tools/lib/libabsl_raw_logging_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_log_severity.a /opt/homebrew/opt/or-tools/lib/libprotob [... cut ...]
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks'
Undefined symbols for architecture arm64:
  "absl::lts_20230125::ParseCommandLine(int, char**)", referenced from:
      InitGoogle(char const*, int*, char***, bool) in libmpl2_lib.a(bus_synthesis.cpp.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/mpl2/test/cpp/mpl2_test] Error 1
make[1]: *** [src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/all] Error 2
make: *** [all] Error 2
```

stefanottili, 2023-06-19:
A June 19 2023 update for anybody trying to use OpenRoad on a MacOS M1 machine.
The built system is very fragile, but I was able to get a working executable.

1) tools/OpenROAD/etc/DependencyInstaller.sh using homebrew updates some things, works.

2) ./build_openroad --latest --local will eventually fail building OpenRoad
It can't be run a 2nd time, you'll have to use --clean-force.
The "Removing MPL2 and PAR" message is gone, so it seems it will build with these.
 
3) cd tools/OpenRoad/build; make
This will eventually fail with warnings about header generated by the wrong version of .. something ... but ...

4) make -j 1  will continue. It will eventually fail, but ....

5) OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad
seems to be a working executable.
At least I can run read_lef/read_def/global_route/detail_route using the ispd19_test6 lef/def.

maliberty, 2023-06-19:
We don't have an M1 machine to test on.  Unless there is a community contribution, I don't see a path forward.

stefanottili, 2023-06-19:
Over the course of the last year OR built just fine --local on my AppleSilicon Mac, it just recently deteriorated.
I understand what without a developer using one of these machines, this is likely to continue to happen.

I'm unfortunately lacking both knowledge about the build_openroad/cmake/OR-code and the time to dive into this.

But I believe that this bug report should at least help the next AppleSilicon MacOS user.
And given some other recent bug reports and the popularity of these machines among students there might be more than just me.

Maybe this can be pointed to in FAQ/README ??
I got a working openroad executable after all.

vvbandeira, 2023-06-21:
@maliberty looks like `or-tools` depends on `protobuf@21`. Maybe there was already a package installed with the `protobuf` dependency, and both got installed to meet. @stefanottili could you run the following and look at the generated file for `protobuf`?
```
brew deps --tree $(brew list) > brew-deps.log
```

stefanottili, 2023-06-22:
I've uninstalled protobuf to be able to make progress, I'm assuming it was a "leftover from another time".

stefanottili, 2023-08-16:
linking is not the issue, mpl2_test and openroad issue runtime errors because libortools.dylib is not placed behind the absl libraries in the link command.

maliberty, 2023-08-16:
Its not a link issue except for the order during linking??  I see no solution to reordering the link line as it is not set by OR.  I think this is a fruitless line of discussion as it is an or-tools issue.

stefanottili, 2024-04-11:
As of April 2024 or-tools should be installed with “brew install or-tools” which installs 9.9
This allows OpenRoad to link and create executables without runtime error.

maliberty, 2024-04-12:
@vvbandeira will look at using or-tools from brew

stefanottili, 2024-04-25:
Fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/4978

closed 2024-04-25 by stefanottili