The-OpenROAD-Project/OpenROAD-flow-scripts #1111: Error While reading the Tech LEF file of tsmc16 and tsmc7

TTTiko, 2023-05-31:
### Subject

[Documentation] for documentation errors.

### Describe the bug

when I read the tech lef file of tsmc16 and tsmc7, the openroad will incur the error report below. 

### Expected Behavior

I would like to know what I can do to avoid this tech lef error, as I am very anxious to test the relevant design.

### Environment

```shell
I am currently working on the latest version
```

### To Reproduce

source "helpers.tcl"
set LIB_DIR "./test_lef"
set tech_lef "$LIB_DIR/ts16ncfslogl16hsl096f_m10f2f1f2f2f0f2_RDL.lef"
read_lef $tech_lef

### Relevant log output

```shell
[INFO ODB-0222] Reading LEF file: ./test_lef/ts16ncfslogl16hsl096f_m10f2f1f2f2f0f2_RDL.lef
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = std::pair<int, int>; _Alloc = std::allocator<std::pair<int, int> >; std::vector<_Tp, _Alloc>::reference = std::pair<int, int>&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Stack trace:
 0# 0x000055F5A1192681 in openroad
 1# 0x00007F8D5D619210 in /lib/x86_64-linux-gnu/libc.so.6
 2# gsignal in /lib/x86_64-linux-gnu/libc.so.6
 3# abort in /lib/x86_64-linux-gnu/libc.so.6
 4# 0x000055F5A1192748 in openroad
 5# lefTechLayerCutSpacingTable::setCutClass(boost::fusion::vector<std::vector<boost::fusion::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, std::allocator<boost::fusion::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > >, std::vector<boost::fusion::vector<boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double>, boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double> >, std::allocator<boost::fusion::vector<boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double>, boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double> > > >, std::vector<boost::fusion::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::vector<boost::fusion::vector<boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double>, boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double> >, std::allocator<boost::fusion::vector<boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double>, boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double> > > > >, std::allocator<boost::fusion::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, boost::optional<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::vector<boost::fusion::vector<boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double>, boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double> >, std::allocator<boost::fusion::vector<boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double>, boost::variant<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double> > > > > > > >&, odb::lefTechLayerCutSpacingTableParser*, odb::lefin*) in openroad
 6# _ZN5boost6detail8function21function_obj_invoker4INS_6spirit2qi6detail13parser_binderINS4_6actionINS4_8sequenceINS_6fusion4consINS4_14literal_stringIRA9_KcLb1EEENSA_INS4_4plusINS8_INSA_INS4_9referenceIKNS4_4ruleIN9__gnu_cxx17__normal_iteratorIPcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEEFSR_vENS_5proto7exprns_4exprINSU_6tagns_3tag8terminalENSU_7argsns_4termINS3_3tag9char_codeINS12_5spaceENS3_13char_encoding5asciiEEEEELl0EEENS3_11unused_typeES1A_EEEENSA_INS4_8optionalINS4_11alternativeINSA_INSB_IRA5_SC_Lb0EEENSA_INSB_IRA4_SC_Lb0EEENS9_4nil_EEEEEEEEES1M_EEEEEEEENSA_INSG_INS8_INSA_INS1F_INSA_INSB_IRA2_SC_Lb0EEENSA_INS4_15any_real_parserIdNS4_13real_policiesIdEEEES1M_EEEEEENSA_IS24_S1M_EEEEEEEENSA_INS4_6kleeneINS8_INSA_IS1D_NSA_IS1Q_NSA_IS28_S1M_EEEEEEEEEES1M_EEEEEEEEEENS_3_bi6bind_tIvPFvRNS9_6vectorIJSt6vectorINS2M_IJSR_NS_8optionalISR_EEEEESaIS2Q_EES2N_INS2M_IJNS_7variantISR_JdEEES2U_EEESaIS2V_EES2N_INS2M_IJSR_S2P_S2X_EEESaIS2Y_EEEEEPN3odb33lefTechLayerCutSpacingTableParserEPNS33_5lefinEENS2K_5list3INS_3argILi1EEENS2K_5valueIS35_EENS3D_IS37_EEEEEEEEN4mpl_5bool_ILb0EEEEEbRSS_RKSS_RNS3_7contextINSA_IRS1A_S1M_EENS2M_IJEEEEERKNS4_10char_classIS17_EEE6invokeERNS1_15function_bufferES3N_S3P_S3V_S3Z_ in openroad
 7# _ZN5boost6detail8function21function_obj_invoker4INS_6spirit2qi6detail13parser_binderINS4_8sequenceINS_6fusion4consINS4_6actionINS4_14literal_stringIRA13_KcLb1EEENS_3_bi6bind_tIvPFvPN3odb33lefTechLayerCutSpacingTableParserEENSG_5list1INSG_5valueISK_EEEEEEEENS9_INS4_8optionalINS7_INS9_INSB_IRA8_SC_Lb1EEENS9_INSA_INS4_15any_real_parserIdNS4_13real_policiesIdEEEENSH_IvPFvdSK_PNSI_5lefinEENSG_5list3INS_3argILi1EEESP_NSO_IS12_EEEEEEEENS8_4nil_EEEEEEEEENS9_INST_INSA_INSB_IRA9_SC_Lb1EEESR_EEEENS9_INST_INS4_11alternativeINS9_INSA_ISW_SR_EENS9_INSA_INSB_IRA10_SC_Lb1EEESR_EENS9_IS1N_S1C_EEEEEEEEEENS9_INST_INS4_9referenceIKNS4_4ruleIN9__gnu_cxx17__normal_iteratorIPcNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEEENS_5proto7exprns_4exprINS29_6tagns_3tag8terminalENS29_7argsns_4termINS3_3tag9char_codeINS2H_5spaceENS3_13char_encoding5asciiEEEEELl0EEENS3_11unused_typeES2P_S2P_EEEEEENS9_IS2T_NS9_IS2T_NS9_IS2T_NS9_IS2T_NS9_IS2T_NS9_IS2T_NS9_IS2T_NS9_IS2S_NS9_INSB_IRA2_SC_Lb1EEES1C_EEEEEEEEEEEEEEEEEEEEEEEEEEEEEEN4mpl_5bool_ILb0EEEEEbRS28_RKS28_RNS3_7contextINS9_IRS2P_S1C_EENS8_6vectorIJEEEEERKNS4_10char_classIS2M_EEE6invokeERNS1_15function_bufferES3G_S3I_S3P_S3T_ in openroad
 8# boost::detail::function::function_obj_invoker4<boost::spirit::qi::detail::parser_binder<boost::spirit::qi::plus<boost::spirit::qi::alternative<boost::fusion::cons<boost::spirit::qi::reference<boost::spirit::qi::rule<__gnu_cxx::__normal_iterator<char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, boost::proto::exprns_::expr<boost::proto::tagns_::tag::terminal, boost::proto::argsns_::term<boost::spirit::tag::char_code<boost::spirit::tag::space, boost::spirit::char_encoding::ascii> >, 0l>, boost::spirit::unused_type, boost::spirit::unused_type, boost::spirit::unused_type> const>, boost::fusion::cons<boost::spirit::qi::reference<boost::spirit::qi::rule<__gnu_cxx::__normal_iterator<char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, boost::proto::exprns_::expr<boost::proto::tagns_::tag::terminal, boost::proto::argsns_::term<boost::spirit::tag::char_code<boost::spirit::tag::space, boost::spirit::char_encoding::ascii> >, 0l>, boost::spirit::unused_type, boost::spirit::unused_type, boost::spirit::unused_type> const>, boost::fusion::nil_> > > >, mpl_::bool_<false> >, bool, __gnu_cxx::__normal_iterator<char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, __gnu_cxx::__normal_iterator<char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, boost::spirit::context<boost::fusion::cons<boost::spirit::unused_type&, boost::fusion::nil_>, boost::fusion::vector<> >&, boost::spirit::qi::char_class<boost::spirit::tag::char_code<boost::spirit::tag::space, boost::spirit::char_encoding::ascii> > const&>::invoke(boost::detail::function::function_buffer&, __gnu_cxx::__normal_iterator<char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >&, __gnu_cxx::__normal_iterator<char*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > const&, boost::spirit::context<boost::fusion::cons<boost::spirit::unused_type&, boost::fusion::nil_>, boost::fusion::vector<> >&, boost::spirit::qi::char_class<boost::spirit::tag::char_code<boost::spirit::tag::space, boost::spirit::char_encoding::ascii> > const&) in openroad
[... cut ...]
16# ord::OpenRoad::readLef(char const*, char const*, bool, bool) in openroad
17# 0x000055F5A11A2C76 in openroad
18# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
19# 0x00007F8D5F303924 in /lib/x86_64-linux-gnu/libtcl8.6.so
20# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
21# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
22# ord::tclAppInit(Tcl_Interp*) in openroad
23# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
24# main in openroad
25# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
```

### Screenshots

[image]

maliberty, 2023-05-31:
We have no access to those LEF files so it is hard to identify the issue.  Can you try cutting down the file to identify what syntax is the issue?  If you can frame the problem in terms of LEF syntax we can try to reproduce it in a non-proprietary way.

TTTiko, 2023-05-31:
> We have no access to those LEF files so it is hard to identify the issue. Can you try cutting down the file to identify what syntax is the issue? If you can frame the problem in terms of LEF syntax we can try to reproduce it in a non-proprietary way.

sure, I will have a try!

TTTiko, 2023-06-08:
> We have no access to those LEF files so it is hard to identify the issue. Can you try cutting down the file to identify what syntax is the issue? If you can frame the problem in terms of LEF syntax we can try to reproduce it in a non-proprietary way.

Here are the LEF error syntax  I found by debug
`LAYER VIA0 
    TYPE CUT ;

    PROPERTY LEF58_CUTCLASS "
              CUTCLASS VSINGLECUT WIDTH 0.032 LENGTH 0.032 CUTS 1 ;
              CUTCLASS VDOUBLECUT WIDTH 0.032 LENGTH 0.080 CUTS 2 ;" ;

    PROPERTY LEF58_SPACINGTABLE "
        SPACINGTABLE  PRL -0.026
        CUTCLASS            VSINGLECUT        VDOUBLECUT
        VSINGLECUT      0.041 0.041         0.041 0.041
        VDOUBLECUT         0.041 0.041        0.041 0.041 ;" ;
END VIA0
`

TTTiko, 2023-06-08:
> We have no access to those LEF files so it is hard to identify the issue. Can you try cutting down the file to identify what syntax is the issue? If you can frame the problem in terms of LEF syntax we can try to reproduce it in a non-proprietary way.

Could you please help me find the reason of the error? Thanks in advance!