The-OpenROAD-Project/OpenROAD-flow-scripts #1123: Unbale type anything in TCL commands section of OpenROAD GUI inside docker

vijayank88, 2023-06-08:
### Subject

[Build] for compilation-related issues.

### Describe the bug

With `openroad>` interactive session able to execute tool commands.
If launch `openroad -gui` and type something in `TCL Commands` section nothing printed.

### Expected Behavior

Enable `TCL Commands` in docker based OpenROAD GUI to execute all commands.

### Environment

```shell
With latest master
```

### To Reproduce

Inside docker, launch `openroad -gui` and type something in `TCL Commands` section

oharboe, 2023-06-08:
How are you launching Docker? We're able to use OpenROAD GUI quite successfully from within docker using this script: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/docker_shell

The only problem we have, is that the GUI can lock up if you resize the window or file requester. This pull request fixes that problem for us https://github.com/The-OpenROAD-Project/OpenROAD/pull/3392

vijayank88, 2023-06-08:
@oharboe 
I launched using this [command](https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithDocker.html#enable-gui-support)
How to run `docker_shell` or from which directory it should be launched?

oharboe, 2023-06-08:
@vijayank88 https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/util/README.md

vijayank88, 2023-06-08:
After the installation, if I use `util/docker_shell make` it end up with below error
```
No protocol specified
[WARNING GUI-0076] QXcbConnection: Could not connect to display :1.0
[ERROR GUI-0077] Could not connect to any X display.
Error: final_report.tcl, 68 GUI-0077
Elapsed time: 0:00.90[h:]min:sec. CPU time: user 0.81 sys 0.08 (99%). Peak memory: 104524KB.
make: *** [logs/nangate45/gcd/base/6_report.log] Error 1
```
If I use `util/docker_shell make gui_final` not launched GUI
```
/OpenROAD-flow-scripts/flow//util/utils.mk:148: warning: ignoring old recipe for target `update_sdc_clocks'
ODB_FILE=./results/nangate45/gcd/base/6_final.odb /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  /OpenROAD-flow-scripts/flow//scripts/gui.tcl
No protocol specified
qt.qpa.screen: QXcbConnection: Could not connect to display :1.0
Could not connect to any X display.
make: *** [gui_6_final.odb] Error 1
```
complete log available here: 
[docker_shell_gui_final.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11685129/docker_shell_gui_final.log)

oharboe, 2023-06-08:
@vijayank88 The script probably needs some adjustments on centos7. I only ever tried it on Ubuntu 22.04.

vijayank88, 2023-06-08:
@oharboe 
Ok. I will check with @vvbandeira 
Can you confirm is your docker based `openroad -gui`, able to get `report_cts` in the scripting section?

oharboe, 2023-06-08:
@vijayank88 Yes: the GUI works flawlessly in GUI and all the flows work just fine. We have used it extensively for 6 months. The only problem is the GUI crash, with the fix here: https://github.com/The-OpenROAD-Project/OpenROAD/pull/3392

The docker_shell script probably needs tweaking so that it works on other distributions than Ubuntu 22.04.

vijayank88, 2023-06-08:
Launching docker with following commands blocking the `TCL Commands` usage.
```
docker run --rm -it \
           -u $(id -u ${USER}):$(id -g ${USER}) \
           -v $(pwd)/flow:/OpenROAD-flow-scripts/flow \
           -e DISPLAY=${DISPLAY} \
           -v /tmp/.X11-unix:/tmp/.X11-unix \
           -v ${HOME}/.Xauthority:/.Xauthority \
           --network host \
           --security-opt seccomp=unconfined \
           openroad/flow-$OS_NAME-builder
```

vijayank88, 2023-06-09:
> Launching docker with following commands blocking the `TCL Commands` usage.
>
> ```
> docker run --rm -it \
>            -u $(id -u ${USER}):$(id -g ${USER}) \
>            -v [...]

@vvbandeira 
Can you please test this with OS: Centos 07 and docker based installation?

vvbandeira, 2023-06-12:
The same issue as https://github.com/The-OpenROAD-Project/OpenROAD/issues/1834, will need to investigate further to see why only a few users have this issue, but the workaround is to run inside the docker image:

```
yum install -y xorg-x11-server-Xorg
```

mczyz-antmicro, 2023-07-07:
@vijayank88 Error is:

    xkbcommon: ERROR: failed to add default include path /usr/share/X11/xkb

so have you tried adding this option  to the `docker run` command? It solved the issue for me

    -v /usr/share/X11/xkb:/usr/share/X11/xkb

vijayank88, 2023-07-08:
In Ubuntu20.04 its not still
```
docker run --rm -it            -u $(id -u ${USER}):$(id -g ${USER})            -v $(pwd)/flow:/OpenROAD-flow-scripts/flow            -e DISPLAY=${DISPLAY}            -v /tmp/.X11-unix:/tmp/.X11-unix -v /usr/share/X11/xkb:/usr/share/X11/xkb           -v ${HOME}/.Xauthority:/.Xauthority            --network host            --security-opt seccomp=unconfined            openroad/flow-centos7-builder
bash-4.2$ source env.sh 
OPENROAD: /OpenROAD-flow-scripts/tools/OpenROAD
bash-4.2$ openroad -gui
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
process 11: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.
Most likely, the application was supposed to call dbus_connection_close(), since this is a private connection.
  D-Bus not built with -rdynamic so unable to print a backtrace
Signal 6 received
Stack trace:
 0# 0x0000000000D4AFD7 in openroad
 1# 0x00007F58B9F49400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x00007F58AFC22655 in /lib64/libdbus-1.so.3
 5# _dbus_warn_check_failed in /lib64/libdbus-1.so.3
 6# 0x00007F58B231CBFB in /lib64/libQt5DBus.so.5
 7# QObject::event(QEvent*) in /lib64/libQt5Core.so.5
 8# 0x00007F58BB279A55 in /lib64/libQt5Core.so.5
 9# QCoreApplication::notifyInternal2(QObject*, QEvent*) in /lib64/libQt5Core.so.5
10# QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) in /lib64/libQt5Core.so.5
11# 0x00007F58BB2C8A03 in /lib64/libQt5Core.so.5
12# g_main_context_dispatch in /lib64/libglib-2.0.so.0
13# 0x00007F58B6EC5478 in /lib64/libglib-2.0.so.0
14# g_main_context_iteration in /lib64/libglib-2.0.so.0
15# QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
16# QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) in /lib64/libQt5Core.so.5
17# QThread::exec() in /lib64/libQt5Core.so.5
18# 0x00007F58B231A3B5 in /lib64/libQt5DBus.so.5
19# 0x00007F58BB0CFE71 in /lib64/libQt5Core.so.5
20# 0x00007F58BE9CEEA5 in /lib64/libpthread.so.0
21# clone in /lib64/libc.so.6
Aborted (core dumped)
bash-4.2$ exit
exit
```

vijayank88, 2023-08-16:
@vvbandeira Any update on this issue with Centos7 docker build?

lpawelcz, 2024-02-28:
I've encountered very similar error as in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1123#issuecomment-1582025520:
```
$ ./util/docker_shell make gui_cts 
...
ODB_FILE=./results/nangate45/gcd/base/4_cts.odb /OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui  /OpenROAD-flow-scripts/flow//scripts/gui.tcl
qt.qpa.screen: QXcbConnection: Could not connect to display :0
Could not connect to any X display.
make: *** [gui_4_cts.odb] Error 1
```

For me, adding `--network host` to docker call in `docker_shell` fixed the issue. Here is a PR with that fix: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1818

vijayank88, 2024-07-03:
As docker moved to Ubuntu environment, this issue no more valid.

closed 2024-07-03 by vijayank88