The-OpenROAD-Project/OpenROAD-flow-scripts #1134: Long running times for some stages

oharboe, 2023-06-12:
### Description

@louiic FYI

megaboom https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1126 has long running times.

Can some of these stages be sped up?

```
make DESIGN_CONFIG=designs/asap7/megaboom/config.mk elapsed
```

```
Log                       Elapsed seconds
1_1_yosys                       5394
2_1_floorplan                   1850
2_2_floorplan_io                  12
2_3_tdms_place                  4428
2_4_mplace                       210
2_5_tapcell                       11
2_6_pdn                          335
3_1_place_gp_skip_io            2563
3_2_place_iop                     14
3_3_place_gp                   66276
3_4_resizer                     5788
3_5_opendp                      5049
4_1_cts                        57232
4_2_cts_fillcell                  61
No elapsed time found in logs/asap7/megaboom/base/5_1_fastroute.log
```

oharboe, 2023-06-12:
I don't think the synthesis of 2 hours isn't too bad.

However for a 590um^2 design, the following running times strike me as surprising given the speed at which I've seen 4mm^2 designs run: 2_1_floorplan, 2_3_tdms_place, 3_4_resizer, 4_1_cts.

louiic, 2023-06-12:
> I think if we pre place SRAM, it will go faster.
>
> Macro placement isn't taking that long. Do you mean replace registers with an SRAM?

Matt,

Yes … I do mean macro = SRAM.

They are in the netlist now.

Louii

tspyrou, 2023-06-22:
@louiic The sram work to reduce the netlist size is the highest priority that @oharboe has for you right now.

oharboe, 2023-07-05:
Covered by a few specific tasks(create fakeram for register file) for megaboom in https://github.com/Pinata-Consulting/pii-ascenium/issues/11

If that isn't enough, then we have to reconsider.

closed 2023-07-05 by oharboe