The-OpenROAD-Project/OpenROAD-flow-scripts #1135: ERROR: Duplicate MACRO name:

gkamendje, 2023-06-12:
### Subject

[Documentation] for documentation errors.

### Describe the bug

When I try to run 
`make results/myplatform/mydesign/base/6_1_merged.gds` I get the following error
`ERROR: Duplicate MACRO name: myCellName (inside MACRO) (line=15, cell=, file=mylef.lef) in Layout.read`

### Expected Behavior

The merged GDS file should have been created. 

### Environment

```shell
Here is the output of `openroad -version v2.0-8513-gca2e30f46`

[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.17.3
-- The CXX compiler identification is GNU 10.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8513-gca2e30f46
-- System name: Linux
-- Compiler: GNU 10.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/libgtest.a
-- The C compiler identification is GNU 10.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-c++11-narrowing - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register
-- Performing Test C_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-register - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format
-- PerfoCMake Error at src/mpl2/CMakeLists.txt:42 (find_package):
  By not providing "Findortools.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ortools", but
  CMake did not find one.

  Could not find a package configuration file provided by "ortools" with any
  of the following names:

    ortoolsConfig.cmake
    ortools-config.cmake

  Add the installation prefix of "ortools" to CMAKE_PREFIX_PATH or set
  "ortools_DIR" to a directory containing one of the above files.  If
  "ortools" provides a separate development package or SDK, be sure it has
  been installed.

rming Test CXX_COMPILER_SUPPORTS__-Wno-format - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test C_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-reserved-user-defined-literal - Failed
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive
-- Performing Test C_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive
-- Performing Test CXX_COMPILER_SUPPORTS__-fpermissive - Success
-- Performing Test C_COMPILER_SUPPORTS__-x
-- Performing Test C_COMPILER_SUPPORTS__-x - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__-x
-- Performing Test CXX_COMPILER_SUPPORTS__-x - Failed
-- Performing Test C_COMPILER_SUPPORTS__c++
-- Performing Test C_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test CXX_COMPILER_SUPPORTS__c++
-- Performing Test CXX_COMPILER_SUPPORTS__c++ - Failed
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-but-set-variable - Success
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- TCL readline library: /usr/lib/x86_64-linux-gnu/libtclreadline.so
-- TCL readline header: /usr/include/x86_64-linux-gnu
-- Found SWIG: /usr/bin/swig4.0 (found suitable version "4.0.1", minimum required is "3.0")
-- Found Boost: /usr/local/lib/cmake/Boost-1.78.0/BoostConfig.cmake (found version "1.78.0")
-- boost: 1.78.0
-- Found Python3: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found version "3.8.10") found components: Development
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11")
-- Found Threads: TRUE
-- spdlog: 1.9.2
-- Found BISON: /usr/bin/bison (found version "3.5.1")
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen missing components: dot
-- STA version: 2.4.0
-- STA git sha: 555493cba6e476bd2ff0b9a543de7a781276c2b3
-- System name: Linux
-- Compiler: GNU 10.3.0
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /usr/local
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- SSTA: 0
-- STA executable: /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Configuring incomplete, errors occurred!
```
```

### To Reproduce

I do see the issue only with the platform I am setting up.  Public platforms Sky130* and nand45 all work fine.   
Unfortunately due to NDA reasons I cannot upload any .def or .lef file here. 

### Additional Context

I do see that a similar issue was reported in #942 and #944. However, I am not sure I fully understand what the fixes for those issues where and how to apply it to my case.

gkamendje, 2023-06-12:
Here is how  Klayout gets called
```
[INFO][FLOW] Using platform directory ./platforms/myplatform
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' stdbuf -o L /usr/bin/klayout -zz -rd design_name=my_design_top \
        -rd in_def=results/myplatform/mydesign/base/6_final.def \
        -rd in_files="./platforms/myplatform/gds/mygds.gds  " \
        -rd config_file=./platforms/myplatform/fill.json \
        -rd seal_file="" \
        -rd out_file=results/myplatform/mydesign/base/6_1_merged.gds \
        -rd tech_file=./objects/myplatform/mydesign/base/klayout.lyt \
        -rd layer_map= \
        -r ./util/def2stream.py) 2>&1 | tee ./logs/myplatform/mydesign/base/6_1_merge.log

```
Unfortunately there is not much in the log file (6_1_merge.log) and I am assuming here that KLayout does not generate a separate log file.  
```
Warning: Reading layer map file myStreamout018.map, line 31: NAME record ignored for purpose: myName
Warning: Reading layer map file myStreamout018.map, line 32: NAME record ignored for purpose: myName
Warning: Reading layer map file myStreamout018.map, line 33: NAME record ignored for purpose: myName
Warning: Reading layer map file myStreamout018.map, line 34: NAME record ignored for purpose: myName
Warning: Reading layer map file myStreamout018.map, line 35: NAME record ignored for purpose: myName
Warning: Reading layer map file myStreamout018.map, line 36: NAME record ignored for purpose: myName
[INFO] Reporting cells prior to loading DEF ...
[INFO] Reading DEF ...
ERROR: Duplicate MACRO name: MyCell (inside MACRO) (line=15, cell=, file=MyLef.lef) in Layout.read
  ./util/def2stream.py:113
Command exited with non-zero status 1
Elapsed time: 0:01.55[h:]min:sec. CPU time: user 1.38 sys 0.12 (96%). Peak memory: 394212KB.
```

Could it be that the following definition of  `MyCell` in `Mylef.lef` is an issue? (this works fine with other EDA tools so I doubt this could be a problem)
```
MACRO MyCell
  CLASS CORE ;
  ORIGIN 0 0 ;
  FOREIGN MyCell 0 0 ;
  SIZE 15.12 BY 4.48 ;
```

maliberty, 2023-06-12:
If your top cell name MyCell or is that something you are instantiating in top?

maliberty, 2023-06-12:
Perhaps you are getting both the LEF and the GDS for the same cell and that is causing the problem?

gkamendje, 2023-06-12:
How is this supposed to work? I can see that the GDS file is passed as an argument to Klayout whereas, def2stream.py reads the LEF file. Is this the expected behavior?

maliberty, 2023-06-12:
Looking more closely I see it reads the LEF, then reads the GDS, and substitutes it.  Could you be reading two LEFs with MyCell (or the same LEF twice)?

gkamendje, 2023-06-13:
I figured out that the entry  `<lef-files> </lef-files>` was duplicated in the  file `klayout.lyt` which is passed as tech file to KLayout 
`-rd tech_file=./objects/myplatform/mydesign/base/klayout.lyt`
Removing one of the entries fixed the issue.

closed 2023-06-13 by gkamendje