The-OpenROAD-Project/OpenROAD-flow-scripts #1137: dpicall.cc:25:10: fatal error: ffi.h: No such file or directory

kotak31, 2023-06-13:
### Subject

[Build] for compilation-related issues.

### Describe the bug

frontends/ast/dpicall.cc:25:10: fatal error: ffi.h: No such file or directory
   25 | #include <ffi.h>
      |          ^~~~~~~
compilation terminated.
make: *** [Makefile:704: frontends/ast/dpicall.o] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory '/home/kotakosa/Desktop/ResearchVLSI/OpenROAD-flow-scripts/tools/yosys'

### Expected Behavior

Im trying to install OpenRoad Project local installation scripts as per this https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildLocally.md

I've entered the following command in my terminal -  ./build_openroad.sh --local
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11739854/build_openroad.log)

It resulted in error as follows.
frontends/ast/dpicall.cc:25:10: fatal error: ffi.h: No such file or directory
   25 | #include <ffi.h>
      |          ^~~~~~~
compilation terminated.
make: *** [Makefile:704: frontends/ast/dpicall.o] Error 1
make: *** Waiting for unfinished jobs....
make: Leaving directory '/home/kotakosa/Desktop/ResearchVLSI/OpenROAD-flow-scripts/tools/yosys'

### Environment

```shell
kotakosa@vemuri-test:~/Desktop/ResearchVLSI/OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.4.0-150-generic
os: Linux Mint 20 (Ulyana)
./etc/Env.sh: line 57: cmake: command not found
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh

./build_openroad.sh --local
[build_openroad.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11739876/build_openroad.log)

vijayank88, 2023-06-14:
@kotak31 
Is `sudo ./setup.sh` completed without any errors? 
Can you share tail of log as well for `setup.sh` run?

vvbandeira, 2023-06-20:
@kotak31, did you had a chance to try the suggestion above? Are you still having the issue?

rovinski, 2023-06-23:
Just an FYI - the error means that the `libffi` package is either not installed on your system or is not found during compilation. The setup script mentioned above should fix that, or you can install the package on your own.

gustad, 2024-07-14:
Depending upon the distro you might need `libffi-devel`.