The-OpenROAD-Project/OpenROAD-flow-scripts #1164: Unable to build

gudeh, 2023-06-21:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I am using Ubuntu 22.04 and build is failing. Trying from a new clone of https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts

### Expected Behavior

Build successfully

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-45-generic
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8780-ga7f84d55d
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")  
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
[... 190 lines cut ...]
[ 93%] Building CXX object src/dft/CMakeFiles/dft.dir/CMakeFiles/dft.dir/dftTCL_wrap.cxx.o
[ 93%] Building CXX object src/dft/CMakeFiles/dft.dir/dft-tclInitVar.cc.o
[ 93%] Building CXX object src/dft/CMakeFiles/dft.dir/src/Dft.cpp.o
[ 93%] Building CXX object src/dft/CMakeFiles/dft.dir/src/MakeDft.cpp.o
[ 93%] Linking CXX static library dft.a
[ 93%] Built target dft
[ 95%] Swig compile src/DftGen-py.i for python
[ 95%] Generating dft_py-pythonInitVar.cc
Consolidate compiler generated dependencies of target dft_py
[ 95%] Building CXX object src/dft/CMakeFiles/dft_py.dir/CMakeFiles/dft_py.dir/DftGen-pyPYTHON_wrap.cxx.o
[ 95%] Building CXX object src/dft/CMakeFiles/dft_py.dir/dft_py-pythonInitVar.cc.o
[ 95%] Linking CXX static library _dft_py.a
[ 95%] Built target dft_py
[ 95%] Building CXX object src/CMakeFiles/openroad.dir/Design.cc.o
[ 95%] Building CXX object src/CMakeFiles/openroad.dir/Tech.cc.o
[ 95%] Building CXX object src/CMakeFiles/openroad.dir/OpenRoad.cc.o
[ 95%] Building CXX object src/CMakeFiles/openroad.dir/Main.cc.o
[ 95%] Linking CXX executable openroad
/usr/bin/ld: mpl2/libmpl2_lib.a(bus_synthesis.cpp.o): in function `InitGoogle(char const*, int*, char***, bool)':
bus_synthesis.cpp:(.text+0x18df): undefined reference to `absl::lts_20220623::ParseCommandLine(int, char**)'
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/CMakeFiles/openroad.dir/build.make:425: src/openroad] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1741: src/CMakeFiles/openroad.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

vijayank88, 2023-06-21:
If you're a beginner can you try docker based installation? https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithDocker.md 

Try the following again if you still want to proceed with local installation.
```
git checkout master
git pull
./build_openroad.sh --local --clean -threads=2
```

stefanottili, 2023-06-21:
I see the same build failure with --local on Apple Silicon MacOS.
Good news is that by the time OR 'make' reaches 98% there is a working OR executable at
OpenROAD-flow-scripts/tools/OpenROAD/build/src/openroad

Fixing this build failure for Ubuntu will fix the MacOS build too.

```
build_openroad.sh --local --latest --clean-force

[ 81%] Linking CXX executable mpl2_test
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks'
Undefined symbols for architecture arm64:
  "absl::lts_20230125::ParseCommandLine(int, char**)", referenced from:
      InitGoogle(char const*, int*, char***, bool) in libmpl2_lib.a(bus_synthesis.cpp.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/build.make:270: src/mpl2/test/cpp/mpl2_test] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:4665: src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....

cd tools/OpenROAD/build
make -j 8 (fails eventually)
make (this continues)

[ 98%] Linking CXX executable mpl2_test
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks'
Undefined symbols for architecture arm64:
  "absl::lts_20230125::ParseCommandLine(int, char**)", referenced from:
      InitGoogle(char const*, int*, char***, bool) in libmpl2_lib.a(bus_synthesis.cpp.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/mpl2/test/cpp/mpl2_test] Error 1
make[1]: *** [src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/all] Error 2
make: *** [all] Error 2
```

maliberty, 2023-06-21:
There are some unnecessary includes in bus_synthesis.cpp.  Does this change help?

```
-#include <absl/flags/flag.h>
-#include <absl/strings/match.h>
-#include <absl/strings/string_view.h>
-#include <ortools/base/commandlineflags.h>
-#include <ortools/base/init_google.h>
-#include <ortools/base/logging.h>
 #include <ortools/linear_solver/linear_solver.h>
-#include <ortools/linear_solver/linear_solver.pb.h>
```

stefanottili, 2023-06-21:
```
[ 96%] Building CXX object src/mpl2/CMakeFiles/mpl2_lib.dir/src/bus_synthesis.cpp.o
[ 96%] Linking CXX static library libmpl2_lib.a
[ 98%] Built target mpl2_lib
[ 98%] Built target mpl2
[ 98%] Linking CXX executable mpl2_test
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks'
ERROR: Something is wrong with flag 'flagfile' in file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc'. One possibility: file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc' is being linked both statically and dynamically into this executable. e.g. some files listed as srcs to a test and also listed as srcs of some shared lib deps of the same test.
CMake Error at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:112 (message):
  Error running test executable.

    Path: '/Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp/mpl2_test'
    Result: 1
    Output:
      

Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:225 (gtest_discover_tests_impl)
```

stefanottili, 2023-06-21:
that works, yay. But a following 'make install' will fail.
```
[ 98%] Built target dft_config_lib
[ 98%] Built target dft_clock_domain_lib
[ 98%] Built target dft_cells_lib
[ 98%] Built target dft_architect_lib
[ 98%] Built target dft_replace_lib
[ 98%] Built target dft_stitch_lib
[ 98%] Built target dft
[100%] Built target dft_py
[100%] Built target openroad
```
```
make install
[ 51%] Built target libabc
[ 51%] Built target abc
[ 51%] Built target utl_lib
...
[ 98%] Linking CXX executable mpl2_test
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks'
ERROR: Something is wrong with flag 'flagfile' in file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc'. One possibility: file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc' is being linked both statically and dynamically into this executable. e.g. some files listed as srcs to a test and also listed as srcs of some shared lib deps of the same test.
CMake Error at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:112 (message):
  Error running test executable.

    Path: '/Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp/mpl2_test'
    Result: 1
    Output:
      

Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:225 (gtest_discover_tests_impl)
```

maliberty, 2023-06-21:
My guess is that google test is the problem then.  @quantamhd any thoughts?

QuantamHD, 2023-06-21:
This looks like this is the mpl2 related or tools dep.

maliberty, 2023-06-21:
@QuantamHD if the OR app can build it doesn't seem related to those but is specific to the unit test.  The only real delta there is google test.

maliberty, 2023-06-21:
My first guess is that google test is using shared libraries somehow

stefanottili, 2023-06-21:
```
ERROR: Something is wrong with flag 'flagfile' in file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc'.
One possibility: file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc' 
is being linked both statically and dynamically into this executable.
e.g. some files listed as srcs to a test and also listed as srcs of some shared lib deps of the same test.
```

vvbandeira, 2023-06-22:
> ```
> ERROR: Something is wrong with flag 'flagfile' in file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc'.
> One possibility: file [...]

This is something we have seen before. It comes and goes, especially with Apple silicon macOS users.

maliberty, 2023-06-22:
Could you give the actual compile command issued for mpl2_test?

stefanottili, 2023-06-23:
I've sent "make VERBOSE=1" directly to Matt, please let me know if and how I could help debug this issue.

vvbandeira, 2023-06-23:
@stefanottili could you also send it here? if it is too large you can put in something like https://pastebin.com/

stefanottili, 2023-06-23:
```
make VERBOSE=1
...
[ 98%] Linking CXX executable mpl2_test
cd /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E cmake_link_script CMakeFiles/mpl2_test.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/mpl2_test.dir/mpl2_test.cc.o -o mpl2_test -F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks  -Wl,-rpath,/opt/homebrew/opt/or-tools/lib -Wl,-rpath,/opt/homebrew/lib ../../../../lib/libgtest.a ../../../../lib/libgtest_main.a ../../libmpl2_lib.a ../../../../lib/libgtest.a /opt/homebrew/opt/or-tools/lib/libortools.9.5.dylib /opt/homebrew/opt/or-tools/lib/libabsl_flags_parse.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_marshalling.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_reflection.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_config.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_private_handle_accessor.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_program_name.a /opt/homebrew/opt/or-tools/lib/libabsl_random_distributions.a /opt/homebrew/opt/or-tools/lib/libabsl_random_seed_sequences.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_pool_urbg.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes_impl.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_slow.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_platform.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_seed_material.a /opt/homebrew/opt/or-tools/lib/libabsl_random_seed_gen_exception.a /opt/homebrew/opt/or-tools/lib/libabsl_raw_hash_set.a /opt/homebrew/opt/or-tools/lib/libabsl_hashtablez_sampler.a /opt/homebrew/opt/or-tools/lib/libabsl_hash.a /opt/homebrew/opt/or-tools/lib/libabsl_city.a /opt/homebrew/opt/or-tools/lib/libabsl_low_level_hash.a /opt/homebrew/opt/or-tools/lib/libabsl_leak_check.a /opt/homebrew/opt/or-tools/lib/libabsl_statusor.a /opt/homebrew/opt/or-tools/lib/libabsl_status.a /opt/homebrew/opt/or-tools/lib/libabsl_cord.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_info.a /opt/homebrew/opt/or-tools/lib/libabsl_cord_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_functions.a /opt/homebrew/opt/or-tools/lib/libabsl_exponential_biased.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_handle.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_optional_access.a /opt/homebrew/opt/or-tools/lib/libabsl_strerror.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_variant_access.a /opt/homebrew/opt/or-tools/lib/libabsl_str_format_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_synchronization.a /opt/homebrew/opt/or-tools/lib/libabsl_stacktrace.a /opt/homebrew/opt/or-tools/lib/libabsl_symbolize.a /opt/homebrew/opt/or-tools/lib/libabsl_debugging_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_demangle_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_graphcycles_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_malloc_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_time.a /opt/homebrew/opt/or-tools/lib/libabsl_strings.a /opt/homebrew/opt/or-tools/lib/libabsl_strings_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_base.a /opt/homebrew/opt/or-tools/lib/libabsl_spinlock_wait.a /opt/homebrew/opt/or-tools/lib/libabsl_throw_delegate.a /opt/homebrew/opt/or-tools/lib/libabsl_int128.a /opt/homebrew/opt/or-tools/lib/libabsl_civil_time.a /opt/homebrew/opt/or-tools/lib/libabsl_time_zone.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_any_cast_impl.a /opt/homebrew/opt/or-tools/lib/libabsl_raw_logging_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_log_severity.a /opt/homebrew/opt/or-tools/lib/libprotobuf.a /opt/homebrew/opt/or-tools/lib/libre2.a /opt/homebrew/opt/or-tools/lib/libCbcSolver.a /opt/homebrew/opt/or-tools/lib/libOsiCbc.a /opt/homebrew/opt/or-tools/lib/libCbc.a /opt/homebrew/opt/or-tools/lib/libCgl.a /opt/homebrew/opt/or-tools/lib/libClpSolver.a /opt/homebrew/opt/or-tools/lib/libOsiClp.a /opt/homebrew/opt/or-tools/lib/libClp.a /opt/homebrew/opt/or-tools/lib/libOsi.a /opt/homebrew/opt/or-tools/lib/libCoinUtils.a /opt/homebrew/opt/or-tools/lib/libscip.a -ldl ../../../par/libpar_lib.a ../../../dbSta/src/libdbSta_lib.a /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/libOpenSTA.a /opt/homebrew/opt/or-tools/lib/libz.a ../../../odb/src/cdl/libcdl.a ../../../odb/src/defin/libdefin.a ../../../odb/src/def/libdef.a ../../../odb/src/def/libdefzlib.a ../../../odb/src/lefin/liblefin.a ../../../odb/src/lef/liblef.a ../../../odb/src/lef/liblefzlib.a ../../../odb/src/db/libdb.a ../../../odb/src/defout/libdefout.a ../../../odb/src/lefout/liblefout.a ../../../odb/src/zutil/libzutil.a ../../../odb/src/db/libdb.a ../../../odb/src/defout/libdefout.a ../../../odb/src/lefout/liblefout.a ../../../odb/src/zutil/libzutil.a /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib ../../../odb/src/tm/libtm.a ../../../utl/libutl_lib.a /opt/homebrew/lib/li [... cut ...] to a test and also listed as srcs of some shared lib deps of the same test.
CMake Error at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:112 (message):
  Error running test executable.

    Path: '/Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp/mpl2_test'
    Result: 1
    Output:
      
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:225 (gtest_discover_tests_impl)
```

vvbandeira, 2023-06-23:
@stefanottili, I was able to get it to pass this error by updating the `or-tools` to the latest version. You can do this in your machine with the following (patch file attached). I still get some compiler warnings and some regression test fail, but is a step forward. I will work on a PR to update the version.

```
rm -rf $(brew --prefix or-tools)
git apply or-tools-update.patch
./etc/DependencyInstaller.sh
```

[or-tools-update.patch](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11850973/or-tools-update.patch)

stefanottili, 2023-06-23:
a) ./build_openroad.sh --local --latest --clean-force 

unfortunately still fails for me. See "make VERBOSE=1" output at e) below
```
2023-06-23 13:02:05 (8.06 MB/s) - ‘or-tools_arm64_macOS-13.2.1_cpp_v9.6.2534.tar.gz’ saved [32492889/32492889]

grep VERSION /opt/homebrew/opt/or-tools/Makefile
VERSION := 9.6.2534
```
```
[ 98%] Linking CXX executable mpl2_test
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks'
ERROR: Something is wrong with flag 'flagfile' in file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc'. One possibility: file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc' is being linked both statically and dynamically into this executable. e.g. some files listed as srcs to a test and also listed as srcs of some shared lib deps of the same test.
CMake Error at /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:112 (message):
  Error running test executable.
```
b) instead of DependcyInstaller trying to install a x86 klayout dmg:

I'd rather compile klayout natively for M1 using homebrew.

I also get '“klayout” cannot be opened because the developer cannot be verified.' 
Usually one can use "System Settings->Privacy&Security" to set it as "good", but I didn't see that option here.
```
QMAKE_RESOURCE_FLAGS += -threshold 1000000. // not sure whether it's still necessary
build4mac.py -r HB31 -p HBAuto -q Qt5Brew -m '--jobs=8' -n -u
```
c) Are these "removing" features or-tools related ?
```
-- Removing MPL2 and PAR to avoid run time fatal error.
-- Removing PAR_PY to avoid run time fatal error.
```
I'd like to be able to give mpl2 a try after all.

d) the compile throws some warnings

Are these to be expected or Apple Clang specific and worth reporting ?
```
class Hypergraph
OpenROAD/src/par/src/Hypergraph.h:58:1: note: did you mean class here?
struct Hypergraph;

/opt/homebrew/Cellar/lemon-graph/1.3.1/include/lemon/connectivity.h:268:36: warning: unused typedef 'Node' [-Wunused-local-typedef]
  typedef typename Digraph::Node Node;

OpenROAD/third-party/abc/src/bdd/dsd/dsdProc.c:1572:15: warning: variable 'bRes' is used uninitialized whenever 'if' condition is false [-Wsometimes-uninitialized]
    else if ( pR->Type == DSD_NODE_PRIME )

OpenROAD/src/mpl2/src/hier_rtlmp.cpp:239:9: warning: variable 'tot_num_layer' set but not used [-Wunused-but-set-variable]
    int tot_num_layer = 0;
```

e) make VERBOSE=1
```
[100%] Linking CXX executable mpl2_test
cd /Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E cmake_link_script CMakeFiles/mpl2_test.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/mpl2_test.dir/mpl2_test.cc.o -o mpl2_test -F/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks  -Wl,-rpath,/opt/homebrew/opt/or-tools/lib -Wl,-rpath,/opt/homebrew/lib ../../../../lib/libgtest.a ../../../../lib/libgtest_main.a ../../libmpl2_lib.a ../../../../lib/libgtest.a /opt/homebrew/opt/or-tools/lib/libortools.9.6.dylib /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_check_op.a /opt/homebrew/opt/or-tools/lib/libabsl_die_if_null.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_parse.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_conditions.a /opt/homebrew/opt/or-tools/lib/libabsl_log_flags.a /opt/homebrew/opt/or-tools/lib/libabsl_flags.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_marshalling.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_reflection.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_config.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_program_name.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_private_handle_accessor.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_log_initialize.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_message.a /opt/homebrew/opt/or-tools/lib/libabsl_examine_stack.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_format.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_proto.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_nullguard.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_log_sink_set.a /opt/homebrew/opt/or-tools/lib/libabsl_log_globals.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_globals.a /opt/homebrew/opt/or-tools/lib/libabsl_log_sink.a /opt/homebrew/opt/or-tools/lib/libabsl_log_entry.a /opt/homebrew/opt/or-tools/lib/libabsl_random_distributions.a /opt/homebrew/opt/or-tools/lib/libabsl_random_seed_sequences.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_pool_urbg.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes_impl.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_slow.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_platform.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_seed_materi [... cut ...]
  Error running test executable.

    Path: '/Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/build/src/mpl2/test/cpp/mpl2_test'
    Result: 1
    Output:
      

Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.26.4/share/cmake/Modules/GoogleTestAddTests.cmake:225 (gtest_discover_tests_impl)

make[2]: *** [src/mpl2/test/cpp/mpl2_test] Error 1
make[2]: *** Deleting file `src/mpl2/test/cpp/mpl2_test'
make[1]: *** [src/mpl2/test/cpp/CMakeFiles/mpl2_test.dir/all] Error 2
```

stefanottili, 2023-06-23:
https://github.com/google/or-tools/issues/3709

gudeh, 2023-06-23:
> @stefanottili, I was able to get it to pass this error by updating the `or-tools` to the latest version. You can do this in your machine with the following (patch file attached). I still get some [...]

These commands solved the issue for me. I guess applying the first two lines to DependencyInstaller. I remember having a similar error and updating or-tools to solve it.

maliberty, 2023-06-24:
@vvbandeira would you see if we can apply this patch across all platforms

stefanottili, 2023-06-24:
Getting the latest or-tools is NOT fixing the issue, as https://github.com/google/or-tools/issues/3709 also states.

grep VERSION /opt/homebrew/opt/or-tools/Makefile
VERSION := 9.6.2534

stefanottili, 2023-07-17:
I stand corrected, applying the patch did fix the build issue for M1, yay.

Please apply the patch to etc/DependencyInstaller.sh so that this issue can be closed.

stefanottili, 2023-07-17:
The follow up question to being able to build with or-tools is whether mpl2 can be enabled for M1.

Currently I'm getting:
-- Removing MPL2 and PAR to avoid run time fatal error.
-- Removing PAR_PY to avoid run time fatal error.

How would I enable mpl2 to test whether the build succeeds ?

maliberty, 2023-07-17:
See https://github.com/The-OpenROAD-Project/OpenROAD/blob/19dccc498d617ec8dca414f757d2a15488c18aa4/src/CMakeLists.txt#L339

stefanottili, 2023-07-17:
commenting out the removal of MPL2/PAR in src/CMakeLists.txt
```
339 #if (NOT ${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
340   # mpl2 aborts with link error on darwin and par causes abseil link error at startup on apple silicon so do not link it.
341   target_link_libraries(openroad mpl2 par)
342   target_compile_definitions(openroad PRIVATE ENABLE_MPL2)
343   target_compile_definitions(openroad PRIVATE ENABLE_PAR)
344 #else()
345 #  message(STATUS "Removing MPL2 and PAR to avoid run time fatal error.")
346 #endif()
```
and a fresh build with --clean-force fails on mpl2_test again, but it's building par_lib, mpl2_lib and mpl

cd tools/OpenRoad/build
make openroad -j 8
```
[ 84%] Built target mpl
[ 84%] Built target mpl_py
[ 86%] Built target par_lib
[ 87%] Built target mpl2_lib
[ 87%] Built target mpl2
```
but then linking fails with 
```
[ 100%] Linking CXX executable openroad
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks'
Undefined symbols for architecture arm64:
  "_PyInit__par_py", referenced from:
      _main in Main.cc.o
  "sta::par_py_python_inits", referenced from:
      _main in Main.cc.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

maliberty, 2023-07-17:
Do you see par_py_python_inits in build/src/par/par_py-pythonInitVar.cc ?

If so, how about:
```
nm -g build/src/par/CMakeFiles/par_py.dir/par_py-pythonInitVar.cc.o|grep par_py_python_inits
0000000000000000 D _ZN3sta19par_py_python_initsE
```

stefanottili, 2023-07-17:
yes, and yes
```
  1 // TCL init file encoded by TclEncode.tcl
  2 namespace sta {
  3 const char *par_py_python_inits[] = {
  4 "035032084104105115032102 ...
 27 0
 28 };
 29 } // namespace
```
```
nm -g build/src/par/CMakeFiles/par_py.dir/par_py-pythonInitVar.cc.o|grep par_py_python_inits
000000000000ae90 D __ZN3sta19par_py_python_initsE
```

stefanottili, 2023-07-17:
the failing link command is:
```
cd /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build/src && /opt/homebrew/Cellar/cmake/3.26.4/bin/cmake -E cmake_link_script CMakeFiles/openroad.dir/link.txt --verbose=1
/Library/Developer/CommandLineTools/usr/bin/c++ -O3 -DNDEBUG -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.3.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names CMakeFiles/openroad.dir/Design.cc.o CMakeFiles/openroad.dir/Tech.cc.o CMakeFiles/openroad.dir/OpenRoad.cc.o CMakeFiles/openroad.dir/Main.cc.o -o openroad -F/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks -F/opt/homebrew/opt/qt@5/lib  -Wl,-rpath,/opt/homebrew/opt/or-tools/lib -Wl,-rpath,/opt/homebrew/lib ifp/src/ifp.a pad/pad.a openroad_swig.a gpl/gpl.a dpl/dpl.a dpo/dpo.a fin/fin.a rsz/src/rsz.a ppl/ppl.a stt/stt.a dbSta/src/dbSta.a /Users/stefanthiede/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/libOpenSTA.a odb/src/swig/tcl/odbtcl.a rcx/src/rcx.a rmp/src/rmp.a cts/src/cts.a grt/grt.a tap/src/tap.a gui/gui.a drt/drt.a dst/dst.a mpl/mpl.a psm/src/psm.a ant/src/ant.a upf/src/upf.a utl/utl.a pdn/src/pdn.a dft/dft.a ../third-party/abc/libabc.a /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib mpl2/mpl2.a par/par.a /opt/homebrew/opt/or-tools/lib/libz.a _openroad_swig_py.a odb/src/swig/python/_odb_py.a ifp/src/_ifp_py.a utl/_utl_py.a ant/src/_ant_py.a grt/_grt_py.a gpl/_gpl_py.a dpl/_dpl_py.a mpl/_mpl_py.a ppl/_ppl_py.a tap/src/_tap_py.a cts/src/_cts_py.a drt/_drt_py.a dpo/_dpo_py.a fin/_fin_py.a rcx/src/_rcx_py.a rmp/src/_rmp_py.a stt/_stt_py.a psm/src/_psm_py.a pdn/src/_pdn_py.a dft/_dft_py.a upf/src/libupf_lib.a mpl2/libmpl2_lib.a /opt/homebrew/opt/or-tools/lib/libortools.9.6.dylib /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_check_op.a /opt/homebrew/opt/or-tools/lib/libabsl_die_if_null.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_parse.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_usage_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_conditions.a /opt/homebrew/opt/or-tools/lib/libabsl_log_flags.a /opt/homebrew/opt/or-tools/lib/libabsl_flags.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_marshalling.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_reflection.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_config.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_program_name.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_private_handle_accessor.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag.a /opt/homebrew/opt/or-tools/lib/libabsl_flags_commandlineflag_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_log_initialize.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_message.a /opt/homebrew/opt/or-tools/lib/libabsl_examine_stack.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_format.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_proto.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_nullguard.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_log_sink_set.a /opt/homebrew/opt/or-tools/lib/libabsl_log_globals.a /opt/homebrew/opt/or-tools/lib/libabsl_log_internal_globals.a /opt/homebrew/opt/or-tools/lib/libabsl_log_sink.a /opt/homebrew/opt/or-tools/lib/libabsl_log_entry.a /opt/homebrew/opt/or-tools/lib/libabsl_random_distributions.a /opt/homebrew/opt/or-tools/lib/libabsl_random_seed_sequences.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_pool_urbg.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_hwaes_impl.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_randen_slow.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_platform.a /opt/homebrew/opt/or-tools/lib/libabsl_random_internal_seed_material.a /opt/homebrew/opt/or-tools/lib/libabsl_random_seed_gen_exception.a /opt/homebrew/opt/or-tools/lib/libabsl_raw_hash_set.a /opt/homebrew/opt/or-tools/lib/libabsl_hashtablez_sampler.a /opt/homebrew/opt/or-tools/lib/libabsl_hash.a /opt/homebrew/opt/or-tools/lib/libabsl_city.a /opt/homebrew/opt/or-tools/lib/libabsl_low_level_hash.a /opt/homebrew/opt/or-tools/lib/libabsl_leak_check.a /opt/homebrew/opt/or-tools/lib/libabsl_statusor.a /opt/homebrew/opt/or-tools/lib/libabsl_status.a /opt/homebrew/opt/or-tools/lib/libabsl_cord.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_info.a /opt/homebrew/opt/or-tools/lib/libabsl_cord_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_functions.a /opt/homebrew/opt/or-tools/lib/libabsl_exponential_biased.a /opt/homebrew/opt/or-tools/lib/libabsl_cordz_handle.a /opt/homebrew/opt/or-tools/lib/libabsl_crc_cord_state.a /opt/homebrew/opt/or-tools/lib/libabsl_crc32c.a /opt/homebrew/opt/or-tools/lib/libabsl_crc_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_crc_cpu_detect.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_optional_access.a /opt/homebrew/opt/or-tools/lib/libabsl_strerror.a /opt/homebrew/opt/or-tools/lib/libabsl_bad_variant_access.a /opt/homebrew/opt/or-tools/lib/libabsl_str_format_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_synchronization.a /opt/homebrew/opt/or-tools/lib/libabsl_stacktrace.a /opt/homebrew/opt/or-tools/lib/libabsl_symbolize.a /opt/homebrew/opt/or-tools/lib/libabsl_debugging_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_demangle_internal.a /opt/homebrew/opt/or-tools/lib/libabsl_graphcy [... cut ...]
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk/System/Library/Frameworks'
Undefined symbols for architecture arm64:
  "_PyInit__par_py", referenced from:
      _main in Main.cc.o
  "sta::par_py_python_inits", referenced from:
      _main in Main.cc.o
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [src/openroad] Error 1
make[2]: *** [src/CMakeFiles/openroad.dir/all] Error 2
make[1]: *** [src/CMakeFiles/openroad.dir/rule] Error 2
make: *** [openroad] Error 2
```

maliberty, 2023-07-17:
Does it match
```
nm build/src/CMakeFiles/openroad.dir/Main.cc.o|grep par_py_python_inits
                 U _ZN3sta19par_py_python_initsE
```

stefanottili, 2023-07-17:
```
nm -g build/src/par/CMakeFiles/par_py.dir/par_py-pythonInitVar.cc.o|grep par_py_python_inits
000000000000ae90 D __ZN3sta19par_py_python_initsE
nm build/src/CMakeFiles/openroad.dir/Main.cc.o|grep par_py_python_inits                     
                 U __ZN3sta19par_py_python_initsE
```

maliberty, 2023-07-17:
In your link line I see par/par.a but in linux I see par/par.a and par/_par_py.a.  The later has the missing symbol.

maliberty, 2023-07-17:
What version of cmake do you have?

maliberty, 2023-07-17:
Does this work on other directories with python APIs?

maliberty, 2023-07-17:
Can you try adding par_py to the list of similar in src/CMakeLists.txt ?

maliberty, 2023-07-17:
Actually there is a second Darwin check that must be the issue in that file where par_py is conditional.

maliberty, 2023-07-17:
```
  if (NOT ${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
    # par_py causes abseil link error at startup on apple silicon so do not link it.
    target_link_libraries(openroad par_py)
  else()
    message(STATUS "Removing PAR_PY to avoid run time fatal error.")
  endif()
```

stefanottili, 2023-07-17:
```
cmake --version
cmake version 3.26.4
CMake suite maintained and supported by Kitware (kitware.com/cmake).
```
adding par/_par_py.a to the link command still gives me
```
Undefined symbols for architecture arm64:
  "_PyInit__par_py", referenced from:
      _main in Main.cc.o
  "sta::par_py_python_inits", referenced from:
      _main in Main.cc.o
ld: symbol(s) not found for architecture arm64
```
```
nm par/_par_py.a | grep par
partitionmgr-pyPYTHON_wrap.cxx.o:
0000000000000000 T _PyInit__par_py
000000000001d350 t __ZL24SwigPyObject_richcompareP12SwigPyObjectS0_i
000000000002a2d0 d __ZL27_swigc__p_par__PartitionMgr
0000000000029f20 d __ZL27_swigt__p_par__PartitionMgr
                 U __ZN3par12PartitionMgr16tritonPartDesignEjfjbibbffffPKcS2_S2_S2_fffffbRKNSt3__16vectorIfNS3_9allocatorIfEEEES9_S9_iiififiiiiifibiiii
                 U __ZN3par12PartitionMgr20readPartitioningFileERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEES9_
                 U __ZN3par12PartitionMgr20tritonPartHypergraphEjfjiiiPKcS2_S2_S2_S2_RKNSt3__16vectorIfNS3_9allocatorIfEEEES9_S9_iiififiiiiifibiiii
                 U __ZN3par12PartitionMgr21writePartitionVerilogEPKcS2_S2_
                 U __ZN3par12PartitionMgr23PartitionKWaySimpleModeEjfjRKNSt3__16vectorINS2_IiNS1_9allocatorIiEEEENS3_IS5_EEEERKNS2_IfNS3_IfEEEESD_
                 U __ZN3par12PartitionMgr26evaluateHypergraphSolutionEjfiiPKcS2_S2_S2_RKNSt3__16vectorIfNS3_9allocatorIfEEEES9_
                 U __ZN3par12PartitionMgr26evaluatePartDesignSolutionEjfbibffffPKcS2_S2_S2_S2_S2_fffffbRKNSt3__16vectorIfNS3_9allocatorIfEEEES9_
                 U __ZN3par12PartitionMgr4initEPN3odb10dbDatabaseEPN3sta9dbNetworkEPNS4_5dbStaEPN3utl6LoggerE
par_py-pythonInitVar.cc.o:
000000000000ae90 D __ZN3sta19par_py_python_initsE
```

maliberty, 2023-07-17:
(The original error on Ubuntu certainly had nothing to do with all this)

maliberty, 2023-07-17:
Does _par_py.a come after Main.cc.o in the link line?  They seem to have matching symbols.

stefanottili, 2023-07-17:
jump that was it, 'make openroad' succeeds.
But trying to run it shows why MPL2 and PAR have been disabled ;-(
Same issue as with mpl2_test, something 'being linked both statically and dynamically'.
```
417 #  if (NOT ${CMAKE_SYSTEM_NAME} MATCHES "Darwin")
418     # par_py causes abseil link error at startup on apple silicon so do not link it.
419     target_link_libraries(openroad par_py)
420 #  else()
421 #    message(STATUS "Removing PAR_PY to avoid run time fatal error.")
422 #  endif()
```
```
src/openroad 
ERROR: Something is wrong with flag 'flagfile' in file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc'. One possibility: file 'Users/corentinl/work/main/build_make/_deps/absl-src/absl/flags/parse.cc' is being linked both statically and dynamically into this executable. e.g. some files listed as srcs to a test and also listed as srcs of some shared lib deps of the same test.
```

stefanottili, 2023-07-17:
these are the shared libs in the link-command
```
/opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
/opt/homebrew/opt/or-tools/lib/libortools.9.6.dylib
/opt/homebrew/opt/or-tools/lib/libortools.9.6.dylib
/opt/homebrew/lib/libomp.dylib
/opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
/opt/homebrew/lib/libspdlog.1.9.2.dylib
/opt/homebrew/lib/libfmt.8.1.1.dylib
/opt/homebrew/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/lib/libpython3.11.dylib
```

maliberty, 2023-07-17:
Similar issues in https://github.com/google/or-tools/issues/2171

maliberty, 2023-07-17:
I wonder if the brew version is out of date.  Can you build or-tools from src?

maliberty, 2023-07-17:
9.6.2534 is the latest release.  We are are on 9.5.2237.  You could change the DependencyInstaller.sh to try the latest and see if that helps.

maliberty, 2023-07-17:
Perhaps ```set (CMAKE_FIND_LIBRARY_SUFFIXES ".a")``` would eliminate the shared library?

stefanottili, 2023-07-17:
here we go !!

removing /opt/homebrew/opt/or-tools/lib/libortools.9.6.dylib from the link command
gives me a working openroad executable with mpl2/par linked in, yay.

Now somebody with cmake knowledge "just" needs to put this in the right files.

maliberty, 2023-07-17:
Great!  Would you give the bad link line as it might have a clue why we have two (I'm guessing from both par & mpl2 somehow).

maliberty, 2023-07-17:
I see it is linked twice on linux as well but it doesn't mind.  I'll see if I can fix that on linux and then have you give it a try.

stefanottili, 2024-04-10:
using "brew install or-tools" which installs 9.9 instead of the DependencyInstaller.sh wget ortools...tar.gz and openroad build with mpl2 and par compiled in doesn't show a runtime error.

So when the DependencyInstaller is updated, mpl2 and par can be unconditionally be compiled in on MacOS.
Tested on Sonoma 14.4.1 M1.

stefanottili, 2024-04-11:
This issue has never been resolved, but it can be now.

gudeh, 2024-04-11:
Great! Indeed I do not see the problem on my end for some time. I will close this one. Feel free to open a new one.

closed 2024-04-11 by gudeh