The-OpenROAD-Project/OpenROAD-flow-scripts #1166: io_placement_issue doesn't place empty PLACE_PINS_ARGS into vars.sh generated

oharboe, 2023-06-21:
### Subject

[Stage]: Other. Please describe below.

### Describe the bug

Run `make io_placement_issue`, the vars*.sh file generated doesn't contain an empty PLACE_PINS_ARGS declaration.

This causes a problem because the existence of an empty PLACE_PINS_ARGS environment variable is required by https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/3548d6706f23c0c6b9fe43505787ee54b527ee13/flow/scripts/io_placement.tcl#L12

### Expected Behavior

Should output an empty PLACE_PINS_ARGS if needed into the vars*.sh file

### Environment

```shell
latest master
```

### To Reproduce

see above

vijayank88, 2023-06-21:
@oharboe 
As per my understanding, by default variables captured from design `config.mk` and platform `config.mk` into `vars.sh` during test case generation with `make *_issue`.
Have you included `PLACE_PIN_ARGS` in `config.mk`?

oharboe, 2023-06-21:
> @oharboe As per my understanding, by default variables captured from design `config.mk` and platform `config.mk` into `vars.sh` during test case generation with `make *_issue`. Have you included `PLACE_PIN_ARGS` in `config.mk` with additional arguments?

No.

This is a problem when there are no `PLACE_PINS_ARGS` defined in the `config.mk`, in which case `PLACE_PINS_ARGS` is defined, but empty. `vars*.sh` does not contain a defined, but empty definition of `PLACE_PINS_ARGS`, which causes the error.

vijayank88, 2023-06-21:
@vvbandeira 
If test case generated with `make io_placement_issue` and running again fails with un-defined variable
[vijayan@chipware-vijayan flow]$ ./run-me-gcd-sky130hd-base.sh 
```
OpenROAD v2.0-8693-g67ac6cf4b 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: io_placement.tcl, 13 can't read "::env(PLACE_PINS_ARGS)": no such variable
openroad> 
```

PR #1167 (2023-06-21): makefile: defined but empty variables are now included in make issue
fixes #1166
commit b9bf8001 (2023-06-21): makefile: defined but empty variables are now included in make issue

fixes #1166
closed 2023-06-21 by vvbandeira