The-OpenROAD-Project/OpenROAD-flow-scripts #1168: Update vars-*_timestamp.sh to avoid overwrtie with previous vars-*.sh

vijayank88, 2023-06-21:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

If we run `make *_issue` it creates `*.tar.gz` , `var*.sh`, `var*.tcl` and `var*.gdb`.

For same design if we run the same again for other stage of the flow we loose `var*` and it is updated to latest issue `var*.sh`.

### Expected Behavior

For tar file its named with time stamp `*_2023-06-20_13-18.tar.gz`, create similar time stamp for `var*_2023-06-20_13-18.sh`.

### Environment

```shell
With latest commit
```

### To Reproduce

`make floorplan_issue`

vvbandeira, 2023-06-21:
Not sure if this is a feature or an issue. The `var*` files are not lost; they are still in the respective tar file. Having only one `var` file for each run of `make *_issue` guarantees that any changes made to the local file work. Having multiple var files can increase the probability of user error. Example: user runs `make *_issue` twice, then uses the first set of `var` with the lasted results.

vijayank88, 2023-06-21:
This is useful when running with changes and creating test case for same design with different stages like floorplan,place,cts etc.
If I want to compare for previous, I need to untar the previous run to access the same. Got it.

closed 2023-06-21 by vijayank88

vvbandeira, 2023-06-21:
@vijayank88, FYI you can also extract only the var files. https://unix.stackexchange.com/a/149460