The-OpenROAD-Project/OpenROAD-flow-scripts #1172: cd flow; make (nandgate45/gcd) fails with 'realpath: illegal option -- -' on M1 MacOS

stefanottili, 2023-06-23:
### Subject

[Documentation] for documentation errors.

### Describe the bug

./build_openroad.sh --local --latest; cd tools/OpenROAD/build; make -j 8; make
mv src/openroad ../../install/OpenROAD/bin/

cd flow; make

nandgate/gcd fails after tap cell insertion step 5_1  with
realpath: illegal option -- - and eventually
ERROR ORD-0007] ./results/nangate45/gcd/base/2_floorplan.odb does not exist.

OpenROAD v2.0-8830-g112620f51 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: grid
        0.20 real         0.19 user         0.00 sys
realpath: illegal option -- -
usage: realpath [-q] [path ...]
ln -sf  results/nangate45/gcd/base/2_floorplan.odb
(/usr/bin/time /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/global_place_skip_io.tcl -metrics ./logs/nangate45/gcd/base/3_1_place_gp_skip_io.json) 2>&1 | tee ./logs/nangate45/gcd/base/3_1_place_gp_skip_io.log
OpenROAD v2.0-8830-g112620f51 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR ORD-0007] ./results/nangate45/gcd/base/2_floorplan.odb does not exist.

-rw-r--r--  1 user  staff   34664 Jun 22 18:08 results/nangate45/gcd/base/1_1_yosys.v
-rw-r--r--  1 user  staff   34664 Jun 22 18:08 results/nangate45/gcd/base/1_synth.v
-rw-r--r--  1 user  staff     456 Jun 22 18:08 results/nangate45/gcd/base/1_synth.sdc
-rw-r--r--  1 user  staff  497423 Jun 22 18:08 results/nangate45/gcd/base/2_1_floorplan.odb
-rw-r--r--  1 user  staff    5745 Jun 22 18:08 results/nangate45/gcd/base/2_floorplan.sdc
-rw-r--r--  1 user  staff  500951 Jun 22 18:08 results/nangate45/gcd/base/2_2_floorplan_io.odb
-rw-r--r--  1 user  staff  500951 Jun 22 18:08 results/nangate45/gcd/base/2_3_floorplan_tdms.odb
-rw-r--r--  1 user  staff  500951 Jun 22 18:08 results/nangate45/gcd/base/2_4_floorplan_macro.odb
-rw-r--r--  1 user  staff  508765 Jun 22 18:09 results/nangate45/gcd/base/2_5_floorplan_tapcell.odb
-rw-r--r--  1 user  staff  525160 Jun 22 18:09 results/nangate45/gcd/base/2_6_floorplan_pdn.odb

lrwxr-xr-x  1 user  staff      42 Jun 22 18:09 2_floorplan.odb@ -> results/nangate45/gcd/base/2_floorplan.odb
lrwxr-xr-x  1 user  staff      38 Jun 22 18:09 3_place.odb@ -> results/nangate45/gcd/base/3_place.odb
-rw-r--r--  1 user  staff    5745 Jun 22 18:09 results/nangate45/gcd/base/3_place.sdc
lrwxr-xr-x  1 user  staff      36 Jun 22 18:09 4_cts.odb@ -> results/nangate45/gcd/base/4_cts.odb
lrwxr-xr-x  1 user  staff      38 Jun 22 18:09 5_route.odb@ -> results/nangate45/gcd/base/5_route.odb
lrwxr-xr-x  1 user  staff      39 Jun 22 18:09 6_1_fill.odb@ -> results/nangate45/gcd/base/6_1_fill.odb

### Expected Behavior

cd flow; make 

should finish running gcd

### Environment

```shell
Git commit: 112620f51c8844e440838e73ac530b484c361250
kernel: Darwin 22.5.0
os: 		macOS 		13.4.1
cmake version 3.26.4
CMake Warning at src/CMakeLists.txt:233 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

-- The CXX compiler identification is AppleClang 14.0.3.14030022
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8830-g112620f51
[... 146 lines cut ...]
### Relevant log output

```shell
OpenROAD v2.0-8830-g112620f51 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO TAP-0004] Inserted 44 endcaps.
[INFO TAP-0005] Inserted 0 tapcells.
        0.20 real         0.18 user         0.01 sys
(/usr/bin/time /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/pdn.tcl -metrics ./logs/nangate45/gcd/base/2_6_pdn.json) 2>&1 | tee ./logs/nangate45/gcd/base/2_6_pdn.log
OpenROAD v2.0-8830-g112620f51 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: grid
        0.20 real         0.19 user         0.01 sys
realpath: illegal option -- -
usage: realpath [-q] [path ...]
ln -sf  results/nangate45/gcd/base/2_floorplan.odb
(/usr/bin/time /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/global_place_skip_io.tcl -metrics ./logs/nangate45/gcd/base/3_1_place_gp_skip_io.json) 2>&1 | tee ./logs/nangate45/gcd/base/3_1_place_gp_skip_io.log
OpenROAD v2.0-8830-g112620f51 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR ORD-0007] ./results/nangate45/gcd/base/2_floorplan.odb does not exist.
Error: global_place_skip_io.tcl, 2 ORD-0007
```

stefanottili, 2023-06-23:
realpath on MacOS doesn't have the --relative-to option that's used in flow/Makefile
```
flow % grep real Makefile
	ln -sf $(shell realpath --relative-to=$(dir $@) $<) $@

which realpath 
/bin/realpath

realpath --relative-to
realpath: illegal option -- -
usage: realpath [-q] [path ...]
```

stefanottili, 2023-06-23:
git pull --recurse-submodules

solved the issue. cd flow; make ran thru.

I was naively assuming that ./build_openroad --local --latest would pull the latest, but it doesn't.

closed 2023-06-23 by stefanottili

vvbandeira, 2023-06-23:
> git pull --recurse-submodules
> 
> solved the issue. cd flow; make ran thru.
> 
> I was naively assuming that ./build_openroad --local --latest would pull the latest, but it doesn't.

@stefanottili 

Please update your ORFS to the latest master. The `realpath` is no longer used and the symlinks are also not used. They both were removed on #1109 ~2 weeks ago.

Unless you have the variable `OPENROAD_FLOW_NO_GIT_INIT` set you are correct in your assumption, the build script does update the submodules. See [here](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/build_openroad.sh#L323-L327).

---

Aditional info: 
The version of `realpath` in macOS is different from the GNU version included with Linux distributions -- this is the case with many other bins, e.g., grep, sed, etc. You can install the GNU version with `brew install coreutils`.