The-OpenROAD-Project/OpenROAD-flow-scripts #1176: Update mock-array so that it can demonstrate routing by abutment

oharboe, 2023-06-23:
### Description

Using #1174, create a  `flow/settings.mk` file:

```
export DESIGN_CONFIG=designs/asap7/mock-array/config.mk
# Routing by abutment parameters
export MOCK_ARRAY_TABLE         = 8 8 4 4 4 4
export MACRO_BLOCKAGE_HALO      = 0
```

Run `make` and the result should be macros routed by abutment.

Actual result:

```
[INFO DRT-0195] Start 15th optimization iteration.
    Completing 10% with 56 violations.
    elapsed time = 00:00:00, memory = 1484.50 (MB).
    Completing 20% with 56 violations.
    elapsed time = 00:00:01, memory = 1811.88 (MB).
    Completing 30% with 56 violations.
    elapsed time = 00:00:01, memory = 1485.00 (MB).
    Completing 40% with 56 violations.
    elapsed time = 00:00:01, memory = 1487.32 (MB).
    Completing 50% with 56 violations.
    elapsed time = 00:00:03, memory = 1810.07 (MB).
    Completing 60% with 56 violations.
    elapsed time = 00:00:03, memory = 1484.50 (MB).
    Completing 70% with 56 violations.
    elapsed time = 00:00:04, memory = 1842.55 (MB).
    Completing 80% with 56 violations.
    elapsed time = 00:00:05, memory = 1484.50 (MB).
    Completing 90% with 56 violations.
    elapsed time = 00:00:06, memory = 1778.36 (MB).
    Completing 100% with 56 violations.
    elapsed time = 00:00:06, memory = 1484.50 (MB).
[INFO DRT-0199]   Number of violations = 56.
[deleted]
```

[image]

[image]

### Suggested Solution

Make clock work.

oharboe, 2023-06-24:
To load the children:

```
>>> read_def -child results/asap7/mock-array_Element/base/6_final.def

[INFO ODB-0135] Reading DEF file: results/asap7/mock-array_Element/base/6_final.def
[INFO ODB-0128] Design: Element
[INFO ODB-0138]     Created 80 pins.
[INFO ODB-0139]     Created 552 components and 1512 component-terminals.
[INFO ODB-0140]     Created 2 special nets and 1104 connections.
[INFO ODB-0141]     Created 228 nets and 408 connections.
[INFO ODB-0142] Finished DEF file: results/asap7/mock-array_Element/base/6_final.def
```

[image]

oharboe, 2023-06-24:
Here we can see the routing by abutment:

[image]

oharboe, 2023-06-24:
@maliberty What is the significance of the very slightly different green colors on the wire?

[image]

maliberty, 2023-06-26:
Most likely you have pin shapes overlapping wire segments.  I don't think it has much importance as the gds will be the same.

maliberty, 2023-06-26:
Is the issue here about clocks (the title) or rcx (the log)?

oharboe, 2023-06-26:
> Is the issue here about clocks (the title) or rcx (the log)?

I was, naively, hoping that openroad could "make cts work" when routing by abutment...

maliberty, 2023-06-26:
What do you want the final clock tree to look like?  Once there is no top level routing the tool has nothing to do.  The clock will be defined by the block designer and the top level array.

oharboe, 2023-06-26:
> What do you want the final clock tree to look like? Once there is no top level routing the tool has nothing to do. The clock will be defined by the block designer and the top level array.

I dont think there is anything that CTS can legally do as it would have to put the CTS inside the macros. I believe the blockages prevent this, even if it can be seen in the OpenROAD GUI that there is enough space inside the macros. 

There would have to be some sort of "scavenging" of space for the CTS inside the macros. I am sure that "solution" comes with its own problems.

Thoughts?

maliberty, 2023-06-26:
I think it is too large a leap to do scavenging.  I think you are better off shrinking the macro and leaving some top level space to put in the clock tree.

closed 2023-06-26 by oharboe

oharboe, 2023-06-29:
@nerriav Try horizontal routing by abutment only:

restrict clock to left edge and use `settings.mk`

```
export DESIGN_CONFIG=designs/asap7/mock-array/config.mk
# Routing by abutment parameters
export MOCK_ARRAY_TABLE         = 8 8 4 4 4 6
export MACRO_BLOCKAGE_HALO      = 0
```