The-OpenROAD-Project/OpenROAD-flow-scripts #1178: gcd builds for sky130hd and nangate45 but not for asap7

stefanottili, 2023-06-23:
### Subject

[Design] for example design issues (i.e. the design does not pass functional validation, etc.)

### Describe the bug

I can build gcd for sky103hd and nangate45, but asap7 fails in 1_1_yosys
Comparing nangate45 log with logs/asap7/gcd/base/1_1_yosys.log shows

nangate45:
[FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 460

asap7:
[FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 
ERROR: TCL interpreter returned an error: empty expression in expression ""

### Expected Behavior

asap7 gcd to run thru

### Environment

```shell
M1 MacOS built Jun 23 2023

OpenROAD v2.0-8836-g0ea657420
Yosys 0.13+15 (git sha1 bc027b2ca, gcc 14.0.3 -fPIC -Os)
```

### To Reproduce

cd flow; make 

### Relevant log output

```shell
flow/Makefile

 74 # DESIGN_CONFIG=./designs/asap7/ethmac/config.mk
 75  DESIGN_CONFIG=./designs/asap7/gcd/config.mk
 76 # DESIGN_CONFIG=./designs/asap7/ibex/config.mk

16.16. Finished OPT passes. (There is nothing left to do.)
Using ABC speed script.
[FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 
ERROR: TCL interpreter returned an error: empty expression in expression ""
        1.12 real         1.05 user         0.05 sys
mkdir -p ./results/asap7/gcd/base ./logs/asap7/gcd/base ./reports/asap7/gcd/base
cp results/asap7/gcd/base/1_1_yosys.v results/asap7/gcd/base/1_synth.v
cp: results/asap7/gcd/base/1_1_yosys.v: No such file or directory

logs/nangate45/gcd/base/1_1_yosys.log
[FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 460
```

maliberty, 2023-06-23:
Are you using the current yosys from ORFS?  It passes in the nightly CI.

stefanottili, 2023-06-24:
./build_openroad.sh --local --latest 
installs yosys in 
13394216 Jun 23 16:50 tools/install/yosys/bin/yosys

I'm totally unfamiliar with the script/constraint setup for ORFS, where is the clock supposed to come from ?

util/genMassive.py:263:    CurAbcClkPeriod = CurAttrs.get('ABC_CP')
util/genMassive.py:370:    if CurAbcClkPeriod != 'empty':
util/genMassive.py:371:        fo.write('export ABC_CLOCK_PERIOD_IN_PS = %s\n' % CurAbcClkPeriod)

I'm assuming hat 390ns clk_period for asap7 is on the generous side, but that's the not cause of my problem.

diff designs/asap7/gcd/constraint.sdc designs/nangate45/gcd/constraint.sdc
5c5
< set clk_period 390 
---
> set clk_period 0.46 
10c10
< create_clock -name $clk_name -period $clk_period  $clk_port
---
> create_clock -name $clk_name -period $clk_period $clk_port

rovinski, 2023-06-25:
FYI ASAP7 is in ps so that's 390ps.

vijayank88, 2023-07-12:
@stefanottili 
Try following command:
```
make DESIGN_CONFIG=./designs/asap7/gcd/config.mk clean_all
make DESIGN_CONFIG=./designs/asap7/gcd/config.mk
```

stefanottili, 2023-07-13:
```
make DESIGN_CONFIG=./designs/asap7/gcd/config.mk clean_all
make DESIGN_CONFIG=./designs/asap7/gcd/config.mk
[INFO][FLOW] Using platform directory ./platforms/asap7
[INFO-FLOW] ASU ASAP7 - version 2
Default PVT selection: BC
mkdir -p ./results/asap7/gcd/base ./logs/asap7/gcd/base ./reports/asap7/gcd/base
(/usr/bin/time /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys -v 3 -c ./scripts/synth.tcl) 2>&1 | tee ./logs/asap7/gcd/base/1_1_yosys.log
1. Executing Verilog-2005 frontend: ./designs/src/gcd/gcd.v
2. Executing Liberty frontend.
...
16.16. Finished OPT passes. (There is nothing left to do.)
Using ABC speed script.
[FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 
ERROR: TCL interpreter returned an error: empty expression
in expression ""
        1.69 real         1.11 user         0.06 sys
mkdir -p ./results/asap7/gcd/base ./logs/asap7/gcd/base ./reports/asap7/gcd/base
cp results/asap7/gcd/base/1_1_yosys.v results/asap7/gcd/base/1_synth.v
cp: results/asap7/gcd/base/1_1_yosys.v: No such file or directory
```

vijayank88, 2023-07-13:
@stefanottili 
Goto `tools/OpenROAD` and share log of `./etc/Env.sh`

maliberty, 2023-07-13:
It is passing in the nightly CI for linux https://jenkins.openroad.tools/job/OpenROAD-flow-scripts-Nightly-Public/2744/execution/node/652/log/

We don't officially support mac (due to lack of hw) but many community users use it.  It seems like some tcl issue.  How did you install your dependencies?

stefanottili, 2023-07-13:
tools/OpenROAD/etc/DependencyInstaller.sh uses homebrew and I wouldn't suspect this because nangate45 and sky130 gcd works just fine.

after removing the flow #'s (because the differ so much) I was able to diff nangate45 vs asap7 gcd 1_1_yosys.log
```
diff  logs/nangate45/gcd/base/1_1_yosys.log logs/asap7/gcd/base/1_1_yosys.log 
```
Why does asap7 use the ABC area script whereas nangate45 uses the speed one ?
Why would the speed script fail to set the ABC_CLOCK_PERIOD_IN_PS whereas the other doesn't ?
```
199,217c208,212
< Using ABC area script.
< [FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 460
...
> Using ABC speed script.
> [FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 
> ERROR: TCL interpreter returned an error: empty expression
```
```
3c3,7
<  Executing Verilog-2005 frontend: ./platforms/nangate45/cells_clkgate.v
---
>  Executing Liberty frontend.
>  Executing Liberty frontend.
>  Executing Liberty frontend.
>  Executing Liberty frontend.
>  Executing Verilog-2005 frontend: ./platforms/asap7/yoSys/cells_clkgate_R.v
168a173
>  Executing EXTRACT_FA pass (find and extract full/half adders).
170c175
<  Executing Verilog-2005 frontend: ./platforms/nangate45/cells_latch.v
---
>  Executing Verilog-2005 frontend: ./platforms/asap7/yoSys/cells_adders_R.v
171a177,188
>  Executing TECHMAP pass (map to technology primitives).
>  Executing Verilog-2005 frontend: /Users/user/OpenROAD-flow-scripts/tools/install/yosys/bin/../share/yosys/techmap.v
>  Continuing TECHMAP pass.
>  Executing OPT pass (performing simple optimizations).
>  Executing OPT_EXPR pass (perform const folding).
>  Executing OPT_MERGE pass (detect identical cells).
>  Executing OPT_DFF pass (perform DFF optimizations).
>  Executing OPT_CLEAN pass (remove unused cells and wires).
>  Finished fast OPT passes.
>  Executing TECHMAP pass (map to technology primitives).
>  Executing Verilog-2005 frontend: ./platforms/asap7/yoSys/cells_latch_R.v
>  Continuing TECHMAP pass.
173,180d189
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFFRS_X1' - skipping.
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFFRS_X2' - skipping.
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFFR_X1' - skipping.
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFFR_X2' - skipping.
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFFS_X1' - skipping.
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFFS_X2' - skipping.
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFF_X1' - skipping.
< Warning: Found unsupported expression 'SE*SI+D*!SE' in pin attribute of cell 'SDFF_X2' - skipping.
199,217c208,212
< Using ABC area script.
< [FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 460
<  Executing ABC pass (technology mapping using ABC).
<  Extracting gate netlist of module `\gcd' to `<abc-temp-dir>/input.blif'..
<  Executing ABC.
<  Re-integrating ABC results.
<  Executing SETUNDEF pass (replace undef values with defined constants).
<  Executing SPLITNETS pass (splitting up multi-bit signals).
<  Executing OPT_CLEAN pass (remove unused cells and wires).
<  Executing HILOMAP pass (mapping to constant drivers).
<  Executing INSBUF pass (insert buffer cells for connected wires).
<  Executing CHECK pass (checking for obvious problems).
<  Printing statistics.
<  Executing Verilog backend.
< Warnings: 8 unique messages, 72 total
< End of script. Logfile hash: cbf4d8ee81, CPU: user 0.34s system 0.02s
< Yosys 0.13+15 (git sha1 bc027b2ca, gcc 14.0.3 -fPIC -Os)
< Time spent: 37% 2x abc (0 sec), 11% 2x read_liberty (0 sec), ...
<         0.56 real         0.53 user         0.03 sys
---
> Using ABC speed script.
> [FLOW] Set ABC_CLOCK_PERIOD_IN_PS to: 
> ERROR: TCL interpreter returned an error: empty expression
> in expression ""
>         1.74 real         1.12 user         0.07 sys
```

stefanottili, 2023-07-14:
> We don't officially support mac (due to lack of hw)

It used to work just fine all of last year. Introducing mpl2 using or-tool static/shared linking the same file broke the build. Disabling of mpl2 doesn’t quite work, the build fails on a test, but it can be manually forced into creating a working openroad binary.

1. klayout runs ci including macOS.

2. Since OR promotes the use of AWS, the lack of physical hardware shouldn’t be a showstopper, right ?
The guys demonstrating klayout and OR running on AWS at SEMI assured me that they have both x86 and apple silicon for rent.

rovinski, 2023-07-17:
@stefanottili the ABC period is admittedly extracted using a brittle shell expression:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/69a30ddc0e2da34486d177f89ecf752050dd8ad4/flow/platforms/asap7/config.mk#L60

The expression is slightly different between ASAP7 and other platforms which could be causing the empty expression you see. Can you try evaluating that expression in your shell and seeing what you get?

As a short term workaround, you can simply set `ABC_CLOCK_PERIOD_IN_PS` in your config file to match the SDC or even an arbitrary positive number. We don't find that changing this parameter has much effect on the final output.

stefanottili, 2023-07-18:
putting 'export ABC_CLOCK_PERIOD_IN_PS = 390' in line 12 of designs/asap7/gcd/config.mk allows to run asap7/gcd.

There is no "shell" on MacOS with homebrew and the sed command doesn't seem to work either.

Since the scripting works for sky130hs, sky130hd and nangate45, why is the scripting done differently for asap7 ?

rovinski, 2023-07-18:
> There is no "shell" on MacOS with homebrew and the sed command doesn't seem to work either.

Make is supposed to use the shell specified by `$SHELL` on your system.

This is the difference between the two platforms:

```diff
+export ABC_CLOCK_PERIOD_IN_PS ?= $(shell sed -nE "s/^set\s+clk_period\s+(\S+).*|.*-period\s+(\S+).*/\1\2/p" $(SDC_FILE) | head -1 | awk '{print $$1}')
-export ABC_CLOCK_PERIOD_IN_PS ?= $(shell sed -nE "s/^set clk_period (.+)|.* -period (.+) .*/\1\2/p" $(SDC_FILE) | head -1 | awk '{print $$1*1000}')
```

I thought the only difference was the `*1000` but on closer inspection it seems like the ASAP7 regex uses shorthand characters `\s` and `\S` for (non-) whitespace characters.

Technically the ASAP7 version is more precise but it looks like your version of sed doesn't support it by default. Can you try these two things and let me know if one of them works?

1. Try adding a `-r` flag to `sed`
2. Try replacing the `sed` expression with the one from nangate45. Keep the `awk` expression the same.

stefanottili, 2023-07-18:
SHELL=/bin/zsh

brew install gnu-sed 

and adding
PATH="/opt/homebrew/opt/gnu-sed/libexec/gnubin:$PATH"
to ~/.zshrc did the trick.
I can now run asap7/mock-array with mpl2 since openroad can be built with par/mpl2.

Might be worth adding gnu-sed to the etc/DependencyInstaller.sh for Apple Silicon.

maliberty, 2023-07-18:
@vvbandeira see note about sed dependency above

stefanottili, 2023-07-18:
openroad compiled with mpl2, gnu-sed allows to run asap7/gcd, trying to run riscv32i gives me:

HierRTLMP Flow enabled...
Call Macro Placer  -halo_width 3 -report_directory ./objects/asap7/riscv32i/base/rtlmp
Error: macro_place.tcl, 141 invalid command name "rtl_macro_placer"

stefanottili, 2023-07-18:
Removing #ifdef ENABLE_MP2/ENABLE_PAR in OpenROAD.cc, manually linking openroad with ortools.dylib behind libabsl_flags_reflection.a followed by 'make install' got things working.

asap7/riscv32i is running.