The-OpenROAD-Project/OpenROAD-flow-scripts #1181: Error occurred while building

lllyuhao, 2023-06-25:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I am using Ubuntu 20.04 and build is failing. It seems like something wrong with the spdlog/fmt.h. 

### Expected Behavior

Build successfully

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-73-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8836-g0ea657420
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/libgtest.a  
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
[... 134 lines cut ...]
/usr/bin/ld: libutl_lib.a(ScopedTemporaryFile.cpp.o): in function `utl::ScopedTemporaryFile::~ScopedTemporaryFile() [clone .part.0]':
ScopedTemporaryFile.cpp:(.text+0x3e0): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: ScopedTemporaryFile.cpp:(.text+0x41f): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: ScopedTemporaryFile.cpp:(.text+0x549): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: ScopedTemporaryFile.cpp:(.text+0x588): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: libutl_lib.a(ScopedTemporaryFile.cpp.o): in function `utl::ScopedTemporaryFile::ScopedTemporaryFile(utl::Logger*)':
ScopedTemporaryFile.cpp:(.text+0x8b3): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: ScopedTemporaryFile.cpp:(.text+0x8fa): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: ScopedTemporaryFile.cpp:(.text+0xa1c): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: ScopedTemporaryFile.cpp:(.text+0xa63): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: libutl_lib.a(ScopedTemporaryFile.cpp.o): in function `void spdlog::logger::log_<fmt::v7::basic_string_view<char>, char const*, char const*, int, char [4096]>(spdlog::source_loc, spdlog::level::level_enum, fmt::v7::basic_string_view<char> const&, char const* const&, char const* const&, int const&, char const (&) [4096])':
ScopedTemporaryFile.cpp:(.text._ZN6spdlog6logger4log_IN3fmt2v717basic_string_viewIcEEJPKcS7_iA4096_cEEEvNS_10source_locENS_5level10level_enumERKT_DpRKT0_[_ZN6spdlog6logger4log_IN3fmt2v717basic_string_viewIcEEJPKcS7_iA4096_cEEEvNS_10source_locENS_5level10level_enumERKT_DpRKT0_]+0x11f): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: ScopedTemporaryFile.cpp:(.text._ZN6spdlog6logger4log_IN3fmt2v717basic_string_viewIcEEJPKcS7_iA4096_cEEEvNS_10source_locENS_5level10level_enumERKT_DpRKT0_[_ZN6spdlog6logger4log_IN3fmt2v717basic_string_viewIcEEJPKcS7_iA4096_cEEEvNS_10source_locENS_5level10level_enumERKT_DpRKT0_]+0x15b): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: libutl_lib.a(Logger.cpp.o): in function `utl::Logger::flushMetrics()':
Logger.cpp:(.text+0x16df): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: Logger.cpp:(.text+0x1730): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
[... cut ...] ails::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: libutl_lib.a(Metrics.cpp.o): in function `utl::MetricsEntry::assembleJSON(std::__cxx11::list<utl::MetricsEntry, std::allocator<utl::MetricsEntry> > const&)':
Metrics.cpp:(.text+0x22c): undefined reference to `fmt::v7::detail::vformat[abi:cxx11](fmt::v7::basic_string_view<char>, fmt::v7::format_args)'
collect2: error: ld returned 1 exit status
```

vijayank88, 2023-06-26:
Try steps from here: https://openroad-flow-scripts.readthedocs.io/en/latest/user/BuildWithDocker.html
Seems you missed `--recursive`

lllyuhao, 2023-06-27:
Sry, I forgot to type `--recursive` in my expression.  I have solved the error about the "lemon-config" . After running setup.sh successfully, I cloned the OpenRoad.git and tried to build it. But I ran into a new problem as shown below：
`CMake Error at src/mpl2/CMakeLists.txt:42 (find_package):
  By not providing "Findortools.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "ortools", but
  CMake did not find one.

  Could not find a package configuration file provided by "ortools" with any
  of the following names:

    ortoolsConfig.cmake
    ortools-config.cmake

  Add the installation prefix of "ortools" to CMAKE_PREFIX_PATH or set
  "ortools_DIR" to a directory containing one of the above files.  If
  "ortools" provides a separate development package or SDK, be sure it has
  been installed.
`

vijayank88, 2023-06-27:
> After running setup.sh successfully, I cloned the OpenRoad.git and tried to build it.

Don't understand your steps. Why you're cloning `OpenROAD.git` again? Where it is mentioned?

RaidToRadar, 2023-06-30:
I am also having issues building with docker:
```sh
#0 730.6 [ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/bufferTreeDescriptor.cpp.o

#0 731.0 [ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/gui_autogen/3YJK5W5UP7/qrc_resource.cpp.o

#0 733.6 [ 78%] Linking CXX static library gui.a

#0 733.8 [ 78%] Built target gui

#0 733.8 gmake: *** [Makefile:146: all] Error 2

------

Dockerfile.builder:14

--------------------

  12 |     WORKDIR /OpenROAD-flow-scripts

  13 |     

  14 | >>> RUN ./build_openroad.sh --no_init --local --threads ${numThreads}

  15 |     

--------------------

ERROR: failed to solve: process "/bin/sh -c ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 2

```

vijayank88, 2023-06-30:
@RaidToRadar 
How many threads used for the installation? Can you share command used?

RaidToRadar, 2023-06-30:
I used all 4 and I have allocate 4 cores to the vbox. This is the command that I followed:
```sh
export OS_NAME=ubuntu22.04
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./etc/DockerHelper.sh create -target=dev -os=$OS_NAME
./etc/DockerHelper.sh create -target=builder -os=$OS_NAME
```
The error arises on while running the builder DockerHelper.sh

vijayank88, 2023-06-30:
Can you try with single thread?
```
./etc/DockerHelper.sh create -target=builder -os=$OS_NAME -threads=1
```

RaidToRadar, 2023-06-30:
The same error is prompted
```sh
#0 4060.2 [ 95%] Built target parseTest
#0 4060.2 [ 95%] Building CXX object src/odb/test/cpp/CMakeFiles/OdbGTests.dir/TestDbWire.cc.o
#0 4062.0 [ 95%] Building CXX object src/odb/test/cpp/CMakeFiles/OdbGTests.dir/TestAbstractLef.cc.o
#0 4062.0 /OpenROAD-flow-scripts/tools/OpenROAD/src/odb/test/cpp/TestAbstractLef.cc:13:10: fatal error: gmock/gmock.h: No such file or directory
#0 4062.0    13 | #include "gmock/gmock.h"
#0 4062.0       |          ^~~~~~~~~~~~~~~
#0 4062.0 compilation terminated.
#0 4062.0 gmake[2]: *** [src/odb/test/cpp/CMakeFiles/OdbGTests.dir/build.make:90: src/odb/test/cpp/CMakeFiles/OdbGTests.dir/TestAbstractLef.cc.o] Error 1
#0 4062.0 gmake[1]: *** [CMakeFiles/Makefile2:2303: src/odb/test/cpp/CMakeFiles/OdbGTests.dir/all] Error 2
#0 4062.0 gmake: *** [Makefile:146: all] Error 2
------
Dockerfile.builder:14
--------------------
  12 |     WORKDIR /OpenROAD-flow-scripts
  13 |     
  14 | >>> RUN ./build_openroad.sh --no_init --local --threads ${numThreads}
  15 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c ./build_openroad.sh --no_init --local --threads ${numThreads}" did not complete successfully: exit code: 2
vboxuser@vbox-inator:~/OpenROAD-flow-scripts$ 
```

vijayank88, 2023-06-30:
@RaidToRadar 
The error is not same, reported by owner of this issue.
It is similar to this issue and got resolved: https://github.com/The-OpenROAD-Project/OpenROAD/issues/3520
So do clean build again with following steps:
```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh --clean --latest --threads 1
```
If there is an issue, open a new issue with the `build_openroad.lo`

RaidToRadar, 2023-06-30:
Okay thank you, perhaps my issue is that I was attempting to build it with ubuntu22.04. It was able to build with the centos version.

lllyuhao, 2023-07-02:
> > After running setup.sh successfully, I cloned the OpenRoad.git and tried to build it.
> 
> Don't understand your steps. Why you're cloning `OpenROAD.git` again? Where it is mentioned?

Sry, I didn't explain clearly. Forget about the previous bug. I have successfully built the Openroad.git and now I'm trying to build the Openroad-flow-scripts.git. After finish running` sudo ./setup.sh`, here is the error occurred while running `./build_openroad.sh --local`.

```
[ 78%] Swig compile src/partitionmgr.i for tcl
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/odb/test/cpp/TestAbstractLef.cc:13:10: fatal error: gmock/gmock.h: No such file or directory
   13 | #include "gmock/gmock.h"
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [src/odb/test/cpp/CMakeFiles/OdbGTests.dir/build.make:90: src/odb/test/cpp/CMakeFiles/OdbGTests.dir/TestAbstractLef.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2303: src/odb/test/cpp/CMakeFiles/OdbGTests.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Consolidate compiler generated dependencies of target mpl2_lib
[ 78%] Built target dpo
[ 78%] Built target mpl2_lib
Consolidate compiler generated dependencies of target grt_lib
Consolidate compiler generated dependencies of target gui
Unable to open file /home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/build/src/par/CMakeFiles/par.dir/partitionmgr.i.d: Permission denied
make[2]: *** [src/par/CMakeFiles/par.dir/build.make:76: src/par/CMakeFiles/par.dir/partitionmgrTCL_wrap.cxx] Error 1
make[1]: *** [CMakeFiles/Makefile2:4809: src/par/CMakeFiles/par.dir/all] Error 2
[ 78%] Built target grt_lib
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/search.cpp.o
Unable to open file /home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/build/src/par/CMakeFiles/par_py.dir/partitionmgr-py.i.d: Permission denied
make[2]: *** [src/par/CMakeFiles/par_py.dir/build.make:76: src/par/CMakeFiles/par_py.dir/partitionmgr-pyPYTHON_wrap.cxx] Error 1
make[1]: *** [CMakeFiles/Makefile2:4847: src/par/CMakeFiles/par_py.dir/all] Error 2
[ 78%] Built target libabc
In file included from /home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.cpp:33:
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:264:10: error: ‘mutex’ in namespace ‘std’ does not name a type
  264 |     std::mutex shapes_init_mutex_;
      |          ^~~~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:41:1: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
   40 | #include "odb/dbBlockCallBackObj.h"
  +++ |+#include <mutex>
   41 | #include "odb/geom.h"
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:267:10: error: ‘mutex’ in namespace ‘std’ does not name a type
  267 |     std::mutex fills_init_mutex_;
      |          ^~~~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:267:5: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
  267 |     std::mutex fills_init_mutex_;
      |     ^~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:270:10: error: ‘mutex’ in namespace ‘std’ does not name a type
  270 |     std::mutex insts_init_mutex_;
      |          ^~~~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:270:5: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
  270 |     std::mutex insts_init_mutex_;
      |     ^~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:273:10: error: ‘mutex’ in namespace ‘std’ does not name a type
  273 |     std::mutex blockages_init_mutex_;
      |          ^~~~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:273:5: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
  273 |     std::mutex blockages_init_mutex_;
      |     ^~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:276:10: error: ‘mutex’ in namespace ‘std’ does not name a type
  276 |     std::mutex obstructions_init_mutex_;
      |          ^~~~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:276:5: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
  276 |     std::mutex obstructions_init_mutex_;
      |     ^~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:279:10: error: ‘mutex’ in namespace ‘std’ does not name a type
  279 |     std::mutex rows_init_mutex_;
      |          ^~~~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.h:279:5: note: ‘std::mutex’ is defined in header ‘<mutex>’; did you forget to ‘#include <mutex>’?
  279 |     std::mutex rows_init_mutex_;
      |     ^~~
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.cpp: In member function ‘void gui::Search::updateShapes(odb::dbBlock*)’:
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.cpp:251:8: error: ‘lock_guard’ is not a member of ‘std’
  251 |   std::lock_guard<std::mutex> lock(data.shapes_init_mutex_);
[... 81 lines cut ...]
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.cpp:346:31: error: ‘lock’ was not declared in this scope; did you mean ‘lockf’?
  346 |   std::lock_guard<std::mutex> lock(data.obstructions_init_mutex_);
      |                               ^~~~
[... cut ...]
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/gui/src/search.cpp:363:31: error: ‘lock’ was not declared in this scope; did you mean ‘lockf’?
  363 |   std::lock_guard<std::mutex> lock(data.rows_init_mutex_);
      |                               ^~~~
      |                               lockf
make[2]: *** [src/gui/CMakeFiles/gui.dir/build.make:288: src/gui/CMakeFiles/gui.dir/src/search.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:5017: src/gui/CMakeFiles/gui.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
```

lllyuhao, 2023-07-02:
And here is the output of `./etc/Env.sh` in /tools/Openroad. Could you give me some help?
```
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-76-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8836-g0ea657420
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/libgtest.a  
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-sign-compare - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test C_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-deprecated - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-c++11-narrowing
[... 60 lines cut ...]
-- TCL header: /usr/include/tcl.h
-- Found re2: /opt/or-tools/lib/cmake/re2/re2Config.cmake (found version "9.0.0") 
-- Found Clp: /opt/or-tools/lib/cmake/Clp/ClpConfig.cmake (found version "1.17.7") 
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.3.9") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found version "1.76.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.TwkXfFR1XE

```

maliberty, 2023-07-02:
Please try with ```./build_openroad.sh --local --latest```

lllyuhao, 2023-07-04:
Here is the output of `./build_openroad.sh --local --latest`. It didn't make any difference.
```
[ 78%] Swig compile src/partitionmgr.i for tcl
Unable to open file /home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/build/src/par/CMakeFiles/par.dir/partitionmgr.i.d: Permission denied
make[2]: *** [src/par/CMakeFiles/par.dir/build.make:76: src/par/CMakeFiles/par.dir/partitionmgrTCL_wrap.cxx] Error 1
make[1]: *** [CMakeFiles/Makefile2:4974: src/par/CMakeFiles/par.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/staGui.cpp.o
/home/lengyuhao/OpenROAD-flow-scripts/tools/OpenROAD/src/mpl2/src/rtl_mp.cpp:127:1: fatal error: opening dependency file CMakeFiles/mpl2_lib.dir/src/rtl_mp.cpp.o.d: Permission denied
  127 | }  // namespace mpl2
      | ^
compilation terminated.
make[2]: *** [src/mpl2/CMakeFiles/mpl2_lib.dir/build.make:76: src/mpl2/CMakeFiles/mpl2_lib.dir/src/rtl_mp.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4589: src/mpl2/CMakeFiles/mpl2_lib.dir/all] Error 2
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/staGuiInterface.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/timingWidget.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/drcWidget.cpp.o
[ 78%] Linking CXX static library libFastRoute4.1.a
[ 78%] Built target FastRoute4.1
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/ruler.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/heatMap.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/heatMapSetup.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/heatMapPlacementDensity.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/browserWidget.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/globalConnectDialog.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/clockWidget.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/colorGenerator.cpp.o
[ 78%] Building CXX object src/gui/CMakeFiles/gui.dir/src/bufferTreeDescriptor.cpp.o
[ 78%] Linking CXX static library gui.a
[ 78%] Built target gui
[ 78%] Linking CXX static library _odb_py.a
[ 78%] Built target odb_py
[ 78%] Linking CXX static library odbtcl.a
[ 78%] Built target odbtcl
make: *** [Makefile:146: all] Error 2
```

vijayank88, 2023-07-04:
@lllyuhao 
Do not mix up docker and local installation in same directory. 
See in the above thread, user able to complete docker based installation successfully.
So try in new directory with clean build again, with docker based to avoid dependency issues.
```
cd
mkdir ORFS && cd ORFS
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh --clean --latest --threads 1
```
Also can you confirm is docker able to run without sudo access. Check with below command
```
docker run hello-world
```

lllyuhao, 2023-07-04:
I have always tried local installation and never tried docker installation.

vijayank88, 2023-07-04:
@lllyuhao 
As per `To Reproduce` you mentioned during issue creation
```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
./build_openroad.sh
```
It is for docker based installation. Also I do not found word, you tried local installation in above thread.
If you tried local installation mean, did you completed `sudo ./setup.sh` for dependency installer successfully?
Try these again:
```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
./build_openroad.sh --local --latest --clean --threads 1
```