The-OpenROAD-Project/OpenROAD-flow-scripts #1182: make route failed, congestion in class pad

xuyang-stack, 2023-06-26:
make route
failed:
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/nangate45/gcd/base/congestion.rpt in the DRC viewer.
[image]
[image]
[image]

**During the routing phase, congestion occurs. Upon checking the Design Rule Check (DRC) report, I noticed that the congestion is occurring within the module itself. This is expected because there are several obstruction (obs) regions in that area, and congestion is normal when attempting to route in such congested areas. However, ABC_GPIO should be treated as an independent module where internal routing is not necessary. It should only require connections from its pins to other modules. Why is this issue occurring?**

In my own design, I made the following additions:

tcl file:
place_pad -master ABC_GPIO -row IO_SOUTH -location 420 {U_GPIOA_2}

v file:
   ABC_GPIO U_GPIOA_2 (
    .C(reset_io ),
    .I(1'b0 ),
    .IE(1'b1 ),
    .OEN(1'b1 ),
    .PAD(reset_pad )
  );

lef file:
MACRO ABC_GPIO
  CLASS  PAD ;
  FOREIGN ABC_GPIO 0.000 0.000 ;
  ORIGIN 0.000 0.000 ;
  SIZE 30.000 BY 177.000 ;
  SYMMETRY R90  ;
  SITE IOSite ;
  PIN IE
    DIRECTION INPUT ;
    PORT
      LAYER M5 ;
        RECT 19.095 176.460 21.055 177.000 ;
      LAYER M4 ;
        RECT 19.095 176.460 21.055 177.000 ;
      LAYER M3 ;
        RECT 19.095 176.460 21.055 177.000 ;
      LAYER M2 ;
        RECT 19.095 176.460 21.055 177.000 ;
        RECT 19.780 170.220 19.900 177.000 ;
        RECT 18.905 170.220 19.900 170.340 ;
      LAYER M1 ;
        RECT 19.095 176.460 21.055 177.000 ;
    END
  AntennaGateArea 1 ;
  AntennaDiffArea 0.590 ;
  END IE
  PIN C
    DIRECTION OUTPUT ;
    PORT
      LAYER M5 ;
        RECT 26.860 176.460 28.820 177.000 ;
      LAYER M4 ;
        RECT 26.860 176.460 28.820 177.000 ;
      LAYER M3 ;
        RECT 26.860 176.460 28.820 177.000 ;
      LAYER M2 ;
        RECT 26.860 176.460 28.820 177.000 ;
       ***************
        RECT 21.790 170.505 21.910 172.690 ;
      LAYER M1 ;
        RECT 26.860 176.460 28.820 177.000 ;
    END
  AntennaGateArea 0 ;
  AntennaDiffArea 0.590 ;
  END C
  PIN I
    DIRECTION INPUT ;
    PORT
      LAYER M5 ;
        RECT 1.180 176.460 3.140 177.000 ;
      LAYER M4 ;
        RECT 1.180 176.460 3.140 177.000 ;
      LAYER M3 ;
        RECT 1.180 176.460 3.140 177.000 ;
      LAYER M2 ;
        RECT 5.440 170.210 5.720 170.330 ;
      ********************
        RECT 2.080 170.210 2.200 177.000 ;
      LAYER M1 ;
        RECT 1.180 176.460 3.140 177.000 ;
    END
  AntennaGateArea 0.5 ;
  AntennaDiffArea 0.590 ;
  END I
  PIN PAD
    DIRECTION INOUT ;
    PORT
      LAYER TM2 ;
        RECT 2.000 0.000 28.000 3.000 ;
      LAYER TM1 ;
        RECT 2.000 0.000 28.000 3.000 ;
      LAYER M7 ;
        RECT 2.000 0.000 28.000 3.000 ;
      LAYER M6 ;
        RECT 2.000 0.000 28.000 3.000 ;
      LAYER M5 ;
        RECT 2.000 0.000 28.000 3.000 ;
      LAYER M4 ;
        RECT 2.000 0.000 28.000 3.000 ;
      LAYER M3 ;
        RECT 25.125 106.150 27.475 136.480 ;
        *************
        RECT 2.525 106.130 4.845 136.490 ;
      LAYER M2 ;
        RECT 25.125 106.150 27.475 136.480 ;
        ************
        RECT 2.525 106.130 4.845 136.490 ;
    END
  END PAD
  PIN OEN
    DIRECTION INPUT ;
    PORT
      LAYER M5 ;
        RECT 4.500 176.460 6.460 177.000 ;
      LAYER M4 ;
        RECT 4.500 176.460 6.460 177.000 ;
      LAYER M3 ;
        RECT 4.500 176.460 6.460 177.000 ;
      LAYER M2 ;
        RECT 4.500 176.460 6.460 177.000 ;
        RECT 3.910 170.210 5.300 170.330 ;
        RECT 4.680 170.210 4.800 177.000 ;
      LAYER M1 ;
        RECT 4.500 176.460 6.460 177.000 ;
    END
  AntennaGateArea 0.4 ;
  AntennaDiffArea 0.590 ;
  END OEN
  OBS 
      LAYER TM2 ;
        RECT 0.000 0.000 1.290 177.000 ;
        RECT 28.710 0.000 30.000 177.000 ;
        RECT 0.000 3.710 30.000 177.000 ;
      LAYER TV2 ;
        RECT 0.000 0.000 30.000 177.000 ;
      LAYER M7 ;
        RECT 0.000 0.000 1.415 177.000 ;
        RECT 28.585 0.000 30.000 177.000 ;
        RECT 0.000 3.585 30.000 177.000 ;
      LAYER M6 ;
        RECT 0.000 0.000 1.415 177.000 ;
        RECT 28.585 0.000 30.000 177.000 ;
        RECT 0.000 3.585 30.000 177.000 ;
      LAYER TM1 ;
        RECT 0.000 0.000 1.270 177.000 ;
        RECT 28.730 0.000 30.000 177.000 ;
        RECT 0.000 3.730 30.000 177.000 ;
      LAYER M5 ;
        RECT 0.000 0.000 1.415 175.875 ;
      *************
        RECT 29.405 0.000 30.000 177.000 ;
      LAYER M4 ;
        RECT 0.000 0.000 1.415 175.875 ;
       ***************
        RECT 29.405 0.000 30.000 177.000 ;
      LAYER M3 ;
        RECT 1.150 108.100 1.955 136.480 ;
        *****************
        RECT 26.815 0.000 30.000 95.235 ;
      LAYER V6 ;
        RECT 0.000 0.000 30.000 177.000 ;
      LAYER V5 ;
        RECT 0.000 0.000 30.000 177.000 ;
      LAYER V4 ;
        RECT 0.000 0.000 30.000 177.000 ;
      LAYER TV1 ;
        RECT 0.000 0.000 30.000 177.000 ;
      LAYER V3 ;
        RECT 0.000 0.000 30.000 177.000 ;
      LAYER V2 ;
        RECT 0.000 0.000 30.000 177.000 ;
      LAYER M2 ;
        RECT 1.350 170.210 1.630 170.330 ;
        ****************
        RECT 29.405 173.275 30.000 177.000 ;
      LAYER M1 ;
        RECT 0.000 0.000 30.000 175.875 ;
        RECT 0.000 0.000 0.595 177.000 ;
        RECT 3.725 0.000 3.915 177.000 ;
        RECT 7.045 0.000 18.510 177.000 ;
        RECT 21.640 0.000 26.275 177.000 ;
        RECT 29.405 0.000 30.000 177.000 ;
      LAYER V1 ;
        RECT 0.000 0.000 30.000 177.000 ;
  END 
END ABC_GPIO

_Originally posted by @xuyang-stack in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/3526_

maliberty, 2023-06-26:
Please attach a runable test case that shows the problem rather than file snippets.

xuyang-stack, 2023-06-26:
> Sorry, I cannot provide a complete test case. However, I can provide a detailed description of the investigation I have conducted: U_GPIOA_2/PAD and reset_pad are two pins in the reset_pad net. They are located in different layers and have positional differences. If I remove this net and do not perform routing, there will be no congestion .

@@ -811,7 +883,13 @@ void GlobalRouter::initNets(std::vector<Net*>& nets)
   }
 
   for (Net* net : nets) {
+    logger_->report("initNets name {} pin_count {} isLocal {} hasWires {}", net->getName(), net->getNumPins(), net->isLocal() ,net->hasWires());
     int pin_count = net->getNumPins();
     
     **output:
    initNets name reset_pad pin_count 2 isLocal false hasWires false**
    
    
    @@ -2926,7 +3036,12 @@ void GlobalRouter::makeItermPins(Net* net,
           getITermName(iterm),
           getLayerName(max_routing_layer, db_));
     }
-
+    logger_->report("makeItermPins pin name {} isPad {} isMacro {} pin_pos {},{}",
+      getITermName(iterm),
+      connected_to_pad,
+      connected_to_macro,
+      pin_pos.getX(),
+      pin_pos.getY());

@@ -2993,7 +3108,10 @@ void GlobalRouter::makeBtermPins(Net* net,
           "Pin {} does not have geometries in a valid routing layer.",
           pin_name);
     }
-
+    logger_->report("makeBtermPins pin name {} pin_pos {},{}",
+      pin_name,
+      pin_pos.getX(),
+      pin_pos.getY());

     **output:
     makeItermPins pin name U_GPIOA_2/PAD isPad true isMacro false pin_pos 422525,106130
     makeBtermPins pin name reset_pad pin_pos 438700,0**

I would like to understand if OpenROAD supports wire-binding for GPIO modules of this type. Alternatively, how can I prevent OpenROAD from routing the networks within this module? Or is it possible that I have made a deployment mistake?

maliberty, 2023-06-26:
Are these nets between the io driver and the bondpad or the core?

xuyang-stack, 2023-06-27:
> Are these nets between the io driver and the bondpad or the core?
these nets between the io driver and the bondpad

xuyang-stack, 2023-06-27:
touch log and modify code:

[GlobalRouter.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11877717/GlobalRouter.zip)

initNetlist db_net name reset_pad
ITermName U_GPIOA_2/PAD box 422.0,0.0 448.0,3.0   tl 9 ll -1
ITermName U_GPIOA_2/PAD box 422.0,0.0 448.0,3.0   tl 8 ll -1
ITermName U_GPIOA_2/PAD box 422.0,0.0 448.0,3.0   tl 7 ll -1
ITermName U_GPIOA_2/PAD box 422.0,0.0 448.0,3.0   tl 6 ll -1
ITermName U_GPIOA_2/PAD box 422.0,0.0 448.0,3.0   tl 5 ll -1
ITermName U_GPIOA_2/PAD box 422.0,0.0 448.0,3.0   tl 4 ll -1
ITermName U_GPIOA_2/PAD box 445.125,106.15 447.475,136.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.175,106.11 447.445,136.52   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,106.075 447.395,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,106.045 447.38,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 422.655,106.0 447.33,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.995 447.33,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 422.705,105.95 447.28,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.945 447.28,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 422.755,105.9 447.23,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.895 447.23,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 422.805,105.85 447.18,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.845 447.18,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 422.855,105.8 447.13,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.795 447.13,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 422.905,105.75 447.08,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.745 447.08,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 422.955,105.7 447.03,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.695 447.03,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.005,105.65 446.98,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.645 446.98,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.055,105.6 446.93,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.595 446.93,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.105,105.55 446.88,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.545 446.88,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.155,105.5 446.83,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.495 446.83,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.205,105.45 446.78,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.445 446.78,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.255,105.4 446.73,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.395 446.73,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.305,105.35 446.68,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.345 446.68,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.355,105.3 446.63,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.295 446.63,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.405,105.25 446.58,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.245 446.58,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.455,105.2 446.53,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.195 446.53,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.505,105.15 446.48,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.145 446.48,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.555,105.1 446.43,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.095 446.43,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.605,105.05 446.38,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,105.045 446.38,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.655,105.0 446.33,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,104.995 446.33,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.705,104.95 446.28,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,104.945 446.28,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.755,104.915 446.23,106.48   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,0.16 446.23,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,0.125 446.205,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.82,0.1 446.155,3.0   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,0.075 446.155,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.87,0.05 446.105,3.0   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.205,0.025 446.105,136.545   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.92,0.01 446.055,3.0   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.175,106.075 447.395,136.53   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 445.125,106.11 447.445,136.49   tl 3 ll -1
ITermName U_GPIOA_2/PAD box 423.77,102.48 446.23,106.48   tl 3 ll -1
[... 110 lines cut ...]
ITermName U_GPIOA_2/PAD box 436.085,95.82 438.405,136.49   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 429.8,95.82 434.1,106.98   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 431.565,95.82 433.915,136.48   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 431.615,95.82 433.885,136.52   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 431.645,95.82 433.835,136.545   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 431.615,95.82 433.835,136.53   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 431.565,95.82 433.885,136.49   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 427.045,102.48 429.395,136.48   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 427.095,102.48 429.365,136.52   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 427.125,102.48 429.315,136.545   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 423.77,0.15 428.27,100.32   tl 2 ll -1
[... cut ...]
ITermName U_GPIOA_2/PAD box 422.575,106.09 424.795,136.53   tl 2 ll -1
ITermName U_GPIOA_2/PAD box 422.525,106.13 424.845,136.49   tl 2 ll -1
makeItermPins pin name U_GPIOA_2/PAD isPad true isMacro false pin_pos 422525,106130
makeBtermPins pin name reset_pad pin_pos 438700,0
pinAccessPointPositions pin name U_GPIOA_2/PAD isPort false  ap true
pin name U_GPIOA_2/PAD , pos 435750,1050, layer 8
pinAccessPointPositions pin name reset_pad isPort true  ap true
pin name reset_pad , pos 439950,1050, layer 9

vijayank88, 2023-06-27:
@xuyang-stack 
I am also trying to implement pad level connections for `gcd` design sky130hd. Can you share me steps for floorplan to global placement steps?

I am struck at following error:
```
OpenROAD v2.0-8870-g8aabb2ef8 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO GPL-0002] DBU: 1000
[INFO GPL-0003] SiteSize: 460 2720
[INFO GPL-0004] CoreAreaLxLy: 210220 212160
[INFO GPL-0005] CoreAreaUxUy: 1339980 1338240
[WARNING GPL-0001] clk toplevel port is not placed!
       Replace will regard clk is placed in (0, 0)
[WARNING GPL-0001] req_msg[0] toplevel port is not placed!
       Replace will regard req_msg[0] is placed in (0, 0)
\
\
[INFO GPL-0100] worst slack 5.5e-10
[INFO GPL-0103] Weighted 29 nets.
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
[ERROR GRT-0042] Pin clk does not have geometries in a valid routing layer.
Error: global_place.tcl, 51 GRT-0042
```

maliberty, 2023-06-27:
I looked at bp_single which has a padring and I see that the nets connecting pads to IOs are marked as special nets.  That excludes them from global routing consideration.  

It would be helpful to see your FOOTPRINT_TCL if possible.  (or @vijayank88 to see yours)

vijayank88, 2023-06-28:
@maliberty 
I dont have access to bp_single. May I get `FOOTPRINAT` and `SIGMAP_FILE` for reference?

vijayank88, 2023-06-28:
@maliberty You can find the attachment here: https://github.com/The-OpenROAD-Project/OpenROAD/issues/3492#issuecomment-1611237064
Extract the file and find `pad.tcl` in `./designs/sky130hd/gcd/pad.tcl` used as `FOOTPRINT_TCL`.

xuyang-stack, 2023-06-28:
> I looked at bp_single which has a padring and I see that the nets connecting pads to IOs are marked as special nets. That excludes them from global routing consideration.
> 
> It would be helpful to see your FOOTPRINT_TCL if possible. (or @vijayank88 to see yours)

I forgot about the footprint. I'm trying to write a footprint script. This script will set the corresponding pad nets as special, which should solve the current issue. Thank you very much.

vijayank88, 2023-06-28:
@xuyang-stack 
I am trying replicate io pad placement based on this example: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/pad/test/skywater130_coyote_tc.tcl

Seems issue with port mapping with IO pads.

vijayank88, 2023-09-07:
@xuyang-stack 
I am trying to create IO pad based test case. Can you share me `gcd.v` that updated with io pad connection?