The-OpenROAD-Project/OpenROAD-flow-scripts #1185: Unable to build docker images for OS_NAME ubuntu20.04

KingKeshav2002, 2023-06-28:
### Subject

[Build] for compilation-related issues.

### Describe the bug

As default docker image facing issue, reference.https://github.com/The-OpenROAD-Project/OpenROAD/issues/3495

Trying to build docker image for Ubuntu20.04 by following steps: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithDocker.md#build-docker-image-for-different-os

But facing issue while building:

```
#0 1096.7 Get:427 http://archive.ubuntu.com/ubuntu focal/universe amd64 libomp-dev amd64 1:10.0-50~exp1 [2824 B]
#0 1098.0 Get:428 http://archive.ubuntu.com/ubuntu focal/main amd64 python3-magic all 2:0.4.15-3 [9376 B]
#0 1098.0 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/p/publicsuffix/publicsuffix_20200303.0012-1_all.deb  403  connecting to archive.ubuntu.com:80: connecting to 91.189.91.38:80: dial tcp 91.189.91.38:80: connectex: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond. [IP: 185.125.190.39 80]
#0 1098.0 E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
#0 1098.0 Fetched 210 MB in 17min 17s (202 kB/s)
------
Dockerfile.dev:15
--------------------
  14 |
  15 | >>> RUN ./DependencyInstaller.sh \
  16 | >>>       && rm -rf /tmp/installer
  17 |
--------------------
ERROR: failed to solve: process "/bin/sh -c ./DependencyInstaller.sh       && rm -rf /tmp/installer" did not complete successfully: exit code: 100
```
I am able to access this externally http://archive.ubuntu.com/ubuntu/pool/main/p/publicsuffix/publicsuffix_20200303.0012-1_all.deb  

### Expected Behavior

Complete docker based installation

### Environment

```shell
~/orfs_local/OpenROAD-flow-scripts$ ./tools/OpenROAD/etc/Env.sh
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8836-g0ea657420
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
[... 98 lines cut ...]
-- Found Cbc: /opt/or-tools/lib/cmake/Cbc/CbcConfig.cmake (found version "2.10.7")
-- Found Eigen3: /home/keshav/orfs_docker/OpenROAD-flow-scripts/dependencies/share/eigen3/cmake/Eigen3Config.cmake (found version "3.4.0")
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1")
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /home/keshav/orfs_docker/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS)
-- Found Boost: /home/keshav/orfs_docker/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /home/keshav/orfs_docker/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread
-- Found Boost: /home/keshav/orfs_docker/OpenROAD-flow-scripts/dependencies/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.jTLz8WqXyA
```

### To Reproduce

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithDocker.md#build-docker-image-for-different-os

vvbandeira, 2023-06-28:
> #0 1098.0 E: Failed to fetch http://archive.ubuntu.com/ubuntu/pool/main/p/publicsuffix/publicsuffix_20200303.0012-1_all.deb  403  connecting to archive.ubuntu.com:80: connecting to 91.189.91.38:80: [...]

This is a network error with the Ubuntu server. Please try again in a couple of minutes.

KingKeshav2002, 2023-06-29:
With fresh installation and using default docker installation using `build_openroad.sh --threads 1` built successfully and so closing this.

closed 2023-06-29 by KingKeshav2002

vijayank88, 2023-06-29:
> This is a network error with the Ubuntu server. Please try again in a couple of minutes.

Thanks @vvbandeira . Seems resolved now. I am able to build for docker image with Ubuntu22.04 in WSL as well.
@KingKeshav2002 Thanks for the update.