The-OpenROAD-Project/OpenROAD-flow-scripts #1186: WARNING GUI-0076 and hangs gui inside docker

vijayank88, 2023-06-29:
### Subject

[Build] for compilation-related issues.

### Describe the bug

Using ORFS with docker based installation .

Flow completes with `make` command successfully.
```
==========================================================================
finish report_design_area
--------------------------------------------------------------------------
Design area 606 u^2 67% utilization.

[WARNING GUI-0076] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
xkbcommon: ERROR: failed to add default include path /usr/share/X11/xkb
[WARNING GUI-0076] Qt: Failed to create XKB context!
[WARNING GUI-0076] Use QT_XKB_CONFIG_ROOT environmental variable to provide an additional search path, add ':' as separator to provide several search paths and/or make sure that XKB configuration data directory contains recent enough contents, to update please see http://cgit.freedesktop.org/xkeyboard-config/ .
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
Elapsed time: 0:02.38[h:]min:sec. CPU time: user 2.18 sys 0.18 (99%). Peak memory: 153012KB.
cp results/nangate45/gcd/base/5_route.sdc results/nangate45/gcd/base/6_final.sdc
```
If I use `openroad -gui` inside docker prints following warning message and gui hangs infinitely.
```
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4276, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4280, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4284, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4288, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4292, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4296, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4300, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4304, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4308, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4312, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4316, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4320, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] QXcbConnection: XCB error: 2 (BadValue), sequence: 4324, resource id: 1536, major code: 131 (Unknown), minor code: 3
[WARNING GUI-0076] message limit reached, this message will no longer print
```

### Expected Behavior

Resolve gui hang issue, inside the docker.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.90.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
-- The CXX compiler identification is GNU 8.3.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-8/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8836-g0ea657420
-- System name: Linux
-- Compiler: GNU 8.3.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Could NOT find GTest (missing: GTEST_LIBRARY GTEST_INCLUDE_DIR GTEST_MAIN_LIBRARY) 
[... 128 lines cut ...]
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.f7Dlu9wCaY
```

### To Reproduce

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithDocker.md
Launch docker with: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithDocker.md#enable-gui-support
Inside docker
```
source env.sh
openroad -gui
```

### Screenshots

[image]

oharboe, 2023-07-02:
Try latest ORFS and OpenROAD, see https://github.com/The-OpenROAD-Project/OpenROAD/issues/2675

vijayank88, 2024-07-03:
Has docker environment updated, not reproducible.

closed 2024-07-03 by vijayank88