The-OpenROAD-Project/OpenROAD-flow-scripts #1189: Global placement fails for single Rocket core with Sky130 PDK when running in OpenROAD flow scripts

hoanhe, 2023-06-30:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

But here is the error that happened:
OpenROAD  v2.0-8163-g9a5b2cd
Tool crash after I run global place:
[image]

### Expected Behavior

Multi-thread to run the OpenROAD process maybe?
Is the any way to fix global place fail?

### Environment

```shell
#And this is my config file : 
`export DESIGN_NAME = ChipTop
export PLATFORM    = sky130hd

export VERILOG_FILES = ./designs/src/$(DESIGN_NICKNAME)/*.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/ChipTop.sdc

export ADDER_MAP_FILE :=

export CORE_UTILIZATION = 30
export PLACE_DENISTY_LB_ADDON = 0.1
#export PLACE_DENSITY     = 0.51

export chameleon_DIR = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)

export ADDITIONAL_GDS  = $(chameleon_DIR)/gds/sramgen_sram_4096x8m8w8_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_1024x32m8w8_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_64x32m4w32_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_4096x32m8w8_replica_v1.gds \
                        $(chameleon_DIR)/gds/sramgen_sram_512x32m4w8_replica_v1.gds \

export ADDITIONAL_LIBS = $(chameleon_DIR)/lib/sramgen_sram_4096x8m8w8_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_1024x32m8w8_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_64x32m4w32_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_4096x32m8w8_replica_v1_tt_025C_1v80.lib \
                        $(chameleon_DIR)/lib/sramgen_sram_512x32m4w8_replica_v1_tt_025C_1v80.lib

export ADDITIONAL_LEFS  = $(chameleon_DIR)/lef/sramgen_sram_4096x8m8w8_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_1024x32m8w8_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_64x32m4w32_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_4096x32m8w8_replica_v1.lef \
			$(chameleon_DIR)/lef/sramgen_sram_512x32m4w8_replica_v1.lef \
                       $(chameleon_DIR)/lef/sky130_ef_io_core.lef \
                        $(chameleon_DIR)/lef/sky130_ef_io.lef \
                        $(chameleon_DIR)/lef/sky130_ef_sc_hd.lef \
                        $(chameleon_DIR)/lef/sky130_fd_io_core.lef \
                        $(chameleon_DIR)/lef/sky130_fd_io.lef \
                        $(chameleon_DIR)/lef/sky130_fd_pr.lef \
                        $(chameleon_DIR)/lef/sky130_fd_sc_hd.lef \
                        $(chameleon_DIR)/lef/sky130_fd_sc_hvl.lef 

export IO_CONSTRAINTS = ./designs/$(PLATFORM)/$(DESIGN_NICKNAME)/io.tcl
#export MACRO_PLACEMENT = $(chameleon_DIR)/chameleon.macro_placment.cfg
#export DIE_AREA    = 0.0 0.0 4000 2500
#export CORE_AREA   = 50 50 3950 2450

export MACRO_PLACE_CHANNEL  = 80 80
export MACRO_PLACE_HALO = 2 2

export FP_PDN_RAIL_WIDTH  = 0.48
export FP_PDN_RAIL_OFFSET = 0
export TNS_END_PERCENT   ?= 100

export MIN_ROUTING_LAYER = met1
export MAX_ROUTING_LAYER = met5

`
```

### To Reproduce

I using make global_place_issue and this is result: 
https://drive.google.com/drive/folders/13Z-GtKt8aaMdl9Qf-K2K3cunGLmzVkC1?usp=sharing

vijayank88, 2023-06-30:
@hoanhe 
Can you run `./tools/OpenROAD/etc/Env.sh`  inside OpenROAD-flow-scripts and share the log?

vijayank88, 2023-06-30:
@hoanhe 
The attached tar file has `*_cts.odb`. that means flow completed till cts stage.
But you're reporting error during global placement stage.
I am also facing similar issue and got killed.
```
[NesterovSolve] Iter: 400 overflow: 0.520455 HPWL: 41227765648
./run-me-ChipTop-sky130hd-base.sh: line 3:  6006 Killed                  $ORD_GUI/openroad -no_init ./scripts/global_place.tcl
```
@maliberty your thoughts?

maliberty, 2023-07-02:
You are gating your clock with an and gate tied to one_:
[image]

I suspect this is a mistake.  Did you synthesize the netlist with yosys?  Usually we replace such constants in yosys using hilomap.

The design has huge tns coming in and I believe the resizer is going a bit crazy trying to resolve it.  I also see very high fanout nets like:
[image]

which suggests that repair_design wasn't run before global placement (which ORFS normally does).

hoanhe, 2023-07-03:
@maliberty  In the global place step it is true that the design is high fanout. But according to the flow of ORFS, after the glp step is the resizer step. As for the port clock side, I'm trying to fix it the way you suggested.

hoanhe, 2023-07-03:
I run a design that used a black box for all macros and didn't use their liberty file, and it made it to the route step and stopped due to high congestion. At this test run at the global place, it also got a high fanout net that you said but in the resizer step, I see it has been fixed. I also don't know if using the black box is a good practice since it doesn't use the macro's liberty file.

vijayank88, 2023-07-03:
@hoanhe 
How do you generate netlist from yosys or other tools?
After synthsis it is bit confusing with your config.mk and what I see with your global placement test case.
I don't see sram macros it your attached test case.

hoanhe, 2023-07-03:
> @hoanhe How do you generate netlist from yosys or other tools? After synthsis it is bit confusing with your config.mk and what I see with your global placement test case. I don't see sram macros with your attached test case.
I use Yosys to synthesize, the verilog file in config.mk is not the actual verilog file I use. But it seems the verilog file has problems, I'm checking them again.

hoanhe, 2023-07-04:
I double-checked everything and thought it was ok. I run another build using Hilomap, but the result was still the same. Tool still crashes when run with Sram's lib file. Is there any way to fix it? I have updated my latest data here: https://drive.google.com/drive/folders/1KEBXs1Y8xJlXDb8M6Llh-bNfiY83Jn-E?usp=sharing
Please help me fix this issue!

hoanhe, 2023-07-04:
> @hoanhe How much RAM do you have? Currently with given test case I'm able to complete global placement at my end.

88 macro 
I using list lib: 
sramgen_sram_4096x8m8w8_replica_v1_tt_025C_1v80.lib
sramgen_sram_1024x32m8w8_replica_v1_tt_025C_1v80.lib
sramgen_sram_64x32m4w32_replica_v1_tt_025C_1v80.lib
sramgen_sram_4096x32m8w8_replica_v1_tt_025C_1v80.lib
sramgen_sram_512x32m4w8_replica_v1_tt_025C_1v80.lib

vijayank88, 2023-07-04:
Sorry for the confusion. I am asked about your system resource number of Cores/ CPU RAM in your desktop machine.

vijayank88, 2023-07-04:
`export GPL_TIMING_DRIVEN=0` 
Add this variable in your `config.mk` and try. I am able to complete the global placement with that variable.

hoanhe, 2023-07-04:
> `export GPL_TIMING_DRIVEN=0` Add this variable in your `config.mk` and try. I am able to complete the global placement with that variable.

Okay, Let me try.

hoanhe, 2023-07-05:
> `export GPL_TIMING_DRIVEN=0` Add this variable in your `config.mk` and try. I am able to complete the global placement with that variable.
@vijayank88 
I tried the way you said. it passed step 3_3_global_place but on step 3_4_resizer it encountered the same situation as before. it stops at "[INFO RSZ-0058] Using max wire length 2395um." and keeps using Ram until it runs out and the tool crashes again there. I don't know what the real reason is here that leads to the design using so much Ram to place cell. Is there a comprehensive way to fix it?
[image]

hoanhe, 2023-07-05:
This is a bit out of line but I want to ask Is there any way to create placement blockage tye hard and keep_out_margin in openroad?

vijayank88, 2023-07-05:
> This is a bit out of line but I want to ask Is there any way to create placement blockage type hard and keep_out_margin in openroad?
Are you looking for macro blockage? 
`MACRO_PLACE_HALO` `MACRO_PLACE_CHANNEL`?

hoanhe, 2023-07-05:
> > This is a bit out of line but I want to ask Is there any way to create placement blockage type hard and keep_out_margin in openroad?
> 
> Are you looking for macro blockage? `MACRO_PLACE_HALO` `MACRO_PLACE_CHANNEL`?
> 
> How much frequency you are targeted for this design? @hoanhe

I set the frequency to 100 . 
And looking for macro blockage I see soft blockage, but I don't find had blockage for it

vijayank88, 2023-07-07:
Did you applied Matt's suggestion?
Reduce frequency and try. Then tighten the frequency.

hoanhe, 2023-07-07:
@vijayank88 
I replaced such constants in yosys using hilomap flow Matt's suggestion, but it's not ok.

hoanhe, 2023-07-07:
And I think frequency reduction is not a good solution. Because I reduced the frequency from 200 to 100 but nothing changed.

maliberty, 2023-07-10:
Could you upload the input files so I can try running the full flow myself?  I'm curious what the timing is coming out of yosys as things seem very off.

hoanhe, 2023-07-12:
you can use the verilog file and sdc file results from my step 1 yosys to run ko, the files to run yosys and sdc are many so i can't update them to the google for you. @maliberty

hoanhe, 2023-07-12:
[input_file.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12023476/input_file.tar.gz)
I upload the input files that I use and along with these I use the built-in PDK of sky130hd. Please help me try to run it. Thank you very much. @maliberty

vijayank88, 2023-07-12:
@hoanhe 
All the files in `.sv` extension, but in `config.mk` you're called only `*.v`.
Are you sure yosys synthesized this files?

hoanhe, 2023-07-12:
Yes, it is true that I use the .sv file
I updated the verilog list of sram in here
[input_file.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12024518/input_file.tar.gz)

hoanhe, 2023-07-12:
This is the file I use yosys to synthesized
`read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ALU.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AMOALU.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AXI4IdIndexer.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AXI4UserYanker.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/Arbiter.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueue.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSink.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSink_1.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSink_2.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSink_3.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSource.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSource_1.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSource_2.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncQueueSource_3.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncResetRegVec_w1_i0.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncResetRegVec_w1_i1.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncResetRegVec_w2_i0.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncResetSynchronizerPrimitiveShiftReg_d3_i0.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncResetSynchronizerShiftReg_w1_d3_i0.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncResetSynchronizerShiftReg_w1_d3_i0_1.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/AsyncValidSync.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/Atomics.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/BTB.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/BankBinder.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/BankedStore.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/BreakpointUnit.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/BundleBridgeNexus_15.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CLINT.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CSRFile.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CaptureChain.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CaptureUpdateChain.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CaptureUpdateChain_1.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CaptureUpdateChain_2.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ChipTop.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockCrossingReg_w15.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockCrossingReg_w43.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockCrossingReg_w55.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockDividerN.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockGroupAggregator_6.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockGroupParameterModifier.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockGroupParameterModifier_1.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockGroupResetSynchronizer.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockSinkDomain.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockSinkDomain_1.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockSinkDomain_2.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockSinkDomain_3.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/ClockSinkDomain_4.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CoherenceManagerWrapper.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/CompareRecFN.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/DCache.sv
read_verilog -sv designs/TestHarness/chipyard.TestHarness.RocketConfig/gen-collateral/DCacheDataArray.sv
[... cut ...]
abc -D 20000 -constr ./designs/TestHarness/abc/abc.constr -liberty ./designs/TestHarness/sky130A/libs.ref/sky130_fd_sc_hd/lib/sky130_fd_sc_hd__tt_025C_1v80.lib -showtmp

setundef -zero

splitnets

opt_clean -purge

hilomap -singleton -hicell sky130_fd_sc_hd__conb_1 HI -locell sky130_fd_sc_hd__conb_1 LO

insbuf -buf sky130_fd_sc_hd__buf_4 A X

tee -o ./designs/TestHarness/reports/check.rpt check

tee -o ./designs/TestHarness/reports/stat.rpt stat

write_verilog -noattr -noexpr -nohex -nodec ./designs/TestHarness/results/ChipTop_yosys_0707.v

`

vijayank88, 2023-07-12:
@hoanhe 
With attached input files, i just ran the flow. Seems some files are missing
```
410. Executing SYNTH pass.
410.1. Executing HIERARCHY pass (managing design hierarchy).
410.2. Executing AST frontend in derive mode using pre-parsed AST for module `\ChipTop'.
410.2.1. Analyzing design hierarchy..
ERROR: Module `\GenericDigitalOutIOCell' referenced in module `\ChipTop' in cell `\iocell_uart_0_txd' is not part of the design.
Elapsed time: 0:02.20[h:]min:sec. CPU time: user 2.03 sys 0.16 (99%). Peak memory: 169516KB.
make: *** [Makefile:406: results/sky130hd/ChipTop/base/1_1_yosys.v] Error 1
```

vijayank88, 2023-07-12:
I am running the flow now. Yosys running, lets compare both the netlist once it completes.

vijayank88, 2023-07-12:
@maliberty 
you can copy files from GCP path
```
/home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/flow/designs/sky130hd/ChipTop
[vijayan@chipware-vijayan ChipTop]$ 
```
Seems its get stuck during resizer
```
Perform port buffering...
[INFO RSZ-0027] Inserted 122 input buffers.
[INFO RSZ-0028] Inserted 183 output buffers.
Perform buffer insertion...
[INFO RSZ-0058] Using max wire length 2406um.
```

vijayank88, 2023-07-20:
I think resizer unable to resolve this: fanout of 38301
```
Startpoint: clock_clock (clock source 'clk')
Endpoint: serial_tl_clock (output port clocked by clk)
Path Group: clk
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.00    0.00    0.00   clock clk (rise edge)
                  0.00    0.00    0.00 ^ clock_clock (in)
     1    0.00                           clock_clock (net)
                  0.00    0.00    0.00 ^ iocell_clock_clock/_1_/A (sky130_fd_sc_hd__nand2_1)
                  0.03    0.03    0.03 v iocell_clock_clock/_1_/Y (sky130_fd_sc_hd__nand2_1)
     1    0.00                           iocell_clock_clock/_0_ (net)
                  0.03    0.00    0.03 v iocell_clock_clock/i\\sky130_fd_sc_hd__inv_1/A (sky130_fd_sc_hd__inv_1)
                944.37  656.49  656.51 ^ iocell_clock_clock/i\\sky130_fd_sc_hd__inv_1/Y (sky130_fd_sc_hd__inv_1)
 38031  117.33                           axi4_mem_0_clock (net)
                944.37    0.00  656.51 ^ system/tsi_domain/clock\\sky130_fd_sc_hd__buf_4/A (sky130_fd_sc_hd__buf_4)
                 15.97   45.54  702.06 ^ system/tsi_domain/clock\\sky130_fd_sc_hd__buf_4/X (sky130_fd_sc_hd__buf_4)
     1    0.00                           serial_tl_clock (net)
                 15.97    0.00  702.06 ^ serial_tl_clock (out)
                                702.06   data arrival time

                  0.00   50.00   50.00   clock clk (rise edge)
                          0.00   50.00   clock network delay (ideal)
                          0.00   50.00   clock reconvergence pessimism
                        -10.00   40.00   output external delay
                                 40.00   data required time
-----------------------------------------------------------------------------
                                 40.00   data required time
                               -702.06   data arrival time
-----------------------------------------------------------------------------
                               -662.06   slack (VIOLATED)
```

hoanhe, 2023-07-21:
@vijayank88 
How to solve high fanout reduction in this design from the previous step? Like set max_fanout for example?

hoanhe, 2023-07-21:
Should I set_max_fanout in sdc file from the beginning?
Example:
`set_max_fanout 12 [current_design]`

hoanhe, 2023-07-21:
Hi @vijayank88 
I tried running design with set_max_fanout = 10 in the .sdc file.
Below is the path wns before the resizer. Fanout is 308 but openroad can't get past the resizer step either. I think high fanout is not the reason why the design doesn't work. What do you think about this?
```
==========================================================================
resizer pre report_checks -path_delay max
--------------------------------------------------------------------------
Startpoint: _407814_ (rising edge-triggered flip-flop clocked by clock_clock)
Endpoint: _424341_ (rising edge-triggered flip-flop clocked by clock_clock)
Path Group: clock_clock
Path Type: max

Fanout     Cap    Slew   Delay    Time   Description
-----------------------------------------------------------------------------
                  0.00    0.00    0.00   clock clock_clock (rise edge)
                          0.00    0.00   clock network delay (ideal)
                  0.00    0.00    0.00 ^ _407814_/CLK (sky130_fd_sc_hd__dfxtp_1)
                 26.12   16.88   16.88 ^ _407814_/Q (sky130_fd_sc_hd__dfxtp_1)
   308    2.82                           system.tile_prci_domain.tile_reset_domain_tile.frontend.btb.io_req_bits_addr[12] (net)
                 26.43    2.30   19.18 ^ _246607_/S0 (sky130_fd_sc_hd__mux4_2)
                  0.51    2.04   21.22 v _246607_/X (sky130_fd_sc_hd__mux4_2)
     3    0.02                           _036737_ (net)
                  0.51    0.00   21.22 v _246608_/C (sky130_fd_sc_hd__nand3_1)
                  2.12    1.58   22.80 ^ _246608_/Y (sky130_fd_sc_hd__nand3_1)
    23    0.21                           _036738_ (net)
                  2.12    0.01   22.81 ^ _247090_/A1 (sky130_fd_sc_hd__o21ai_0)
                  0.49    0.65   23.46 v _247090_/Y (sky130_fd_sc_hd__o21ai_0)
     1    0.02                           _037220_ (net)
                  0.49    0.00   23.46 v _247091_/C1 (sky130_fd_sc_hd__a211oi_1)
                  1.38    1.22   24.68 ^ _247091_/Y (sky130_fd_sc_hd__a211oi_1)
     1    0.04                           _037221_ (net)
                  1.38    0.01   24.69 ^ _247095_/C (sky130_fd_sc_hd__nand4_1)
                  0.62    0.70   25.39 v _247095_/Y (sky130_fd_sc_hd__nand4_1)
     1    0.04                           _037225_ (net)
                  0.62    0.01   25.39 v _247096_/C1 (sky130_fd_sc_hd__a221oi_1)
                 72.75   43.77   69.16 ^ _247096_/Y (sky130_fd_sc_hd__a221oi_1)
     5    2.44                           _037226_ (net)
                 79.69   17.05   86.21 ^ _247528_/B (sky130_fd_sc_hd__xor2_1)
                  6.71    0.55   86.76 ^ _247528_/X (sky130_fd_sc_hd__xor2_1)
     1    0.02                           _037658_ (net)
                  6.71    0.00   86.76 ^ _247535_/B (sky130_fd_sc_hd__nand4_1)
                  3.34    0.37   87.14 v _247535_/Y (sky130_fd_sc_hd__nand4_1)
     1    0.00                           _037665_ (net)
                  3.34    0.00   87.14 v _247542_/A (sky130_fd_sc_hd__or4_1)
                  0.26    1.36   88.50 v _247542_/X (sky130_fd_sc_hd__or4_1)
     1    0.00                           _037672_ (net)
                  0.26    0.00   88.50 v _247554_/A (sky130_fd_sc_hd__nor4_1)
                 80.48   52.80  141.30 ^ _247554_/Y (sky130_fd_sc_hd__nor4_1)
     3    2.15                           _037684_ (net)
                 80.48    0.18  141.48 ^ _363289_/B1 (sky130_fd_sc_hd__a22oi_1)
                 11.70  441.28  582.76 v _363289_/Y (sky130_fd_sc_hd__a22oi_1)
     1    2.13                           _134495_ (net)
                 28.03   14.89  597.66 v _363291_/A (sky130_fd_sc_hd__nand2_1)
                  4.66    6.29  603.95 ^ _363291_/Y (sky130_fd_sc_hd__nand2_1)
     1    0.01                           _134497_ (net)
                  4.66    0.00  603.95 ^ _363292_/A1 (sky130_fd_sc_hd__mux2_2)
                  0.19    0.55  604.50 ^ _363292_/X (sky130_fd_sc_hd__mux2_2)
     1    0.00                           _022410_ (net)
                  0.19    0.00  604.50 ^ _424341_/D (sky130_fd_sc_hd__dfxtp_1)
                                604.50   data arrival time

                  0.00   10.00   10.00   clock clock_clock (rise edge)
                          0.00   10.00   clock network delay (ideal)
                         -0.40    9.60   inter-clock uncertainty
                          0.00    9.60   clock reconvergence pessimism
                                  9.60 ^ _424341_/CLK (sky130_fd_sc_hd__dfxtp_1)
                         -0.09    9.51   library setup time
                                  9.51   data required time
-----------------------------------------------------------------------------
                                  9.51   data required time
                               -604.50   data arrival time
-----------------------------------------------------------------------------
                               -594.99   slack (VIOLATED)

```

vijayank88, 2023-07-21:
@hoanhe Attached yosys generated netlist with abc optimized. 
[synth.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12121821/synth.zip)
Just compare with your netlist is same or different?

hoanhe, 2023-07-21:
@vijayank88 
I think our netlist is different. This is my netlist after using the new sdc file generated by yosys.
[1_synth.zip.zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12121960/1_synth.zip.zip)
This is the sdc file I use
```
###############################################################################
# Created by write_sdc
# Mon Jul  3 06:32:13 2023
###############################################################################
current_design ChipTop
###############################################################################
# Timing Constraints
###############################################################################
set_max_fanout 10 [current_design]
set_max_transition 0.5 [current_design]
create_clock -name clock_clock -period 10.0000 [get_ports {clock_clock}]
set_clock_uncertainty -rise_from [get_clocks {clock_clock}] -rise_to [get_clocks {clock_clock}]  -setup 0.4000
set_clock_uncertainty -rise_from [get_clocks {clock_clock}] -fall_to [get_clocks {clock_clock}]  -setup 0.4000
set_clock_uncertainty -fall_from [get_clocks {clock_clock}] -rise_to [get_clocks {clock_clock}]  -setup 0.4000
set_clock_uncertainty -fall_from [get_clocks {clock_clock}] -fall_to [get_clocks {clock_clock}]  -setup 0.4000
###############################################################################
# Environment
###############################################################################
###############################################################################
# Design Rules
###############################################################################
```

hoanhe, 2023-08-07:
Hi @maliberty @vijayank88 
I know you are very busy, but do you have any other solutions for running this design?

maliberty, 2023-08-07:
Sorry - I'll try to get back to this.

maliberty, 2023-08-07:
I just re-ran it with the head of OR (828e699eed090aad73fc5787dda088ec75c1001f) and it no longer fails in global placement.  Please update and try again.

hoanhe, 2023-08-08:
Can you please share the environment you use with me? I tried running it but it doesn't work. And I don't understand what is the head of OR (828e699eed090aad73fc5787dda088ec75c1001f)?

maliberty, 2023-08-08:
Do you use the build.sh script to setup your environment?  If so use --latest to get the most current code.

hoanhe, 2023-08-10:
Hi @maliberty, 
I build env using latest build.sh script version as per your instructions but it still same problem. I use OR version v2.0-9709-g828e699. Can you share your setup config file with sdc file for me?

maliberty, 2023-08-18:
I tried with the build you mentioned and it does fail for me as well. However the current head of master (01e6343a4590283d07e34289ad056aded601b82f) works fine.

maliberty, 2023-12-18:
Please reopen if you still are having problems

closed 2023-12-18 by maliberty