The-OpenROAD-Project/OpenROAD-flow-scripts #1190: Failed at global placement stage 3_3_place_gp.log

vishallparmar25, 2023-06-30:
### Subject

[Stage]: Global Placement.

### Describe the bug

[NesterovSolve] Iter: 310 overflow: 0.813046 HPWL: 5619862180
Command terminated by signal 11
Elapsed time: 40:17.33[h:]min:sec. CPU time: user 2400.71 sys 16.37 (99%). Peak memory: 8148768KB.

Global placement failed with above error, run is not moving forward. 

below is config of that design 
[image]
 

### Expected Behavior

[NesterovSolve] Finished with Overflow: 0.099319

### Environment

```shell
ubuntu@ip-172-31-10-38:~/Opensource/OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh 
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-1027-aws
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8118-gb95ce9901
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")  
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
[... 300 lines cut ...]
[NesterovSolve] Iter: 100 overflow: 0.994539 HPWL: 556015113
[NesterovSolve] Iter: 110 overflow: 0.994309 HPWL: 570179252
[NesterovSolve] Iter: 120 overflow: 0.994137 HPWL: 588302380
[NesterovSolve] Iter: 130 overflow: 0.993935 HPWL: 611556924
[NesterovSolve] Iter: 140 overflow: 0.993599 HPWL: 640819791
[NesterovSolve] Iter: 150 overflow: 0.99318 HPWL: 676510357
[NesterovSolve] Iter: 160 overflow: 0.992646 HPWL: 720452630
[NesterovSolve] Iter: 170 overflow: 0.991751 HPWL: 776133542
[NesterovSolve] Iter: 180 overflow: 0.990317 HPWL: 847105729
[NesterovSolve] Iter: 190 overflow: 0.988058 HPWL: 938727825
[NesterovSolve] Iter: 200 overflow: 0.984748 HPWL: 1055102476
[NesterovSolve] Iter: 210 overflow: 0.98016 HPWL: 1203829580
[NesterovSolve] Iter: 220 overflow: 0.973896 HPWL: 1396138753
[NesterovSolve] Iter: 230 overflow: 0.965628 HPWL: 1647865896
[NesterovSolve] Iter: 240 overflow: 0.954789 HPWL: 1982810776
[NesterovSolve] Iter: 250 overflow: 0.937343 HPWL: 2857919803
[NesterovSolve] Iter: 260 overflow: 0.912159 HPWL: 4391131586
[NesterovSolve] Iter: 270 overflow: 0.891764 HPWL: 5447196435
[NesterovSolve] Iter: 280 overflow: 0.872312 HPWL: 6135796265
[NesterovSolve] Iter: 290 overflow: 0.853829 HPWL: 6444647485
[NesterovSolve] Iter: 300 overflow: 0.839114 HPWL: 6001237274
[NesterovSolve] Iter: 310 overflow: 0.813046 HPWL: 5619862180
Command terminated by signal 11
Elapsed time: 40:17.33[h:]min:sec. CPU time: user 2400.71 sys 16.37 (99%). Peak memory: 8148768KB.
```

vijayank88, 2023-06-30:
`riscv32i` is built in design. Did you changed anything in the source code?

Attach a reproducible test case by running `make global_place_issue` and attach tar.gz here

vishallparmar25, 2023-06-30:
> `riscv32i` is built in design. Did you changed anything in the source code?
> 
> Attach a reproducible test case by running `make global_place_issue` and attach tar.gz here

This is not riscv32i design, it's different design. 
Entire tar file sharing is not possible due to it's large size around 2GB of tar file.

Any specific file you are looking ?

vijayank88, 2023-06-30:
share `*.v`, `.sdc` and `config.mk` in zip file

vijayank88, 2023-06-30:
Attach following files
```
3_2_place_iop.odb 
2_floorplan.sdc
```

vijayank88, 2023-07-04:
@vishallparmar25 
Any update on this?

vishallparmar25, 2023-07-04:
> @vishallparmar25 Any update on this?

It's giving same issue.

vijayank88, 2023-07-04:
> Attach following files
> 
> ```
> 3_2_place_iop.odb 
> 2_floorplan.sdc
> ```

Can you attach these files?

vijayank88, 2023-08-31:
@vishallparmar25 
What is the status on this?