The-OpenROAD-Project/OpenROAD-flow-scripts #1191: Failed at detailed routing stage 5_2_TritonRoute.log

vishallparmar25, 2023-06-30:
### Subject

[Stage]: Detail Router.

### Describe the bug

ORFS flow failed at detail route stage

[INFO DRT-0036] Pad guide region query size = 0.
[INFO DRT-0179] Init gr pin query.
[INFO DRT-0185] Post process initialize RPin region query.
[INFO DRT-0181] Start track assignment.
[INFO DRT-0184] Done with 726268 vertical wires in 15 frboxes and 499275 horizontal wires in 15 frboxes.
[INFO DRT-0186] Done with 55361 vertical wires in 15 frboxes and 81996 horizontal wires in 15 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:03:26, elapsed time = 00:01:00, memory = 5321.84 (MB), peak = 6472.33 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:01, memory = 5321.84 (MB), peak = 6472.33 (MB)
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
Command terminated by signal 9
Elapsed time: 3:26.31[h:]min:sec. CPU time: user 681.69 sys 21.72 (340%). Peak memory: 15534688KB.

### Expected Behavior

[INFO DRT-0198] Complete detail routing.
Total wire length = 127688 um.
Total wire length on LAYER M1 = 0 um.

### Environment

```shell
ubuntu@ip-172-31-10-38:~/Opensource/OpenROAD-flow-scripts/tools/OpenROAD$ ./etc/Env.sh 
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-1027-aws
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8118-gb95ce9901
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")  
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
[... 555 lines cut ...]
[INFO DRT-0186] Done with 55361 vertical wires in 15 frboxes and 81996 horizontal wires in 15 frboxes.
[INFO DRT-0182] Complete track assignment.
[INFO DRT-0267] cpu time = 00:03:26, elapsed time = 00:01:00, memory = 5321.84 (MB), peak = 6472.33 (MB)
[INFO DRT-0187] Start routing data preparation.
[INFO DRT-0267] cpu time = 00:00:01, elapsed time = 00:00:01, memory = 5321.84 (MB), peak = 6472.33 (MB)
[INFO DRT-0194] Start detail routing.
[INFO DRT-0195] Start 0th optimization iteration.
Command terminated by signal 9
Elapsed time: 3:26.31[h:]min:sec. CPU time: user 681.69 sys 21.72 (340%). Peak memory: 15534688KB.
```

### Screenshots

config.mk 

[image]

Instance Count
[image]

Placement Density heat map
[image]

Routing congestion heat map
[image]

vijayank88, 2023-06-30:
@vishallparmar25 
How much RAM your system has?

vijayank88, 2023-06-30:
Attach reproducible test case by running `make detail_route_issue` and upload here.

vishallparmar25, 2023-06-30:
> @vishallparmar25 How much RAM your system has?

16 GB RAM

vijayank88, 2023-06-30:
So it is RAM issue.
```
Command terminated by signal 9
Elapsed time: 3:26.31[h:]min:sec. CPU time: user 681.69 sys 21.72 (340%). Peak memory: 15534688KB.
```
Once it is reached max RAM (15.5GB) it is got killed. Increase swap memory and try again

vishallparmar25, 2023-06-30:
> Attach reproducible test case by running `make detail_route_issue` and upload here.

It's not possible to share entire tar file here it's large in size. 
Any specific file you are looking from that tar ?

vishallparmar25, 2023-06-30:
> So it is RAM issue.
> 
> ```
> Command terminated by signal 9
> Elapsed time: 3:26.31[h:]min:sec. CPU time: user 681.69 sys 21.72 (340%). Peak memory: 15534688KB.
> ```
> 
> Once it is reached max RAM (15.5GB) it is got killed. Increase swap memory and try again

Can you please share steps to increase swap memory, or any link. How much it requires additional memory ?

rovinski, 2023-06-30:
@vishallparmar25 You should try reducing the number of threads before increasing your swap size. The memory used is proportional to the number of threads in DRT.

You can do so by running `make NUM_CORES=x` where `x` is a positive integer.

vijayank88, 2023-07-04:
@vishallparmar25 
Are you able to complete the routing?

vishallparmar25, 2023-07-04:
> @vishallparmar25 Are you able to complete the routing?

No, it's giving same issue, even after increasing 4 Gb of swap memory.

vijayank88, 2023-07-04:
Can you attach following files?
```
5_1_grt.odb
4_cts.sdc
```
I will update you how much RAM its required to complete routing.

vishallparmar25, 2023-07-04:
> @vishallparmar25 Are you able to complete the routing?

Fired one run with 32Gb swap memory + 16 Gb of RAM. Once it will completed i will update you.

rovinski, 2023-07-05:
@vishallparmar25 did you try running with fewer threads yet? Increasing swap is a much worse and much slower solution.

vishallparmar25, 2023-07-05:
> @vishallparmar25 did you try running with fewer threads yet? Increasing swap is a much worse and much slower solution.
Yes, In that case it also failing.

vijayank88, 2023-07-20:
@vishallparmar25 are you working on confidential design?
Still not shared any test case, hence it is difficult to provide the solution.

vishallparmar25, 2023-07-20:
> @vishallparmar25 are you working on confidential design? Still not shared any test case, hence it is difficult to provide the solution.

Yes, i am working on confidential design, that's why.

By the way similar kind of issues i am facing in few more design. only similarity is their cell count are bit higher like 1M -3M cell count.
Does there any upper limit for instance count or cell count for OpenROAD-flow-script ?

vijayank88, 2023-07-20:
No limitation for instance count.
Try with better hardware or try with cloud resources for short-term to complete routing.

vishallparmar25, 2023-07-20:
> No limitation for instance count. Try with better hardware or try with cloud resources for short-term to complete routing.
Thanks for let me know about no upper limit for cell count or instance count.
I am running design with cloud resources only. Below are my cloud configurations.

[image]

[image]

vijayank88, 2023-07-20:
For 3M instance count, the above hardware is not enough as per my knowledge.
One hint you can check during detailed routing how much memory it is taking for routing, if it is getting `signal  9` or `child killed` you have to increase your RAM. Detailed route log format shown below.
```
[INFO DRT-0267] cpu time = 00:07:41, elapsed time = 00:00:28, memory = 1459.91 (MB), peak = 2071.53 (MB)
```
So check `Peak Memory` for your design routing and when it fails and what its peak memory.

rovinski, 2023-07-20:
Yes, 16 GB is no where near enough for a 1-3M inst design. I would estimate somewhere between 48-96 GB minimum.

vishallparmar25, 2023-07-21:
> Yes, 16 GB is no where near enough for a 1-3M inst design. I would estimate somewhere between 48-96 GB minimum.

Can you give me minimum system requirement (Number of Processor, RAM, ROM, etc.)  for 3M instance count design.

vijayank88, 2023-07-21:
Currently our CI runs with 32 core/128GB RAM. Maybe try with same configuration.

rovinski, 2023-07-21:
> Currently our CI runs with 32 core/128GB RAM. Maybe try with same configuration.

However, I also don't think that we run with any 3M instance designs in the CI.

> Can you give me minimum system requirement (Number of Processor, RAM, ROM, etc.) for 3M instance count design.

I can't give your requirements, but I can give estimates. It depends on your specific design. We have loose estimates for scaling, which I have given based on a 1M instance design. A 3M instance design would probably be closer to 144-256 GB of memory.

It is usually not recommended to run designs that large as a single block. You should look into partitioning your design into smaller, more manageable block sizes.

closed 2026-04-16 by github-actions[bot]