The-OpenROAD-Project/OpenROAD-flow-scripts #1199: crash on non existing pin name in set_io_pin_constraint

gkamendje, 2023-07-05:
### Subject

[Stage]: I/O Placement.

### Describe the bug

During floorplaning while running `tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/io_placement_random.tcl ...` while the file io_constraints.tcl contains a pin which is not in the design something like  
`set_io_pin_constraint -pin_names {i_rst_n i_sys_clk my_non_existing_pin } -region left:*` the following error is reported 
```
Signal 11 received
Stack trace:
 0# 0x0000560496055BEB in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F0503F21520 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::ConcretePort::isBus() const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# sta::ConcreteNetwork::isBus(sta::Port const*) const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# sta::NetworkNameAdapter::isBus(sta::Port const*) const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# find_ports_matching(char const*, bool, bool) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# 0x000056049636FE60 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# 0x00007F0507DBA028 in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
11# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# 0x000056049605693B in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# ord::tclAppInit(Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# main in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# 0x00007F0503F08D90 in /lib/x86_64-linux-gnu/libc.so.6
17# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
18# _start in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 11
Elapsed time: 0:05.29[h:]min:sec. CPU time: user 5.19 sys 0.10 (100%). Peak memory: 280096KB.
make: *** [Makefile:474: results/smic018/ltag7/base/2_2_floorplan_io.odb] Error 139
```

### Expected Behavior

The tool should report that the pin could not be found in the design

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.17.3
-- The CXX compiler identification is GNU 11.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9036-g12e051bf4
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 11.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
[... 131 lines cut ...]
Signal 11 received
Stack trace:
 0# 0x0000560496055BEB in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 1# 0x00007F0503F21520 in /lib/x86_64-linux-gnu/libc.so.6
 2# sta::ConcretePort::isBus() const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 3# sta::ConcreteNetwork::isBus(sta::Port const*) const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 4# sta::NetworkNameAdapter::isBus(sta::Port const*) const in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 5# find_ports_matching(char const*, bool, bool) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 6# 0x000056049636FE60 in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
 7# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# 0x00007F0507DBA028 in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
10# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
11# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
12# 0x000056049605693B in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
13# ord::tclAppInit(Tcl_Interp*) in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
14# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
15# main in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
16# 0x00007F0503F08D90 in /lib/x86_64-linux-gnu/libc.so.6
17# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
18# _start in /home/gkamendje/tmp_compile/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
Command terminated by signal 11
Elapsed time: 0:05.29[h:]min:sec. CPU time: user 5.19 sys 0.10 (100%). Peak memory: 280096KB.
make: *** [Makefile:474: results/smic018/ltag7/base/2_2_floorplan_io.odb] Error 139
```

vijayank88, 2023-07-06:
@gkamendje 
Is that built-in example or your own design?
If own design mean  run `make io_placement_random_issue` and share a test case.

gkamendje, 2023-07-06:
@vijayank88 I was able to reproduce the issue with the nangate45/gcd design. Essentially any `set_io_constraint` statement with a pin name with an out of bound array index will reproduce the issue.  For example `set_io_pin_constraint -pin_names {req_msg[32]   } -region left:*` where req_msg is defined as `req_msg[31:0]` will cause the crash. Attaching a test case to the this reply
[io_placement_random_gcd_nangate45_base_2023-07-06_09-55.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/11968967/io_placement_random_gcd_nangate45_base_2023-07-06_09-55.tar.gz)

eder-matheus, 2023-07-07:
@maliberty @openroadie I've found that the crash happens in the OpenSTA code. I've created a fork of the OpenSTA repo with a possible fix for it: https://github.com/eder-matheus/OpenSTA/commit/798175901ea93727091fc048e57010e7392f5a96.

How do we proceed with PRs to OpenSTA? Should I just create one from my fork or should I leave this fix for @openroadie ?

maliberty, 2023-07-07:
Leave it to @openroadie

closed 2023-07-10 by openroadie