The-OpenROAD-Project/OpenROAD-flow-scripts #1200: make mock-array 64 bit data path work with routing by abutment

oharboe, 2023-07-05:
### Description

Update to latest master, to get this performance improvement https://github.com/The-OpenROAD-Project/OpenROAD/pull/3567

Create a `flow/settings.mk`:

```
export DESIGN_CONFIG=designs/asap7/mock-array/config.mk
export MOCK_ARRAY_TABLE         = 8 8 20 20 20 22
export MOCK_ARRAY_DATAWIDTH     = 64
export MACRO_BLOCKAGE_HALO      = 0
```

Create Verilog:

```
make verilog
```

Run build:

```
rm -rf results/ logs/ reports && make
```

Output:

```
[deleted]
cts post-repair report_design_area
--------------------------------------------------------------------------
Design area 119916 u^2 88% utilization.

Placement Analysis
---------------------------------
total displacement       1628.7 u
average displacement        0.2 u
max displacement          168.5 u
original HPWL           62649.8 u
legalized HPWL          63723.9 u
delta HPWL                    2 %

Repair setup and hold violations...
TNS end percent 100
[INFO RSZ-0046] Found 792 endpoints with hold violations.
[INFO RSZ-0032] Inserted 10350 hold buffers.
[deleted]

```

A draft pull request where I tried to make 64 bit datapath default #1198

oharboe, 2023-07-05:
@osamahammad21 FYI, I tried to use your latest detailed routing fixes, but I have run into a CTS snag...

oharboe, 2023-07-05:
I tried moving beyond the cts problems with `make skip_cts`, then I got:

[image]

[image]

[image]

[image]

oharboe, 2023-07-05:
Adding to `settings.mk`, `export GLOBAL_ROUTE_ARGS=-congestion_iterations 100 -verbose`, global routing passes.

oharboe, 2023-07-05:
Routing by abutment:

[image]

oharboe, 2023-07-08:
@rovinski Suggested trying `dont_touch` in https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1206#issuecomment-1626325795

However, what should I set `dont_touch` on? CTS never completes, so it doesn't give me any clues as to what it is that I should set `dont_touch` on...

rovinski, 2023-07-10:
@oharboe you need to set `dont_touch` on all nets which have abutted pins. The point is that you are telling the optimizer that it is not allowed to buffer those nets because there isn't a valid way to insert them.

The point of abutting blocks at the top level is so that you can fix design / timing issues at the block level. If you are failing hold timing at the top level, that means you have to adjust your constraints at the block level so that timing is met at the top level.

oharboe, 2023-07-10:
@rovinski Like so? https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1220

Seems to work...

closed 2023-07-11 by oharboe