The-OpenROAD-Project/OpenROAD-flow-scripts #1201: Current Version of OpenROAD 6_final.log does not have power numbers

dalyles, 2023-07-05:
### Description

In some previous versions of OpenROAD flow scripts, the 6_final.log file had power numbers like this example for <code>gcd</code> for <code>nangate45</code>:

[image]

The current version of OpenROAD flow scripts shows this instead, skipping the power measurements entirely.
[image]

### Suggested Solution

I'm requesting to bring back the power measurements in 6_final.log as this is very critical information for our applications.
Please let me know when this is included again so I can use the current version of OpenROAD flow scripts.

Thank you.

maliberty, 2023-07-05:
I recall a while ago there was a crash, perhaps it never got turned back on

vijayank88, 2023-07-06:
@vvbandeira 
Restore these lines is enough right? https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/commit/6ea30ef768e788300d8c2d818bb40dd588659c87

dalyles, 2023-07-06:
Hi @vvbandeira,

Restoring those lines worked just fine.
I'm able to see the power numbers.

[image]

Thanks!

closed 2023-07-06 by vvbandeira