The-OpenROAD-Project/OpenROAD-flow-scripts #1213: Macro Placement Gets Stuck

BaranErdo, 2023-07-09:
### Subject

[Stage]: Floorplaning.

### Describe the bug

I was running our design with OpenROAD while including the sram macros however it gets stuck at 2_4_mplace part. I was able to succesfully run the flow without including macros. 

### Expected Behavior

I was expecting the run the complete the macro placement at the floorplanning without a problem. 2_3_tdms part was succesfull. 

### Environment

```shell
Please pull the latest changes and try again, if problem persists file a github issue with the re-producible test case.

I actually did git pull after this but some how couldn't get rid of this message.
```

### To Reproduce

I used make macro_place_issue and this is result:
https://drive.google.com/file/d/1rML7XaY-JsGE6NaoLcB7Ux5l10j0rsdl/view?usp=drive_link

vijayank88, 2023-07-12:
@BaranErdo 
Update ORFS to latest commit and run the flow.
Then run `make macro_place_issue` and attach a test case to reproduce the same along with log.
Your current test case has wrong executable paths.

vijayank88, 2023-07-12:
@BaranErdo 
you're using 8 months old flow. `OpenROAD v2.0-5852-g2a9ef5fa1`
Please update the flow and try again.

vijayank88, 2023-07-13:
Still executable fails.
```
./run-me-core_v_mini_mcu-sky130hd-base.sh 
vars-core_v_mini_mcu-sky130hd-base.sh: line 1: export: `1': not a valid identifier
vars-core_v_mini_mcu-sky130hd-base.sh: line 1: export: `-exit': not a valid identifier
vars-core_v_mini_mcu-sky130hd-base.sh: line 1: export: `-no_init': not a valid identifier
vars-core_v_mini_mcu-sky130hd-base.sh: line 1: export: `2_2_floorplan_io.odb': not a valid identifier
vars-core_v_mini_mcu-sky130hd-base.sh: line 1: export: `2_3_floorplan_tdms.odb': not a valid identifier
OpenROAD v2.0-9107-g27b94737f 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: macro_place.tcl, 1 can't read "::env(SCRIPTS_DIR)": no such variable
```
Please update your build again today and generate the test case and attach it.

the variable file looks
```
export RESYNTH_TIMING_RECOVER=0 LSORACLE_KAHYPAR_CONFIG=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/tools/install/LSOracle/share/lsoracle/test.ini SYNTH_ARGS=-flatten WRAPPED_LEFS='' DPO_MAX_DISPLACEMENT=5 1 OPENROAD_CMD=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init GDS_FINAL_FILE=./results/sky130hd/core_v_mini_mcu/base/6_final.gds MAX_UNGROUP_SIZE=100 NPROC=64 CURDIR=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/flow SHELL=/usr/bin/env bash PWR_NETS_VOLTAGES="VDD 1.8" OPENROAD_EXE=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad MAX_ROUTING_LAYER=met5 FLOW_HOME=. DESIGN_CONFIG=./designs/sky130hd/core_v_mini_mcu/config.mk CORE_MARGIN=2 RESULTS_ODB=2_1_floorplan.odb 2_2_floorplan_io.odb 2_3_floorplan_tdms.odb \n='0 LSORACLE_KAHYPAR_CONFIG=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/tools/install/LSOracle/share/lsoracle/test.ini SYNTH_ARGS=-flatten WRAPPED_LEFS='' DPO_MAX_DISPLACEMENT=5 1 OPENROAD_CMD=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init GDS_FINAL_FILE=./results/sky130hd/core_v_mini_mcu/base/6_final.gds MAX_UNGROUP_SIZE=100 NPROC=64 CURDIR=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/flow SHELL=/usr/bin/env bash PWR_NETS_VOLTAGES="VDD 1.8" OPENROAD_EXE=/home/cerdogan/OpenSource/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad MAX_ROUTING_LAYER=met5 FLOW_HOME=. DESIGN_CONFIG=./designs/sky130hd/core_v_mini_mcu/config.mk CORE_MARGIN=2 RESULTS_ODB=2_1_floorplan.odb 2_2_floorplan_io.odb 2_3_floorplan_tdms.odb \n='
```
This has been fixed with latest commit.

BaranErdo, 2023-07-13:
This one should be executable
https://drive.google.com/file/d/10sjrMDC3D9NnyEZ1Dog92msNYzp3Vwu0/view?usp=drive_link

vijayank88, 2023-07-13:
@maliberty 
MPL got stuck with below message.
```
OpenROAD v2.0-9107-g27b94737f 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO MPL-0101] Found 16 macros.
[INFO MPL-0102] West pins 113.
[INFO MPL-0102] East pins 112.
[INFO MPL-0102] North pins 166.
[INFO MPL-0102] South pins 166.
```
Can you please check this?

vijayank88, 2023-07-14:
@BaranErdo 
Can you add below configuration in `config.mk` and update us.
```
export SYNTH_HIERARCHICAL = 1
export RTLMP_FLOW = True
```

vijayank88, 2023-07-14:
@maliberty 
```
export SYNTH_HIERARCHICAL = 1
export RTLMP_FLOW = True
```
These two configuration must be used together or works standalone? 
For example use `RTLMP_FLOW` for macro placement alone.

vijayank88, 2023-07-14:
With `RTLMP_FLOW=True` macro placement completes successfully.
[image]

maliberty, 2023-07-15:
If this is sufficient then I suggest we close this.  I don't want to invest in fixes to the old placer if possible.

> These two configuration must be used together or works standalone?
It is best to use both.  Without SYNTH_HIERARCHICAL there will be no verilog hierarchy for the placer to use.  Without RTLMP_FLOW you'll get the old flow.

closed 2023-07-17 by BaranErdo