The-OpenROAD-Project/OpenROAD-flow-scripts #1216: Write global routing ODB file along with congestion report

vijayank88, 2023-07-10:
### Subject

[Stage]: Global Router.

### Describe the bug

If there is any global routing congestion, flow write `congestion.rpt`, but without writing `5_1_grt.odb`. It makes hard to debug visually where congestion occurs in OpenROAD gui.
```
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/sky130hd/gcd/base/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 21 GRT-0119
Elapsed time: 0:05.06[h:]min:sec. CPU time: user 4.60 sys 0.46 (99%). Peak memory: 558844KB.
make: *** [Makefile:614: results/sky130hd/gcd/base/5_1_grt.odb] Error 1
```

### Expected Behavior

If there is a congestion during global routing stage, write `.odb` along with `congestion.rpt` and exit the flow.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.90.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
-- The CXX compiler identification is GNU 8.3.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-8/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9070-gbb341ffb9
-- System name: Linux
-- Compiler: GNU 8.3.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/rh/devtoolset-8/root/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/local/bin/python3.9 (found version "3.9.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
[... 89 lines cut ...]
-- Found Cbc: /opt/or-tools/lib64/cmake/Cbc/CbcConfig.cmake (found version "2.10.7") 
-- Found Eigen3: /usr/local/share/eigen3/cmake/Eigen3Config.cmake (found version "3.3.90") 
-- Found SCIP: /opt/or-tools/lib/cmake/scip/scip-config.cmake (found version "8.0.1") 
-- GUI is enabled
-- Charts widget is not enabled
-- Found Boost: /usr/local/lib/cmake/Boost-1.72.0/BoostConfig.cmake (found version "1.72.0") found components: serialization 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found suitable version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib64/libtcl.so
-- TCL header: /usr/include/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.5kKBW0Q87v
```

### To Reproduce

Hope description gives idea about requirement

vijayank88, 2023-07-10:
For debug purpose can be created if it's fails with congestion.

vijayank88, 2023-08-10:
@oharboe you may need this option as per discussion https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/discussions/1338

oharboe, 2023-08-10:
> We generally only write the .odb when a step succeeds. If we write it on failure then the next invocation of make will not try to re-run grt. We could write some other .odb just for debug purposes though that may be confusing to novices.

Indeed: the debug .odb would have to be written with a different name.

disk space&runtime vs. usability is an issue: if we write it out when the .rpt files are written that takes disk space and time in the case where the reports might not be used.

if we don't write out the .odb alongside the progress .rpt file, then the user has to go spelunking to get that .odb file written out.

In CTS, you can specify CTS_SNAPSHOTS=1, this writes out extra .odb snapshots with names that don't collide with make dependencies.

Much as I hate options, perhaps a ROUTE_SNAPSHOTS=1?

Or perhaps simply add a generic SNAPSHOTS=1?

maliberty, 2023-08-10:
There is nothing extra to see in the .odb even if we wrote it out.  The markers can be loaded against the cts .odb and will be visually identical.

closed 2026-04-16 by github-actions[bot]