The-OpenROAD-Project/OpenROAD-flow-scripts #1225: platform support

Simonliudan, 2023-07-12:
### Subject

[Design] for example design issues (i.e. the design does not pass functional validation, etc.)

### Describe the bug

i tried GF12 design according to the doc, but there is no GF12 platform files.

### Expected Behavior

to support GF12 platform

### Environment

```shell
ubuntu and CentOS
```

### To Reproduce

in folder:
/flow/design/gf12
/flow/platforms

maliberty, 2023-07-12:
GF12 is not a public PDK and we can't therefore provide the setup.  If you are planning a gf12 tapeout we could work on a three-way NDA with a support contract.

Simonliudan, 2023-07-14:
We have the PDK and would like to know how to install it. Could you provide detailed installation instructions? Additionally, please provide a business email so that we can arrange the NDA process.

vijayank88, 2023-07-14:
@Simonliudan 
Refer this document https://openroad-flow-scripts.readthedocs.io/en/latest/contrib/PlatformBringUp.html
With that you can integrate any pdk with ORFS flow and test existing built-in examples.
If need any further help @maliberty can help you on this.

vijayank88, 2023-07-14:
@Simonliudan 
For NDA contact: tspyrou@precisioninno.com
/cc @tspyrou

Simonliudan, 2023-07-25:
@tspyrou

We would like to add the GF12 PDK to our OR project. Could you provide more detailed documentation and technical support?

vijayank88, 2023-07-25:
@Simonliudan Did you sent a mail to above mail ID for NDA request?

Simonliudan, 2023-07-26:
The email has been sent, but no feedback has been received yet.

maliberty, 2023-07-26:
> The email has been sent, but no feedback has been received yet.

@tspyrou did you receive this email?

closed 2026-04-16 by github-actions[bot]