The-OpenROAD-Project/OpenROAD-flow-scripts #1241: Unable to run sky130hd gcd

dalyles, 2023-07-17:
### Subject

[Documentation] for documentation errors.

### Describe the bug

Hello again,

I'm trying to get started on running the some of the example flows using the <code>sky130</code> technology file.
What ultimately happens when I run make is that ORFS stops at the placement stage. I'll copy/paste
some of the warnings and error messages that showed up on the console since the entire dump is too large to post here.

### Expected Behavior

The expected behavior is a clean run of <code>gcd</code> in the <code>sky130hd</code> technology file.

### Environment

```shell
N/A
```

### To Reproduce

Input: gcd from sky130hd in makefile
Output: Completed run without interrupts through the flow.

### Screenshots

Here are some warnings/errors:
[image]

[image]

The screenshot consists of thousands of lines!
[image]

[image]

[image]

This is the where the run interrupts and teminates:
[image]

maliberty, 2023-07-17:
This design is passing in the CI so there is something wrong in your setup.  Please describe how you are setting up and running.

dalyles, 2023-07-17:
I'm simply sourcing env.sh, and then I go to the makefile and uncomment the path to <code>gcd</code> for <code>sky130hd</code>. I'm not doing anything particularly different.

dalyles, 2023-07-18:
I did a tar for my workspace: 
Try this https://drive.google.com/file/d/1dOsR0v7ycP4Y77LHABrXfxjq2zrPAlos/view?usp=sharing

rovinski, 2023-07-18:
@dalyles your screenshots are from the design `coyote_tc` not `gcd`

rovinski, 2023-07-18:
Also it's much much more helpful to copy and paste formatted code instead of screenshots.

vijayank88, 2023-07-18:
> @dalyles your screenshots are from the design `coyote_tc` not `gcd`

@dalyles Seems you mistakenly uncomment the `coyote_tc` instead of `gcd`. To avoid confusion over editing `Makefile`, directly execute with `make DESIGN_CONFIG=./designs/sky130hd/gcd/config.mk` from `flow` directory.

maliberty, 2023-07-18:
Please try with the correct design.  Reopen if a problem occurs with gcd

closed 2023-07-18 by maliberty

dalyles, 2023-07-18:
Thank you for clarification. It was an oversight on my part. Apologies for that.
I was able to run gcd successfully.

Thanks everybody!