The-OpenROAD-Project/OpenROAD-flow-scripts #1246: FLOW_VARIANT fails to complete the flow

vijayank88, 2023-07-18:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

With default `make` command works fine 
If I run `make FLOW_VARIANT=test1` fails to complete the flow.

### Expected Behavior

`make FLOW_VARIANT=test1` should complete the flow.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 3.10.0-1160.90.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
-- The CXX compiler identification is GNU 8.3.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /opt/rh/devtoolset-8/root/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9070-gbb341ffb9
-- System name: Linux
-- Compiler: GNU 8.3.1
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 8.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/rh/devtoolset-8/root/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/local/bin/python3.9 (found version "3.9.6") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
[... 97 lines cut ...]
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
```

### To Reproduce

Run default design with `make FLOW_VARIANT=test1`

### Relevant log output

```shell
OpenROAD v2.0-9070-gbb341ffb9 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO PDN-0001] Inserting grid: grid
Elapsed time: 0:00.59[h:]min:sec. CPU time: user 0.53 sys 0.06 (99%). Peak memory: 79604KB.
make[1]: Leaving directory '/home/vijayan/OPENROAD_FLOW/ORFS_LOCAL/OpenROAD-flow-scripts/flow'
cp results/sky130hd/gcd/test1/2_6_floorplan_pdn.odb results/sky130hd/gcd/test1/2_floorplan.odb
cp: cannot stat \u2018results/sky130hd/gcd/test1/2_6_floorplan_pdn.odb\u2019: No such file or directory
make: *** [Makefile:508: results/sky130hd/gcd/test1/2_floorplan.odb] Error 1
```

vvbandeira, 2023-07-18:
This is failing because the `UNSET_AND_MAKE` combined with `do-step`.

@maliberty @oharboe While looking into this issue I noticed that #1207 might have introduced unwanted behaviour/unforeseen consequences. 

The flow after #1207 ignores any parameter set by the user via `export VAR=VALUE` or `make VAR=VALUE` since all stages call `UNSET_AND_MAKE`, which in turn unset all variables unconditionally. This makes `make issue_*` and AutoTuner silently broken. Sourcing the `make issue_*` generated `var*.sh` is pointless since the variables are unset by `UNSET_AND_MAKE` and AutoTuner changes the parameters via `make VAR=VALUE`.

Am I missing something or the above is true?

oharboe, 2023-07-18:
> This is failing because the `UNSET_AND_MAKE` combined with `do-step`.
>
> @maliberty @oharboe While looking into this issue I noticed that #1207 might have introduced unwanted behaviour/unforeseen [...]

I'm a bit confused as to what you are asking.

#1207 only affects the generate_abstract target.

The goal of this commit was to make BLOCKS built from the containing macro and BLOCKS built seperately give the same result.

How are you running `make issue_*`?

Can you give a specific example of a sequence of commands that is no longer working as before?

vvbandeira, 2023-07-18:
Let's say I want to see what happens if I change the default value of `PLACE_DENSITY_LB_ADDON` for `nangate45/gcd`.
The default value is `0.2`, before #1207 the make call below would correctly set the variable, after the PR the `PLACE_DENSITY_LB_ADDON=0.4` is ignored. The only way to test the change is by changing the `config.mk` file itself.

```
make PLACE_DENSITY_LB_ADDON=0.4
```

For the `make issue`, in the past this was a valid workflow
```
# user has a problem during cts and runs
make cts_issue
# developer gets a cts_gcd_nangate45_base_2023-07-18_13-55.tar.gz
tar --strip 1 -xf cts_gcd_nangate45_base_2023-07-18_13-55.tar.gz
source vars-gcd-nangate45-base.sh
make cts
```
Now it does not matter what the user had different from the default as they get ignored/unset by `UNSET_AND_MAKE`.

vvbandeira, 2023-07-18:
Maybe it would be more correct to say that #1207 + #1175 are the cause of the problem.

maliberty, 2023-07-18:
@oharboe please advise if you have a solution; otherwise we should revert #1175

oharboe, 2023-07-18:
Some investigations, first modify Makefile:

```
$ git diff Makefile
diff --git a/flow/Makefile b/flow/Makefile
index 32226039..6be9fcac 100644
--- a/flow/Makefile
+++ b/flow/Makefile
@@ -893,7 +893,11 @@ all_verilog : $(foreach file,$(RESULTS_ODB),$(file).v)
 .PHONY: handoff
 handoff : all_defs all_verilog
 
-print-%  : ; @echo $* = $($*)
+.PHONY: print-%
+print-%: ; @echo $* = $($*)
+
+.PHONY: test-unset-and-make-%
+test-unset-and-make-%: ; $(UNSET_AND_MAKE) $*
 
 # Utilities
 #-------------------------------------------------------------------------------
```

Now run some tests:

```
$ echo $FOO
$ make test-unset-and-make-print-FOO
FOO =
$ make FOO=bar test-unset-and-make-print-FOO
FOO =
$ FOO=bar make test-unset-and-make-print-FOO
FOO = bar
$ export FOO=bar2
$ make FOO=bar test-unset-and-make-print-FOO
FOO =
$ make test-unset-and-make-print-FOO
FOO = bar2
```

Hmm... I can't explain why sourcing the vars file wouldn't work. Seems like it should.

oharboe, 2023-07-18:
@vvbandeira Try #1250

```
$ echo $FOO
bar2
$ make FOO=bar test-unset-and-make-print-FOO
FOO = bar
$ make test-unset-and-make-print-FOO
FOO = bar2
$ unset FOO
$ make test-unset-and-make-print-FOO
FOO =
$ make FOO=bar test-unset-and-make-print-FOO
FOO = bar
```

PR #1250 (2023-07-18): makefile: fix passing of variables on command line
fixes #1246
commit f52a9fa1 (2023-07-18): makefile: fix passing of variables on command line

fixes #1246
closed 2023-07-18 by maliberty