The-OpenROAD-Project/OpenROAD-flow-scripts #1272: CTS Running around with 6 Days and not completed !!

lakshmikanthgummadi, 2023-07-24:
### Subject

[Stage]: CTS.

### Describe the bug

ORFS CTS taking around 6 days and still it's running, working on 2M instance count design with cloud resource, 
cloud configs are below
[image]
 below are RAM Config 
 
[image]

Is there any possibility that we can shorten runtime??

### Expected Behavior

CTS has to finish within four hours !

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.19.0-1028-aws
os: Ubuntu 22.04.2 LTS (Jammy Jellyfish)
cmake version 3.24.2
-- The CXX compiler identification is GNU 11.3.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-8118-gb95ce9901
-- System name: Linux
-- Compiler: GNU 11.3.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- Found GTest: /usr/lib/x86_64-linux-gnu/cmake/GTest/GTestConfig.cmake (found version "1.11.0")  
-- The C compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
[... 206 lines cut ...]
set env(ABC_CLOCK_PERIOD_IN_PS) "2000"
set env(MIN_ROUTING_LAYER) "M2"
set env(PROCESS) "7"
set env(GDS_MERGED_FILE) "./results/asap7/riscv_soc/base/6_1_merged.gds"
set env(TEST_SCRIPT) "./test/core_tests.sh"
set env(NUM_CORES) "4"
set env(OPENROAD_CMD) "/home/ubuntu/Opensource/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init"
set env(ABC_LOAD_IN_FF) "3.898"
set env(MAX_ROUTING_LAYER) "M9"
set env(CORNER) "BC"
set env(OPENROAD_ARGS) "-no_init"
set env(OPENROAD_GUI_CMD) "/home/ubuntu/Opensource/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -gui"
set env(WC_VOLTAGE) "0.63"
set env(PLACE_PINS_ARGS) "-min_distance 20 -min_distance_in_tracks"
set env(TC_DFF_LIB_FILE) "./platforms/asap7/lib/asap7sc7p5t_SEQ_RVT_TT_nldm_220123.lib"
set env(FLOW_VARIANT) "base"
set env(RULES_DESIGN) "./designs/asap7/riscv_soc/rules-base.json"
set env(RTLMP_BLOCKAGE_FILE) "./objects/asap7/riscv_soc/base/rtlmp/partition.txt.blockage"
set env(RTLMP_RPT_FILE) "partition.txt"
set env(IR_DROP_LAYER) "M1"
set env(GDS_ALLOW_EMPTY) "fakeram.*"

### Screenshots

[image]

vijayank88, 2023-07-24:
@lakshmikanthgummadi 
Run `make cts_issue` for your design and attach a test case to debug the same.
Maybe 2M instances, the given RAM is not sufficient. It should terminated the run, don't understand why it is still running.
Just run and check `htop` what are the jobs running still.

lakshmikanthgummadi, 2023-07-24:
[image]
 
killed the run manually !

vvbandeira, 2023-07-24:
@lakshmikanthgummadi

We have anecdotal information that a design with 1.7M instances takes over 30h to run on a 32 CPU 128GB RAM machine.

Looks like you have a 4 CPU, and 16GB RAM machine, which might not be sufficient to run such a large design. You have another 32 GB of swap, but if the design uses swap, the run time can be exponentially worse because this will use HDD/SSD as RAM, which is very slow compared to native RAM.

From your last print of `htop`, was `openroad` still running? There's no `openroad` info on the print, and the load average points to no process running for the last past 15 min. If this was before you killed the process, this is useful; otherwise, not so much.

maliberty, 2023-07-24:
Is it possible to provide a test case?

lakshmikanthgummadi, 2023-07-26:
Again fired the run yesterday its been more then 19 hours , and still it is running on cts stage , sharing the log screen shot and htop

[image]

[image]

vijayank88, 2023-07-26:
> @lakshmikanthgummadi Run `make cts_issue` for your design and attach a test case to debug the same. Maybe 2M instances, the given RAM is not sufficient. It should terminated the run, don't understand why it is still running. Just run and check `htop` what are the jobs running still.

@lakshmikanthgummadi Please understand few things it is mandatory to get support.
By sharing log/screenshot won't resolve your issue.
Run `make cts_issue` and attach test case generated in `flow` directory to get solution.

rovinski, 2023-07-26:
@lakshmikanthgummadi a 2M instance design will not finish on a machine with 16GB RAM. Full stop. You need on the order of 8 times more RAM. Swap will not help you. Your only other option is to partition the design down to much much less than 2M instances per block.

lakshmikanthgummadi, 2023-07-26:
> @lakshmikanthgummadi Please understand few things it is mandatory to get support. By sharing log/screenshot won't resolve your issue. Run `make cts_issue` and attach test case generated in `flow` [...]

tar file size was 1113 mb, not allowed to upload here , how to share the file ??

vijayank88, 2023-07-26:
Upload it to google drive and share access link

lakshmikanthgummadi, 2023-07-26:
> Upload it to google drive and share access link

https://drive.google.com/file/d/1OfY6tSnXlllg7-TJEaHfYeYHqPxLMhhn/view?usp=sharing

vijayank88, 2023-07-26:
> > Upload it to google drive and share access link
> 
> https://drive.google.com/file/d/1OfY6tSnXlllg7-TJEaHfYeYHqPxLMhhn/view?usp=sharing

provide public access to file location

lakshmikanthgummadi, 2023-07-26:
> > > Upload it to google drive and share access link
> > 
> > 
> > https://drive.google.com/file/d/1OfY6tSnXlllg7-TJEaHfYeYHqPxLMhhn/view?usp=sharing
> 
> provide public access to file location

given !!

lakshmikanthgummadi, 2023-07-28:
Any Update @ @vijayank88 @rovinski @

vishallparmar25, 2023-07-31:
Same issue i also facing, any update ???

vijayank88, 2023-07-31:
@lakshmikanthgummadi 
Can you share your `RTL`, `config.mk`, `constraint.sdc`?

@vishallparmar25 
Open a new issue by following issue template and attach a test case.

maliberty, 2023-07-31:
Does this make sense:
```
[WARNING RSZ-0069] skipping net u_core/u_icache/data_addr_r[0] with 103883 pins.
```
Is that single bit really a global signal?

maliberty, 2023-07-31:
Something is quite wrong in 3_4_resizer.log
```
==========================================================================
resizer report_wns
--------------------------------------------------------------------------
wns -164492.39
...
resizer report_check_types -max_slew -max_cap -max_fanout -violators
--------------------------------------------------------------------------
max slew

Pin                                    Limit    Slew   Slack
------------------------------------------------------------
u_core/u_icache/_0848_/Y              320.00 252488.52 -252168.52 (VIOLATED)
...
max capacitance

Pin                                    Limit     Cap   Slack
------------------------------------------------------------
u_core/u_icache/_0848_/Y               92.16 84234.69 -84142.52 (VIOLATED)
```

There are huge violations.  In CTS the timing optimization is starting from a horrible place.  I think we need to understand what is happening in 3_4 first.

maliberty, 2023-07-31:
The 3_4 issue comes back to the warning above.  Such a large net produces terrible runtimes in building a Steiner tree and so we skip it.  Please provide more detail on what the intention is.

maliberty, 2023-07-31:
It looks like you have rams that you are synthesizing at gate level (eg icache_data_ram).  You should use a RAM macro for those.

vijayank88, 2023-08-04:
@lakshmikanthgummadi 
Can you update status on this?

lakshmikanthgummadi, 2023-08-07:
> @lakshmikanthgummadi Can you update status on this?

vijayank88, 2023-08-07:
@lakshmikanthgummadi 
Have you replaced with RAM or how you resolved the issue?

lakshmikanthgummadi, 2023-08-07:
> @lakshmikanthgummadi Have you replaced with RAM or how you resolved the issue?

we are migrating to new cloud server once it is done i will update , It may take one week of time ! Thanks