The-OpenROAD-Project/OpenROAD-flow-scripts #1286: Routing congestion problem with sram macros

BaranErdo, 2023-07-27:
### Subject

[Stage]: Global Router.

### Describe the bug

I wa running the routing stage with SRAM macros however I have congestion problems. I tried different halo values but couldn't resolve.  I get this error:

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0103] Extra Run for hard benchmark.
[INFO GRT-0197] Via related to pin nodes: 1355596
[INFO GRT-0198] Via related Steiner nodes: 64455
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 1612346
[INFO GRT-0112] Final usage 3D: 7667022
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/sky130hd/core_v_mini_mcu/base/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 21 GRT-0119
Command exited with non-zero status 1
Elapsed time: 6:18.10[h:]min:sec. CPU time: user 347.76 sys 29.89 (99%). Peak memory: 12729940KB.

### Expected Behavior

I am trying the find a way to do the global routing without problem. Can you help me with this?

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.4.0-148-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.16.3
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9070-gbb341ffb9
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /home/cerdogan/miniconda3/envs/core-v-mini-mcu/bin/python3.8 (found version "3.8.16") found components: Interpreter 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
[... 125 lines cut ...]
  directories:

    runtime library [libtcl8.6.so] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cerdogan/miniconda3/envs/core-v-mini-mcu/lib
    runtime library [libz.so.1] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/cerdogan/miniconda3/envs/core-v-mini-mcu/lib

  Some of these libraries may not be found correctly.

e version "1.80.0", minimum required is "1.78")  
-- TCL library: /usr/lib/x86_64-linux-gnu/libtcl.so
-- TCL header: /usr/include/tcl/tcl.h
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0") found components: serialization system thread 
-- Found Boost: /usr/local/lib/cmake/Boost-1.80.0/BoostConfig.cmake (found version "1.80.0")  
-- TCL readline enabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/tmp.TBZWSxe6xP
```

### To Reproduce

https://drive.google.com/file/d/1EnH2Tmu27ugmVvGK5hilhDGHyOac_eOC/view?usp=sharing

vijayank88, 2023-07-27:
@BaranErdo 
See the routing congestion view.
[image]

You may try reduce placement density and increase macro halo/channel width and check again.

rovinski, 2023-07-27:
Decreasing the area might also be helpful so that the pins and SRAMs aren't so far apart.

vijayank88, 2023-07-31:
@BaranErdo 
Any updates on this? Did you updated area and run the flow?

maliberty, 2023-07-31:
If I looked at the congestion markers tehy are all around your macros:
[image]

I suggest you increase the spacing between your macros and between the macros the boundary.

BaranErdo, 2023-08-02:
Increasing macro halo/channel width and decreasing the area solved the problem. Thank you for the help

closed 2023-08-02 by vijayank88

Wiki1998-dev, 2024-08-30:
Hi @BaranErdo  how u solved it for me same design with parameters of your repository dont work