The-OpenROAD-Project/OpenROAD-flow-scripts #1287: sky130hd fill.def contains + MASK 1

stefanottili, 2023-07-28:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

1) read_db 6_1_fill.odb doesn't contain any fill, why saving it ?

2) density_fill -rules ../../../../platforms/sky130hd/fill.json, write_def 6_1_fill.def, the def contains
- LAYER met1 + MASK 1 RECT ( ... ) ;
Why the + MASK 1 ?

3) should 6_1_final.gds contain fill data ?

### Expected Behavior

1) 6_1_fill.odb should either contain fill data or not be written

2) sky130hd write_def FILLS shouldn't use + MASK 1

### Environment

```shell
v2.0-9530-g8f86421cb
MacOS Ventura Apple Silicon
```

### To Reproduce

read_db results/sky130hd/chameleon/base/6_1_fill.odb
density_fill -rules platforms/sky130hd/fill.json
write_def 6_1_fill.def

### Relevant log output

```shell
589609 FILLS 1970078 ;
 589610     - LAYER met1 + MASK 1 RECT ( 200800 1533260 ) ( 202800 1535260 ) ;
 589611     - LAYER met1 + MASK 1 RECT ( 203100 1533260 ) ( 205100 1535260 ) ;
 589612     - LAYER met1 + MASK 1 RECT ( 205400 1533260 ) ( 207400 1535260 ) ;
```

stefanottili, 2023-07-31:
looking at scripts/run_all.tcl

 54 # Finishing
 55 if {[info exists ::env(USE_FILL)]} {
 56   source -verbose $::env(SCRIPTS_DIR)/density_fill.tcl
 57 }

export USE_FILL

gets fill into 6_1_fill.odb, but then the MASK issue pops up in klayout def/gds merging.

[INFO][FLOW] Using platform directory ./platforms/sky130hd
cp ./results/sky130hd/gcd/base/5_route.sdc ./results/sky130hd/gcd/base/6_final.sdc
(/usr/bin/time stdbuf -o L /Users/stefanthiede/bin/klayout -zz -rd design_name=gcd \
	        -rd in_def=results/sky130hd/gcd/base/6_final.def \
	        -rd in_files="./platforms/sky130hd/gds/sky130_fd_sc_hd.gds  " \
	        -rd config_file=./platforms/sky130hd/fill.json \
	        -rd seal_file="" \
	        -rd out_file=results/sky130hd/gcd/base/6_1_merged.gds \
	        -rd tech_file=./objects/sky130hd/gcd/base/klayout.lyt \
	        -rd layer_map= \
	        -r ./util/def2stream.py) 2>&1 | tee ./logs/sky130hd/gcd/base/6_1_merge.log
Warning: No mapping for layer 'met2', purpose 'FILL' Mask 1 - layer is ignored
Warning: No mapping for layer 'met3', purpose 'FILL' Mask 1 - layer is ignored
Warning: No mapping for layer 'met4', purpose 'FILL' Mask 1 - layer is ignored
Warning: No mapping for layer 'met5', purpose 'FILL' Mask 1 - layer is ignored
[INFO] Reporting cells prior to loading DEF ...
[INFO] Reading DEF ...
[INFO] Clearing cells...
[INFO] Merging GDS/OAS files...
	./platforms/sky130hd/gds/sky130_fd_sc_hd.gds
[INFO] Copying toplevel cell 'gcd'
INFO: Reading config file: ./platforms/sky130hd/fill.json
ERROR: Unrecognized fill:     - LAYER met2 + MASK 1 RECT ( 65000 2720 ) ( 65300 3020 ) ;

  ./util/def2stream.py:79
  ./util/def2stream.py:143

stefanottili, 2023-07-31:
Here's is my current understanding of using fill in ORFS:

OR can add fill, but only uses it for extraction to get more accurate spef, doesn't add it to odb.
So adding it is a rarely tested feature that can be enabled with 'export USE_FILL'

Klayout can add fill, but OR is not using that.

OpenLane uses Magic to add fill for sky130.

I'm assuming that for other technologies the "usual" way of adding fill is to be using physical verification tools to generate fill gds/oas. These tools usually generate gds/oas with array references to keep the file size small for the large number of tiles.

I've not seen OR extraction being able to read fill gds/oas from either Magic or other tools.

Did I miss something ?
Is reading gds/oas fill for extraction on a "todo" list ?

maliberty, 2023-07-31:
The fill is stored in odb and is written in the DEF if present.  Last I looked KLayout didn't add fill automatically but perhaps that has changed.

I am not sure that RCX can handle fill at all today.  It was written before odb even had fill storage.  I suspect reading from gds is much further down the list of issues.

maliberty, 2023-07-31:
The MASK statement is a problem.  This code was tested with private PDKs that use multiple masks.  In OL they don't use OR so it hasn't seen much use in sky130.  I'll fix it  up.

maliberty, 2023-07-31:
The somewhat longer history is that ORFS used to generate fill in KLayout using a script.  However we had problems with klayout crashing and could not release the proprietary data needed to reproduce it.  Eventually it became simpler to just generate the fill in OR.  Eventually this could enable timing aware fill insertion.

closed 2023-07-31 by maliberty