The-OpenROAD-Project/OpenROAD-flow-scripts #1294: sky130hd should map def PIN to both metal and pin layer:datatype

stefanottili, 2023-07-31:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

results/sky130hd/gcd/base/6_1_merged.gds only contains pin geometry, it's missing the net geometry for def PIN's

klayout can do 1:n mappings using the '+' notation, klayout.lyt needs to be updated.

### Expected Behavior

Write both pin and net layer:datatype geometry for def PIN's.

klayout can do 1:n mapping using the '+' notation

adding '+ met3.pin: 70/20'; to ./objects/sky130hd/gcd/base/klayout.lyt generates both 70/16 pin and 70/20 net geometry for met3 pins. This should be done for all pin layers.

25    <layer-map>layer_map(...;'met3.pin : 70/16'; '+ met3.pin : 70/20';

Alternatively one can use a lefdef.map file, the notation there would be
met3 PIN 70 16
met3 PIN 70 20

### Environment

```shell
Git commit: 90f22d15990f0bf8aa33751a712c37048ffa9006
kernel: Darwin 22.6.0
os: macOS 13.5
cmake version 3.27.1
```

### To Reproduce

make DESIGN_CONFIG=./designs/sky130hd/gcd/config.mk

maliberty, 2023-07-31:
@stefanottili do you want to make a PR with your suggestion?

stefanottili, 2023-07-31:
I have no idea where ./objects/sky130hd/gcd/base/klayout.lyt is coming from.
Is it "hand written" or generated by yet another script from some other source ?

I'm assuming that isn't of any priority, because sky130's def2gds conversion is done by magic.
Presumably specified with yet another tech file.

maliberty, 2023-07-31:
From https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/0bce776a9e7e5cd7cb6151120e01a3c389fdbf05/flow/Makefile#L353

the upstream source is https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/sky130hd/sky130hd.lyt which is manual

maliberty, 2023-07-31:
Nobody is taping out from ORFS directly to skywater afaik - skywater users are using OL.