The-OpenROAD-Project/OpenROAD-flow-scripts #1295: Attempting to Run 3rd party cell for sky130hd (and furthermore sky130hs)

dalyles, 2023-07-31:
### Subject

[Documentation] for documentation errors.

### Describe the bug

I'm currently trying to run a cell named picor32.v with the autotuner, however upon completion of the runs, the output results are incomplete as shown here: 

[image]

From my understanding, it doesn't seem to sucessfully make it through the placement stage.

### Expected Behavior

What is expected is to have all output files in logs completed for further analysis.

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.0-76-generic
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.26.4
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9070-gbb341ffb9
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /home/dlyles/mambaforge/bin/python3.10 (found version "3.10.9") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
[... 268 lines cut ...]
```

The file <code>constraint.sdc</code>
```
current_design picorv32

set clk_name  core_clock
set clk_port_name clk
set clk_period 0.46 
set clk_io_pct 0.2

set clk_port [get_ports $clk_port_name]

create_clock -name $clk_name -period $clk_period $clk_port

set non_clock_inputs [lsearch -inline -all -not -exact [all_inputs] $clk_port]

set_input_delay  [expr $clk_period * $clk_io_pct] -clock $clk_name $non_clock_inputs 
set_output_delay [expr $clk_period * $clk_io_pct] -clock $clk_name [all_outputs]
```

### Screenshots

More Detailed Screenshot of output of autotuner at terminating stage:
[image]

maliberty, 2023-07-31:
Please package a specific test case showing the issue (eg make issue).  Setting up and re-running the autotuner to recreate the issue is impractical.

dalyles, 2023-07-31:
Hi @maliberty 

I did a simple make run and I have the results of the run here:

[picorv32_sky130hd.log](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12222239/picorv32_sky130hd.log)

maliberty, 2023-07-31:
Thanks but a log is not a test case.  I need a package set of files and scripts representing a single run of OR that reproduces your result.  If you are using ORFS you can use 'make global_place_issue' to create such. 

This will include you cell library so make sure it is ok to make public.

dalyles, 2023-07-31:
Yes it's fine to for me to have it public. I think what I can do is tar.gz my workspace and the hand it over.
Here it is: https://drive.google.com/file/d/1DtPxpZ8b4lO0sEOaZ5QO8GNJwG18_63L/view?usp=sharing

The makefile is already configured to the run described above.

vijayank88, 2023-08-01:
@dalyles 
Can you confirm with `Best parameters found` run tag not contain complete log files?
[image]

dalyles, 2023-08-01:
Yes it does not contain complete log files. It ends at files starting with prefix 3, which assumes that placement failed.

vijayank88, 2023-08-01:
@dalyles In `sky130hd` platform why used `nangate45` `fastroute.tcl` in `autotuner.json`?
```
    "_FR_FILE_PATH": "../../../platforms/nangate45/fastroute.tcl",
    "_FR_GR_OVERFLOW": {
        "type": "int",
        "minmax": [
            1,
            1
        ],
        "step": 0
```
But sample `make` run failed during global routing
```
[ERROR GRT-0119] Routing congestion too high. Check the congestion heatmap in the GUI and load ./reports/sky130hd/picorv32/base/congestion.rpt in the DRC viewer.
Error: global_route.tcl, 21 GRT-0119
Command exited with non-zero status 1
Elapsed time: 0:07.67[h:]min:sec. CPU time: user 7.60 sys 0.07 (99%). Peak memory: 192836KB.
make: *** [Makefile:625: results/sky130hd/picorv32/base/5_1_grt.odb] Error 1
```
Seems something wrong with your configuration itself?

dalyles, 2023-08-01:
Hi @vijayank88,

Yes, you're right. It seems to be an oversight on my part. I'll follow up once I can confirm that the runs are complete and working properly. Please stayed tuned.

vijayank88, 2023-08-01:
Maybe you can try with clock period value of `20ns` and core utilization of `35`

dalyles, 2023-08-01:
@vijayank88 Thank you. Instead I made sure the run went through properly and then let the autotuner optimize it before I saw your comment. Everything seems to be working well.

maliberty, 2023-08-01:
ok to close?

closed 2023-08-15 by dalyles