The-OpenROAD-Project/OpenROAD-flow-scripts #1297: Unable to build OpenLOAD

dhdhlee, 2023-08-01:
### Subject

[Build] for compilation-related issues.

### Describe the bug

I've tried building openload several times, but I keep getting the error. 
Each time I try to build, I get a different error.
Enter the following commands in this order
/////////////////////////////////////////////////////////////////////////////////////////////////////
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts
cd OpenROAD-flow-scripts
sudo ./setup.sh
./build_openroad.sh --local
/////////////////////////////////////////////////////////////////////////////////////////////////////

In the environment below, the version of OpenROAD is outdated, can I use the --latest option?
Even if I use the --latest option, I get an error.

I repeatedly ran ./build_openroad.sh --local --latest, and yosys built at some point, but openload never finished. I think it ignored the error and proceeded further.

### Expected Behavior

Building openload

### Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.90.1-microsoft-standard-WSL2
os: Ubuntu 20.04.6 LTS (Focal Fossa)
cmake version 3.24.2
-- The CXX compiler identification is GNU 9.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-9542-ge5a7ad7ae
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Python: /usr/bin/python3.8 (found version "3.8.10") found components: Interpreter
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
[... 285 lines cut ...]
[ 41%] Building passes/techmap/aigmap.o
[ 42%] Building passes/techmap/tribuf.o
[ 42%] Building passes/techmap/lut2mux.o
[ 42%] Building passes/techmap/nlutmap.o
[ 43%] Building passes/techmap/shregmap.o
[ 43%] Building passes/techmap/deminout.o
[ 43%] Building passes/techmap/insbuf.o
[ 44%] Building passes/techmap/attrmvcp.o
[ 44%] Building passes/techmap/attrmap.o
[ 44%] Building passes/techmap/zinit.o
[ 45%] Building passes/techmap/dfflegalize.o
[ 45%] Building passes/techmap/dffunmap.o
[ 45%] Building passes/techmap/flowmap.o
[ 46%] Building passes/techmap/extractinv.o
/tmp/ccbFOvTR.s: Assembler messages:
/tmp/ccbFOvTR.s: Internal error (Segmentation fault).
Please report this bug.
make: *** [Makefile:705: passes/techmap/extract_fa.o] Error 1
make: *** Waiting for unfinished jobs....
passes/techmap/abc.cc: In member function ‘virtual void {anonymous}::AbcPass::execute(std::vector<std::__cxx11::basic_string<char> >, Yosys::RTLIL::Design*)’:
passes/techmap/abc.cc:1825:15: warning: ‘g_argidx’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1825 |      cmd_error(args, g_argidx, stringf("Unsupported gate type: %s", g.c_str()));
      |      ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make: Leaving directory '/home/dhlee/OpenROAD-flow-scripts/tools/yosys'
```

vijayank88, 2023-08-01:
@dhdhlee 
Correct name `OpenLOAD` to `OpenROAD`
Try following option as you're doing 1st time installation.
```
cd OpenROAD-flow-scripts
git clean -xdf ./tools
./build_openroad.sh --clean --local --threads 1
```

vijayank88, 2023-08-01:
Also you're using Ubuntu20.04, we encourage you to try prebuilt binaries method of ORFS by using steps from https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithPrebuilt.md

dhdhlee, 2023-08-01:
Thank you very much. I'll give it a try as you suggested.

vijayank88, 2023-08-02:
@dhdhlee Is that working? can we close this issue?

dhdhlee, 2023-08-02:
@vijayank88 
I solved it with your first suggestion. 
Thank you.

closed 2023-08-02 by vijayank88