The-OpenROAD-Project/OpenROAD-flow-scripts #1337: Error126

JJs-Pi, 2023-08-09:
### Subject

[Flow] for any util, flow Makefile, or flow script issues.

### Describe the bug

When I run the flow the following error message appears:
Warnings: 10 unique messages, 76 total
End of script. Logfile hash: 7c98f96e62, CPU: user 20.74s system 0.13s, MEM: 210.57 MB peak
Yosys 0.31+16 (git sha1 c7670b36d, clang 10.0.0-4ubuntu1 -fPIC -Os)
Time spent: 60% 2x abc (31 sec), 14% 1x extract_fa (7 sec), ...
Elapsed time: 0:51.81[h:]min:sec. CPU time: user 50.60 sys 0.90 (99%). Peak memory: 215620KB.
cp ./designs/nangate45/ibex/constraint.sdc ./results/nangate45/ibex/base/1_synth.sdc
mkdir -p ./results/nangate45/ibex/base ./logs/nangate45/ibex/base ./reports/nangate45/ibex/base
cp results/nangate45/ibex/base/1_1_yosys.v results/nangate45/ibex/base/1_synth.v
--: line 0: unset: `do-copy': not a valid identifier
--: line 0: unset: `do-step': not a valid identifier
make DESIGN_CONFIG=./designs/nangate45/ibex/config.mk do-2_1_floorplan
make[1]: Entering directory '/home/vboxuser/OpenROAD-flow-scripts/flow'
[INFO][FLOW] Using platform directory /home/vboxuser/OpenROAD-flow-scripts/flow/platforms/nangate45
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/vboxuser/OpenROAD-flow-scripts/flow -exit -no_init  /home/vboxuser/OpenROAD-flow-scripts/flow/scripts/floorplan.tcl -metrics ./logs/nangate45/ibex/base/2_1_floorplan.json) 2>&1 | tee ./logs/nangate45/ibex/base/2_1_floorplan.log
/usr/bin/time: cannot run /home/vboxuser/OpenROAD-flow-scripts/flow: Permission denied
Command exited with non-zero status 126
Elapsed time: 0:00.00[h:]min:sec. CPU time: user 0.00 sys 0.00 (?%). Peak memory: 972KB.
make[1]: *** [Makefile:503: do-2_1_floorplan] Error 126
make[1]: Leaving directory '/home/vboxuser/OpenROAD-flow-scripts/flow'
make: *** [Makefile:503: results/nangate45/ibex/base/2_1_floorplan.odb] Error 2
vboxuser@Ubuntu-Openroad:~/OpenROAD-flow-scripts/flow$

### Expected Behavior

The 1_1_yosis.v, 1_synth.sdc and 1_synth.v are written but the .odb not.

### Environment

```shell
bash: ./etc/Env.sh: No such file or directory
```

### To Reproduce

Later

### Relevant log output

```shell
2_1_floorplan.log
/usr/bin/time: cannot run /home/vboxuser/OpenROAD-flow-scripts/flow: Permission denied
Command exited with non-zero status 126
Elapsed time: 0:00.00[h:]min:sec. CPU time: user 0.00 sys 0.00 (?%). Peak memory: 972KB.
```

maliberty, 2023-08-09:
"/usr/bin/time: cannot run /home/vboxuser/OpenROAD-flow-scripts/flow: Permission denied" sounds more like a permission issue in your local filesystem.

JJs-Pi, 2023-08-10:
Everything is on rwx. I don't know, what it can be.

vijayank88, 2023-08-10:
> Everything is on rwx. I don't know, what it can be.

@JJs-Pi Run following in terminal and copy paste the log.
```
ll -a /home/vboxuser/OpenROAD-flow-scripts/
```
To get environment 
```
cd /home/vboxuser/OpenROAD-flow-scripts/tools/OpenROAD
./etc/Env.sh
```

maliberty, 2023-08-10:
The first error I can see is
```
--: line 0: unset: do-copy': not a valid identifier --: line 0: unset: do-step': not a valid identifier
```
What version of make do you have?  What OS?

JJs-Pi, 2023-08-11:
@ Vijayank88

 ll -a /home/vboxuser/OpenROAD-flow-scripts
total 100
drwxrwxrwx 10 vboxuser vboxuser  4096 Jul 24 14:45 ./
drwxr-xr-x 20 vboxuser vboxuser  4096 Aug 11 12:31 ../
-rwxrwxr-x  1 vboxuser vboxuser 12733 Jul 24 14:45 build_openroad.sh*
-rwxrwxr-x  1 vboxuser vboxuser   231 Jul 24 14:45 dev_env.sh*
drwxrwxr-x  2 vboxuser vboxuser  4096 Jul 24 14:45 docker/
-rw-rw-r--  1 vboxuser vboxuser   792 Jul 24 14:45 Dockerfile
-rw-rw-r--  1 vboxuser vboxuser   361 Jul 24 14:45 .dockerignore
drwxrwxr-x  6 vboxuser vboxuser  4096 Jul 24 14:45 docs/
-rwxrwxrwx  1 vboxuser vboxuser   756 Jul 24 14:45 env.sh*
drwxrwxrwx  2 vboxuser vboxuser  4096 Jul 24 14:45 etc/
drwxrwxrwx 12 vboxuser vboxuser  4096 Aug  9 11:37 flow/
drwxrwxr-x  8 vboxuser vboxuser  4096 Jul 24 14:45 .git/
-rw-rw-r--  1 vboxuser vboxuser   108 Jul 24 14:45 .gitattributes
drwxrwxr-x  4 vboxuser vboxuser  4096 Jul 24 14:45 .github/
-rw-rw-r--  1 vboxuser vboxuser  1125 Jul 24 14:45 .gitignore
-rw-rw-r--  1 vboxuser vboxuser   290 Jul 24 14:45 .gitmodules
drwxrwxr-x  2 vboxuser vboxuser  4096 Jul 24 14:45 jenkins/
-rw-rw-r--  1 vboxuser vboxuser  2030 Jul 24 14:45 LICENSE_BUILD_RUN_SCRIPTS
-rw-rw-r--  1 vboxuser vboxuser  4538 Jul 24 14:45 README.md
-rwxrwxr-x  1 vboxuser vboxuser   612 Jul 24 14:45 setup.sh*
drwxrwxr-x  7 vboxuser vboxuser  4096 Jul 24 14:45 tools/

My tools/OpenROAD Directory is empty.
My working directory is OpenROAD-flow-scripts/flow

vboxuser@Ubuntu-Openroad:~/OpenROAD-flow-scripts/flow$ ./etc/Env.sh
bash: ./etc/Env.sh: No such file or directory

I can't remember if i use sudo for the openroad build

@maliberty

Make: 4.2.1
Ubuntu 20.04 LTS

Thank all for your help.

vijayank88, 2023-08-11:
> My tools/OpenROAD Directory is empty.
> My working directory is OpenROAD-flow-scripts/flow

Can you point link that you followed the instructions for installation? @JJs-Pi

JJs-Pi, 2023-08-11:
I use it in a virtual machine and i think i used this:
https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildLocally.md

vijayank88, 2023-08-11:
If you followed the instruction properly, then ` /home/vboxuser/OpenROAD-flow-scripts/tools/OpenROAD` will be available and able to run `./etc/Env.sh`. Maybe something went wrong at your end.

Follow below instruction again and install again
```
git checkout master
git pull
git submodule update --init --recursive
sudo ./setup.sh
./build_openroad.sh --local
```
Then verify with https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildLocally.md#verify-installation

JJs-Pi, 2023-08-11:
It is not possible to do the installation again because it is an stand alone pc.
If i be in the OpenROAD-flow-scripts directory and i enter: source ./env.sh
i get back : OPENROAD: /home/vboxuser/OpenROAD-flow-scripts/tools/openROAD

vijayank88, 2023-08-11:
Then try
```
cd /home/vboxuser/OpenROAD-flow-scripts/tools/openROAD
./etc/Env.sh
```

As you're using Ubuntu20.04, use pre-built binaries method from here: https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/docs/user/BuildWithPrebuilt.md as a fresh installation in separate directory.

JJs-Pi, 2023-08-11:
bash: ./etc/Env.sh: No such file or directory

maliberty, 2023-08-11:
"My tools/OpenROAD Directory is empty." is a bad sign.  It suggests you didn't update the submodules and have no build.

I don't understand "It is not possible to do the installation again because it is an stand alone pc.".  Why not?

JJs-Pi, 2023-08-15:
On the first day of installation everything worked. Even the odb files. After that it only works so far that it writes The 1_1_yosis.v, 1_synth.sdc and 1_synth.v files. That means that it works in general. You just have to solve this acess problem. 
The installation was done on a VM and then transferred to this PC. This PC has no internet connection and is not able to update.

vijayank88, 2023-08-19:
@vvbandeira 
just copying files will work as mentioned above by @JJs-Pi ?

vvbandeira, 2023-08-21:
@vijayank88 @JJs-Pi 
Well, it depends.
1. If you are copying just the source code between machines, that is ok as long as you copy the right files to the right place and the machine without internet access already has all the dependencies installed. Otherwise, it will not matter.
2. If you are copying the binary, the machines must use the same OS and have the same setup/dependencies must have been installed in the same place.
3. If you are copying the `.deb` installer, that will need internet access to resolve dependencies.

Either way, this process is very brittle and easy to make a wrong assumption/mistake. Possibles solutions:
1. If you can, bring the machine back to the internet to do the initial setup. After you have a working environment you can disconnect it again if you are concerned with security.
2. Use docker instead and two machines, one with and one without internet. The one with internet you will build the docker image as per our docs, then run:
```
docker save <IMAGE_NAME> | gzip > orfs.tar.gz
# copy the file to the machine without internet and run
docker load --input orfs.tar.gz
```

JJs-Pi, 2023-08-23:
I have reinstalled everything. Oracle VM, Ubuntu 20.04 and the BuildLocally.md. Then exported the appliance and imported it on the stand alone pc. Now everything runs as it should. Thanks anyway for your help.

closed 2023-08-23 by JJs-Pi