The-OpenROAD-Project/OpenROAD-flow-scripts #1342: Evaluate degradation on asap7/riscv32i

vvbandeira, 2023-08-10:
### Description

Design: `asap7/riscv32i`, main metric change `finish__timing__setup__ws` `+81` -> `-5`.

Good:
```
  "run__flow__openroad_version": "v2.0-9791-gea56b56a9",
  "run__flow__scripts_commit": "c83d09863e38a4b01c925621f08e472e39e91f6d",

  "finish__timing__drv__max_cap_limit": 0.0388632,
  "finish__timing__drv__max_slew": 244,
  "finish__timing__drv__hold_violation_count": 0,
  "finish__timing__drv__setup_violation_count": 0,
  "finish__timing__setup__tns": 0,
  "finish__timing__setup__ws": 81.6611,
  "finish__timing__wns_percent_delay": 5.957017,
```

Bad:
```
  "run__flow__openroad_version": "v2.0-9822-g4e4293bbe",
  "run__flow__scripts_commit": "a059a1ba4711a8e63415a41d9488cbbcd1f6778d",

  "finish__timing__drv__max_cap_limit": 0.102151,
  "finish__timing__drv__max_slew": 42,
  "finish__timing__drv__hold_violation_count": 3,
  "finish__timing__drv__setup_violation_count": 1,
  "finish__timing__setup__tns": -5.10636,
  "finish__timing__setup__ws": -5.10636,
  "finish__timing__wns_percent_delay": -0.350328,
```

maliberty, 2023-08-10:
@openroadie I don't know if this is a rsz issue or something else.  Please do an initial triage.

openroadie, 2023-08-12:
@maliberty @vvbandeira as far as I can tell, this is the PR that degraded performance:

https://github.com/The-OpenROAD-Project/OpenROAD/pull/3834

```
Aug   8 : 81.6611: e088bf159a892982cce289c74367193da8b602db
Aug   8 : 81.6611: 83770be334d117a755f6e1a3f3a8e4cd332b8191
Aug   9 : 81.6611: 3934c57955502c90cf0147af0f4b238f137e7afb
Aug   9 : 81.6611: a2fc4465a414194db0acc9395620ce3f6274e802
Aug   9 : 81.6611: 4474922bd6d85e9491e7ffebfb00cbab7e9e64c9
**Aug   9 : -5.1064: 3e12bbef4d35f3f8f922d33935a03c8f35552640** 
Aug   9 : -5.1064: 4e4293bbec4c2d2abb40a9cb78c69bbaa2ab58fc
Aug 10 : -5.1064: b31a3b960dfa4f0cf387e53d6218e9e8f89d3aa8

```
As you can see ... the last SHA before 3e12xx maintained the QOR and then 3e12 has the degraded result.

closed 2025-08-12 by maliberty