The-OpenROAD-Project/OpenROAD-flow-scripts #1355: Make annealing io placement work in a reasonable amount of time on non-abutted mock-array test-case

oharboe, 2023-08-15:
### Description

To make available the changes needed, including changes to the Verilog of mock-array, I have draft pull request and closed it. It can still be accessed as if it was open:

To reproduce, check out https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1354

Then run:

```
make DESIGN_CONFIG=designs/asap7/mock-array/config.mk
```

```
[WARNING PPL-0015] Macro ces_0_0 is not placed.
[WARNING PPL-0015] Macro ces_0_1 is not placed.
[WARNING PPL-0015] Macro ces_0_2 is not placed.
[WARNING PPL-0015] Macro ces_0_3 is not placed.
[WARNING PPL-0015] Macro ces_0_4 is not placed.
[WARNING PPL-0015] Macro ces_0_5 is not placed.
[WARNING PPL-0015] Macro ces_0_6 is not placed.
[WARNING PPL-0015] Macro ces_0_7 is not placed.
Found 0 macro blocks.
Using 2 tracks default min distance between IO pins.
[INFO PPL-0001] Number of slots          9214
[INFO PPL-0002] Number of I/O            2314
[INFO PPL-0003] Number of I/O w/sink     2314
[INFO PPL-0004] Number of I/O w/o sink   1
[doesn't seem to complete, no progress messages]
```

### Suggested Solution

Make io annealing placement faster, or if it can't be fast enough that it can't reasonably be run every time, make it write out a .tcl file that can be used in subsequent runs with MACRO_PLACEMENT_TCL

oharboe, 2023-08-15:
> fwiw I would file this as an OR issue as it isn't really a flow problem

Should I refile? I get these projects mixed up somtimes.

eder-matheus, 2023-08-16:
@oharboe My PR should cover the error in this design. The problem is that pins are constrained to regions without enough room. Now the tool errors out when it happens instead of getting stuck in an infinite loop.

oharboe, 2023-08-16:
> @oharboe My PR should cover the error in this design. The problem is that pins are constrained to regions without enough room. Now the tool errors out when it happens instead of getting stuck in an infinite loop.

Thanks!

@maliberty Here is an example of where debugging the floorplan  and macro layout before debugging IO constraints is useful.

@eder-matheus I fixed a bug in Verilog generation for mock array where rows and cols were swapped.

oharboe, 2023-08-16:
With https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1366, I would have gotten the following error when debugging the floorlplan. The RTL and manual placement doesn't line up so the macros manual floorplan is looking for don't exist.

```
$ make floorplan_debug_macros
[deleted]
Block Element needs to be hardened.
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/oyvind/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/floorplan_debug_macros.tcl -metrics ./logs/asap7/mock-array/base/2_floorplan_debug_macros.json) 2>&1 | tee ./logs/asap7/mock-array/base/2_floorplan_debug_macros.log
OpenROAD v2.0-9875-g7430f83d9 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: macro-placement.tcl, 27 invalid command name "NULL"
Command exited with non-zero status 1
Elapsed time: 0:00.95[h:]min:sec. CPU time: user 0.91 sys 0.04 (100%). Peak memory: 192748KB.
make[2]: *** [Makefile:524: do-2_floorplan_debug_macros] Error 1
make[2]: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/flow'
make[1]: *** [Makefile:524: results/asap7/mock-array/base/2_floorplan_debug_macros.odb] Error 2
make[1]: Leaving directory '/home/oyvind/OpenROAD-flow-scripts/flow'
make: *** [Makefile:922: floorplan_debug_macros] Error 2
```

The error message isn't great, this is because `set inst [$block findInst [format "ces_%d_%d" $row $col]]` doesn't throw an exception when not finding a block, it relies on the caller to check the return value. This is a typical bug of mixing exception throwing programming model and non-exception programming model. I would have like to see findInst throwing an exception and for there to be a hasInst method to check for the presence in the rare cases where that's needed... But I digress...

oharboe, 2023-08-16:
When I rebase https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/1366 and run `make floorplan_debug_macros`, I get the below and everything sails through the flow...

[image]

closed 2023-08-16 by oharboe