The-OpenROAD-Project/OpenROAD-flow-scripts #1365: Command terminated by signal 6

chros098, 2023-08-16:
### Description

After we replaced the process library, we used the original data and scripts. Modify the script according to the process library, and the signal  6 problem occurs when running to ruote. I also want to ask where to set the fanout transition capacitance of sdc.

### Additional Context

[image]
[logs.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12357890/logs.tar.gz)
[results_part1.tzr.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12357891/results_part1.tzr.gz
[results_part2.tzr.gz](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12357892/results_part2.tzr.gz)
)

vijayank88, 2023-08-16:
Please attach valid reproducible test case.
Also missing environment setup. goto `tools/OpenROAD` and run `./etc/Env.sh`, copy & paste resulting logs

chros098, 2023-08-17:
Ok, packaged and uploaded
[image]
[global_place_gcd_nangate45_base_2023-08-16_17-09 (2)(2).zip](https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/files/12365281/global_place_gcd_nangate45_base_2023-08-16_17-09.2.2.zip)

chros098, 2023-08-17:
Can it be analyzed from the log that the problem is caused by poor timing

vijayank88, 2023-08-17:
@chros098 
Unable to run your test case as it's missing the files
```
./run-me-gcd-nangate45-base.sh 
vars-gcd-nangate45-base.sh: line 4: export: `===': not a valid identifier
vars-gcd-nangate45-base.sh: line 5: export: `?=': not a valid identifier
OpenROAD 6a66a6b07a3231b9400ed35d0f4e71eace5f5f4f 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Error: read_liberty.tcl, 25 cannot read file ./platforms/nangate45/lib/NangateOpenCellLibrary_typical.lib.
openroad> 
```

vijayank88, 2023-08-17:
The test case you provided not relevant to the `title` and logs.
Provide steps by step instruction what you've changed and when the crash happened?
Because `nangate45/gcd` is part of CI and able to run successfully.

chros098, 2023-08-17:
yes, I replaced lib, lef, gds and cdl, because I want to use other processes to run flow,
Maybe there is a problem with the file I compressed. I re-uploaded it. Please take a look again
Put it on the cloud disk：
https://drive.google.com/file/d/1blVHIdQNlMkI0LiVkNl2dCvPoFM_d517/view?usp=drive_link

vijayank88, 2023-08-17:
> yes, I replaced lib, lef, gds and cdl, because I want to use other processes to run flow, Maybe there is a problem with the file I compressed. I re-uploaded it. Please take a look again Put it on the cloud disk： https://drive.google.com/file/d/1blVHIdQNlMkI0LiVkNl2dCvPoFM_d517/view?usp=drive_link

Please enable public access.

chros098, 2023-08-17:
sry。done

chros098, 2023-08-17:
https://drive.google.com/file/d/1blVHIdQNlMkI0LiVkNl2dCvPoFM_d517/view?usp=sharing

vijayank88, 2023-08-17:
@osamahammad21 
Can you please check about test case that fails during detail routing stage?

osamahammad21, 2023-08-17:
@chros098 
I am getting this error
[ERROR DRT-0234] via8 does not have single-cut via.
could you please recheck the attached testcase?

chros098, 2023-08-18:
@osamahammad21 
I repackage the test case 
link ：
https://drive.google.com/file/d/1kSC-TpLAFtIfX8hf9aJopLeK079eOpYl/view?usp=sharing

vijayank88, 2023-08-19:
@chros098 
Before fixing the issue, what you're trying to achieve with the changes you made with nangate45 PDK?

osamahammad21, 2023-08-20:
@chros098 I found the part of the code that leads to this segmentation fault. I am discussing it currently with Matt to settle on a solution for this.
The main problem is the ill-formed layer via8 that does not have any vias defined in the tech lef. Could you clarify to me why are you using such layer?

chros098, 2023-08-21:
@vijayank88 We mainly want to use our own process library to run flow, but we are not professional in digital aspects, which will cause many basic problems 。The script is based on nangate45, and the environment is also based on nangate45。

chros098, 2023-08-21:
@osamahammad21 I will check the script of the whole process again, because this script is based on nangate45 and modified from nangate45's script

osamahammad21, 2023-08-22:
@chros098 We've settled to log an error whenever there is a cut layer that doesn't have any via defined. So from our part we will create a PR with the error added. For you to fix this test case, you need to fix that layer by either adding the required vias or removing the layer at all.

chros098, 2023-08-24:
@osamahammad21 @vijayank88
Hello, I have modified the problem of via8, and then continued to run, the problem of signal11 appeared。
The picture below is the error reported
[image]
The data used is packed,put on the cloud disk
https://drive.google.com/file/d/1jefhCSjbj71KR638bsYYdVyGsE_CPalv/view?usp=drive_link
Please help me to look again

vijayank88, 2023-08-24:
@chros098 Enable public access to the above link

chros098, 2023-08-24:
@vijayank88 https://drive.google.com/file/d/1jefhCSjbj71KR638bsYYdVyGsE_CPalv/view?usp=drive_link 
update thanks

vijayank88, 2023-08-24:
@chros098 
The attached test case is related to `global_place` stage. Seems you attached wrong test case, the logs shows still an detail routing crash.
Can you re-attach the right test case as per above screen shot, it crash during finish stage?

maliberty, 2023-08-24:
Did you update the rcx tech file to match whatever other modifications you made?

chros098, 2023-08-28:
hello ，I have two questions：
1.After modifying the tech file, we ran through the entire flow and finally generated the gds file. Regardless of using klayout and calibredrv to open the gds file, the filter cell does not display the layer, but this cell is called, and the gds file run by nangate45 also has this problem。
2.The whole process runs smoothly. We modified the grid strategy-M1-M4-M7.tcl file in the pdn stage, so that M1 is not used as the power grid layer, so that it can run smoothly. If you do not modify this file, the following problem will be reported.
[image]
https://drive.google.com/file/d/13Uk1S_odI0u3XN9ieb-_JKBLOf4btMIF/view?usp=drive_link

vijayank88, 2023-08-28:
@chros098 
Please attach a valid test case whenever you're reporting some issues.
In the above screenshot shows pdn issue, but given test case running `global_place` stage.

chros098, 2023-08-28:
thanks 。
I am asking two questions, one is that the filter cell is empty after gds is generated, without any layers. The second problem is that during the pdn process, when the m1 is used as the power network level, an error will be reported.
The link below is the test case for the second problem
https://drive.google.com/file/d/1d5b6o8sSdT6GQHtJbS2-n2GR6IvD-e7Q/view?usp=drive_link
Thanks again

osamahammad21, 2023-08-28:
PDN is out of my scope. @maliberty could you please check who should be responsible for this?

maliberty, 2023-08-28:
Please use make issue to produce you test cases rather than making a huge tar with tons of extraneous file.

maliberty, 2023-08-28:
The original issue here is resolved.  Please open unrelated issues separately.

closed 2023-08-28 by maliberty

chros098, 2023-08-29:
@maliberty  thanks