The-OpenROAD-Project/OpenROAD-flow-scripts #1393: Autotuner Results and Make Results inconsistent (Using parameters chosen by autotuner do make run), Make failes

dalyles, 2023-08-26:
### Subject

[Documentation] for documentation errors.

### Describe the bug

Hello,

I was running the autotuner on one of my circuits named <code>picorv32</code> was successfully able to be generated without any violations. These are the parameter results that was determined for a clean run by the autotuner.

This can be found at the <code>autotuner-best-xxxxxxxx.json</code>
```
{
    "_SDC_CLK_PERIOD": 1,
    "CORE_UTILIZATION": 39,
    "CAP_MARGIN": 39,
    "CORE_ASPECT_RATIO": 1.1435000182737207,
    "CORE_MARGIN": 2,
    "CELL_PAD_IN_SITES_GLOBAL_PLACEMENT": 1,
    "CELL_PAD_IN_SITES_DETAIL_PLACEMENT": 0,
    "_FR_LAYER_ADJUST": 0.15900339639207847,
    "PLACE_DENSITY_LB_ADDON": 0.436940328162555,
    "_PINS_DISTANCE": 1,
    "CTS_CLUSTER_SIZE": 51,
    "CTS_CLUSTER_DIAMETER": 336,
    "_FR_GR_OVERFLOW": 1,
    "best_result": 92.52495
}
```
What I do then is copy these paramters and put them into the associated <code>config.mk</code> file like so:
```
export DESIGN_NAME = picorv32
export PLATFORM    = nangate45

export VERILOG_FILES = ./designs/src/$(DESIGN_NAME)/picorv32.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NAME)/constraint.sdc
export ABC_AREA      = 1

# Adders degrade GCD
export ADDER_MAP_FILE :=

export CORE_UTILIZATION ?= 55
export PLACE_DENSITY_LB_ADDON = 0.20
export TNS_END_PERCENT = 100

export _SDC_CLK_PERIOD = 15
export CORE_UTILIZATION = 39
export CAP_MARGIN = 39
export CORE_ASPECT_RATIO = 1.1435000182737207
export CORE_MARGIN = 2
export CELL_PAD_IN_SITES_GLOBAL_PLACEMENT = 1
export CELL_PAD_IN_SITES_DETAIL_PLACEMENT = 0
export _FR_LAYER_ADJUST = 0.15900339639207847
export PLACE_DENSITY_LB_ADDON = 0.436940328162555
export _PINS_DISTANCE = 1
export CTS_CLUSTER_SIZE = 51
export CTS_CLUSTER_DIAMETER = 336
export _FR_GR_OVERFLOW = 1
```

My expectation would be that given the parameters determined by the autotuner, this would not result in an error (see "Expected Behavior" below), Is there a possiblity that the way I'm setting up the fixed parameters is not correct?

Thank you

### Expected Behavior

The result should be a clean circuit as was done by the autotuner.  Here's a snapshot of <code>6_report.log</code> results from the autotuner
```
==========================================================================
finish max_slew_violation_count
[... 227 lines cut ...]
        ],
        "step": 1
    },
    "CTS_CLUSTER_DIAMETER": {
        "type": "int",
        "minmax": [
            20,
            400
        ],
        "step": 1
    },
    "_FR_FILE_PATH": "../../../platforms/nangate45/fastroute.tcl",
    "_FR_GR_OVERFLOW": {
        "type": "int",
        "minmax": [
            1,
            1
        ],
        "step": 0
    }
}
```

2) Get parameters determined by the autotuner and inset into <code>config.mk</code>
3) Run make with the given parameters, if the autotuner run is clean, the make run should be clean as well.

maliberty, 2023-08-26:
The _ variables do not correspond to flow variables and are handled specially by the autotuner script.  For example _SDC_CLOCK_PERIOD updates the set_clock_period value in the sdc.

dalyles, 2023-08-28:
Is there some kind of correspondence or mapping with with variables such as _SDC_CLOCK_PERIOD, _FR_LAYER_ADJUST, _PINS_DISTANCE, and _FR_GR_OVERFLOW? I'd like to set those values directly in config.mk so I can be able to have a successful run.

Thank you

maliberty, 2023-08-28:
The point is that they don't correspond to anything in config.mk and are handled specially.  See https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/753d91cddbd990fcb82d0592d59645053f3b2d3a/flow/util/distributed.py#L328

dalyles, 2023-08-29:
Hi @maliberty I see what you're saying here.
According to the code, would it be as straightforward as removing the underscore and prefix
for these variables?  I've given this a try and it seems to have not yielded to a clean result.
Here is what I have for my <code>config.mk</code>:
```
export DESIGN_NAME = picorv32
export PLATFORM    = nangate45

export VERILOG_FILES = ./designs/src/$(DESIGN_NAME)/picorv32.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NAME)/constraint.sdc
export ABC_AREA      = 1

# Adders degrade GCD
export ADDER_MAP_FILE :=

export CORE_UTILIZATION ?= 55
export PLACE_DENSITY_LB_ADDON = 0.20
export TNS_END_PERCENT = 100

export CLK_PERIOD = 1
export CORE_UTILIZATION = 39
export CAP_MARGIN = 39
export CORE_ASPECT_RATIO = 1.1435000182737207
export CORE_MARGIN = 2
export CELL_PAD_IN_SITES_GLOBAL_PLACEMENT = 1
export CELL_PAD_IN_SITES_DETAIL_PLACEMENT = 0
export LAYER_ADJUST = 0.15900339639207847
export PLACE_DENSITY_LB_ADDON = 0.436940328162555
export DISTANCE = 1
export CTS_CLUSTER_SIZE = 51
export CTS_CLUSTER_DIAMETER = 336
export GR_OVERFLOW = 1
```
I guess I have a follow-up question then: Does a clean run on the autotuner also imply that a run using make (with the autotuner parameters chosen) will also be clean?

dalyles, 2023-08-29:
I have a follow up question: if there is no such variables that can be set, how can you compare autotuner vs make runs? Is this even possible?

maliberty, 2023-08-29:
You have to modify the corresponding files the same way that the autotuner does.  For example you have to edit the sdc to set the clock period.

dalyles, 2023-08-29:
Ok, I think I better understand what you're saying.

The autotuner already has the files generated for <code>constraint.sdc</code> and <code>fastroute.tcl</code> which includes the parameters that have the appropriate prefixes, so I can use that in my make run. As for the other parameters, I can manually place them in the <code>config.mk</code> file since they belong in the "options" section in <code>distributed.py</code>. Is this understanding correct?

vijayank88, 2023-08-29:
yes @dalyles

dalyles, 2023-08-29:
I just want to confirm with you if this setup is correct (it worked for me, just need a sanity check)
So for my parameter list which I will show below from the autotuner:

```
{
  "CAP_MARGIN": 39,
  "CELL_PAD_IN_SITES_DETAIL_PLACEMENT": 0,
  "CELL_PAD_IN_SITES_GLOBAL_PLACEMENT": 1,
  "CORE_ASPECT_RATIO": 1.1435000182737207,
  "CORE_MARGIN": 2,
  "CORE_UTILIZATION": 39,
  "CTS_CLUSTER_DIAMETER": 336,
  "CTS_CLUSTER_SIZE": 51,
  "PLACE_DENSITY_LB_ADDON": 0.436940328162555,
  "_FR_GR_OVERFLOW": 1,
  "_FR_LAYER_ADJUST": 0.15900339639207847,
  "_PINS_DISTANCE": 1,
  "_SDC_CLK_PERIOD": 1
}
```

I have the parameters distributed as follows:

<code>config.mk</code>
```
current_design picorv32
  
set clk_name  core_clock
set clk_port_name clk
set clk_period 1
set clk_io_pct 0.2

set clk_port [get_ports $clk_port_name]

create_clock -name $clk_name -period $clk_period $clk_port

set non_clock_inputs [lsearch -inline -all -not -exact [all_inputs] $clk_port]

set_input_delay  [expr $clk_period * $clk_io_pct] -clock $clk_name $non_clock_inputs
set_output_delay [expr $clk_period * $clk_io_pct] -clock $clk_name [all_outputs]
```

<code>fastroute.tcl</code>
```
set_global_routing_layer_adjustment metal2-metal3 0.5
set_global_routing_layer_adjustment metal4-$::env(MAX_ROUTING_LAYER) 0.25

set_routing_layers -signal $::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER)

set_global_routing_layer_adjustment $::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER) 0.15900339639207847
```

And finally, the <code>config.mk</code> (note that I changed the fastroute reference to the updated <code>fastroute.tcl</code>)
```
export DESIGN_NAME = picorv32
export PLATFORM    = nangate45

export VERILOG_FILES = ./designs/src/$(DESIGN_NAME)/picorv32.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NAME)/constraint.sdc
export FASTROUTE_TCL = ./fastroute.tcl
export ABC_AREA      = 1

# Adders degrade GCD
export ADDER_MAP_FILE :=

export CORE_UTILIZATION ?= 55
export PLACE_DENSITY_LB_ADDON = 0.20
export TNS_END_PERCENT = 100

export CAP_MARGIN = 39
export CELL_PAD_IN_SITES_DETAIL_PLACEMENT = 0
export CELL_PAD_IN_SITES_GLOBAL_PLACEMENT = 1
export CORE_ASPECT_RATIO = 1.1435000182737207
export CORE_MARGIN = 2
export CORE_UTILIZATION = 39
export CTS_CLUSTER_DIAMETER = 336
export CTS_CLUSTER_SIZE = 51
export PLACE_DENSITY_LB_ADDON = 0.436940328162555
export _PINS_DISTANCE = 1
```

vijayank88, 2023-08-29:
`constraint.sdc` looks ok.

`fastroute.tcl` should be updated as:
```
set_global_routing_layer_adjustment $::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER) 0.15900339639207847
set_routing_layers -signal $::env(MIN_ROUTING_LAYER)-$::env(MAX_ROUTING_LAYER)
```
`config.mk` as follows
```
export DESIGN_NAME = picorv32
export PLATFORM    = nangate45

export VERILOG_FILES = ./designs/src/$(DESIGN_NAME)/picorv32.v
export SDC_FILE      = ./designs/$(PLATFORM)/$(DESIGN_NAME)/constraint.sdc
export FASTROUTE_TCL = ./fastroute.tcl
export ABC_AREA      = 1

export TNS_END_PERCENT = 100

export CAP_MARGIN = 39
export CELL_PAD_IN_SITES_DETAIL_PLACEMENT = 0
export CELL_PAD_IN_SITES_GLOBAL_PLACEMENT = 1
export CORE_ASPECT_RATIO = 1.1435000182737207
export CORE_MARGIN = 2
export CORE_UTILIZATION = 39
export CTS_CLUSTER_DIAMETER = 336
export CTS_CLUSTER_SIZE = 51
export PLACE_DENSITY_LB_ADDON = 0.436940328162555
export PLACE_PINS_ARGS="-min_distance 1"

dalyles, 2023-08-29:
Hi @vijayank88 ,

Thanks so much for your input, as for the <code>fastroute.tcl</code>, I directly copied and pasted from the autotuner's generated <code>fastroute.tcl</code> file so I didn't touch the first two lines of code, just to add context.

I'll update the last line in <code>config.mk</code> and give a run.

Thank you!

dalyles, 2023-08-29:
For some reason, the last line you gave me in <code>config.mk</code>  <code>export PLACE_PINS_ARGS="-min_distance 1"</code> gives an error when I run:

```
Elapsed time: 0:01.75[h:]min:sec. CPU time: user 1.70 sys 0.05 (100%). Peak memory: 138192KB.
(/usr/bin/time -f 'Elapsed time: %E[h:]min:sec. CPU time: user %U sys %S (%P). Peak memory: %MKB.' /home/dlyles/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad -exit -no_init  ./scripts/io_placement_random.tcl -metrics ./logs/nangate45/picorv32/base/2_2_floorplan_io.json) 2>&1 | tee ./logs/nangate45/picorv32/base/2_2_floorplan_io.log
OpenROAD v2.0-9070-gbb341ffb9 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR STA-0402] place_pins -min_distance 1 is not a known keyword or flag.
Error: io_placement_random.tcl, 14 STA-0402
Command exited with non-zero status 1
Elapsed time: 0:00.28[h:]min:sec. CPU time: user 0.25 sys 0.03 (100%). Peak memory: 102284KB.
make: *** [Makefile:473: results/nangate45/picorv32/base/2_2_floorplan_io.odb] Error 1

```

vijayank88, 2023-08-29:
Just replace with `export PLACE_PINS_ARGS=-min_distance 1`

dalyles, 2023-08-29:
It works!

vijayank88, 2023-09-02:
@dalyles is anything else pending here to resolve?

dalyles, 2023-09-05:
This issue has been resolved.

Thanks!

closed 2023-09-05 by vvbandeira