The-OpenROAD-Project/OpenROAD-flow-scripts #1396: mempool_group source files are missing

gadfort, 2023-08-26:
### Subject

[Design] for example design issues (i.e. the design does not pass functional validation, etc.)

### Describe the bug

mempool_group for nangate45 is missing the source files needed to be able to run this design.

### Expected Behavior

Be able to run or remove the design all together

### Environment

```shell
N/A
```

### To Reproduce

N/A

vijayank88, 2023-08-28:
@gadfort 
Similar issue already reported https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/issues/1236

gadfort, 2023-08-28:
@vijayank88 thanks for pointing that out

closed 2023-08-28 by gadfort