The-OpenROAD-Project/OpenROAD-flow-scripts #1405: The width of the generated gds track and metal becomes 10 times the set

chros098, 2023-08-29:
### Subject

[Stage]: GDS generation.

### Describe the bug

Using the gds generated by openroad, the track and metal width in the gds become 10 times the set value, and the filter cell is empty.

Test cases are placed on the cloud disk
https://drive.google.com/file/d/1vytqgR7SvbHEZ3GHWbgww63wENlhfN5f/view?usp=drive_link

### Expected Behavior

I hope to find out whether it is a setting problem or a problem with our operation。

### Environment

```shell
using an older version
```

### To Reproduce

Test cases are placed on the cloud disk
https://drive.google.com/file/d/1vytqgR7SvbHEZ3GHWbgww63wENlhfN5f/view?usp=drive_link

vijayank88, 2023-08-29:
Based on updated LEF file, have you updated KLayout technology file for GDSII generation `nangate45.lyt`?

chros098, 2023-08-29:
This file is not generated, how is this file generated？

vijayank88, 2023-08-29:
check this: https://openroad-flow-scripts.readthedocs.io/en/latest/contrib/PlatformBringUp.html#klayout

chros098, 2023-09-07:
thanks. After generating the lyt file according to the tutorial, I ran the process again. There are metal layers in the cell, etc. However, when importing gds, I found an error message and there was no via cell. This cell seemed to be automatically generated. May I ask where this needs to be set?
[image]

https://drive.google.com/file/d/1p-gcb9Hb4s9BpW6qx3U_TKyQp5VJZ_Q4/view?usp=drive_link

vijayank88, 2023-09-07:
Attach a layer map file to run klayout for gds generation @chros098

chros098, 2023-09-07:
hlleo here？
[image]

vijayank88, 2023-09-07:
I've requested you to attach `*.lyp` file.

chros098, 2023-09-08:
Are you attaching here?
[image]
lyt and lyp files can be helpful here.
https://drive.google.com/file/d/1kUGkuNiqKuGNnlPttsNefqRd97QrZsG2/view?usp=drive_link

vijayank88, 2023-09-09:
If I'm right you're derived pdk from FreePDK45(nangate45). So try to use same `.lyt` and `.lyp` from [here]( https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/master/flow/platforms/nangate45/FreePDK45.lyt) if layer names not changed mean and run the flow and check able to get GDS file. @chros098

maliberty, 2024-03-06:
This seems to have trailed off.  Please reopen if still relevant.

closed 2024-03-06 by maliberty